MarlinFirmware/MarlinDev #133: SDSSUPPORT should be switchable

justmyopinion, 2015-09-04:
My azteeg x3 motherboard is supplied with a Vicki display with embedded Sd card. However I never use this card facility and it is not even wired up.
However Marlin does not accept this and try to access sdcard and issue not existing dir lists to serial.
I tried to disable SDSUPPORT as Marlin is hanging waiting for reponse from SDCARD and introduce delys of 10-15 seconds to commands, but this is a jungle embedded in different display variants (NEWPANEL,ULTRAPANEL etc) and it looks like several faciliy tests in code has not been supplied with SD support conditions so disabling SDSUPPORT this way makes it "not compileable"..
It seems impossible to switch SD spport off, so my suggestion is that SDSUPPORT should be a general feature to be switched globally outside any depending panel. 
I have only tested this with rel version (1.03?) so any changes in dev is not included.

Wackerbarth, 2015-09-04:
I agree. @thinkyhead, this is your area. What is your opinion?

thinkyhead, 2015-09-04:
Seems like a good idea – for consistency with other controllers. Check out #139 which makes it right.

justmyopinion, 2015-09-05:
With a switchable SDSUPPORT attention is also needed to dependecies like SD_DETECT and display updates.
Display menues should not contain anything related to SD card if SDSUPPORT is disabled and SD_DETECT activity should never occur even if SD_DETECT_PIN is defined.

justmyopinion, 2015-09-14:
I have made some verification of changes in current master for this implementation and as far as my testing is limited it looks to me that problem is solved. 
SDSUPPORT now can be switched on/off as suggested. Thanks

closed 2015-09-14 by justmyopinion