MarlinFirmware/MarlinDev #135: RRD Smart LCD 2004 "no SD-Card" error

o-lampe, 2015-09-04:
When I power On my Ramps with above mentioned LCD and the SD-Card is already inserted, I see " no SD-Card" message in main display. I have to remove and plug it back in to have access to SD-files. 
Not a biggy, but with autostart option enabled, it wouldn´t work.

Wackerbarth, 2015-09-04:
There was an issue about this which was raised earlier and I thought that it had been fixed. Are you using the latest, or at least a very recent, version of the dev firmware?

o-lampe, 2015-09-04:
@Wackerbarth  I´ve downloaded the last version two days ago, maybe I haven´t set all the right options?
Usually I disable #define SDsupport
and enable RRD_Smart Controller only. Worked fine in the past.

thinkyhead, 2015-09-05:
Going forward you will have to enable `SDSUPPORT` explicitly in `Configuration.h`. That shouldn't have any effect on how it behaves, however. If you have `SD_DETECT_PIN` defined (and the correct `SD_DETECT_INVERTED` setting) then Marlin should automatically detect the presence of an already-inserted SD card. If it's not, maybe there's some way to twiddle the detection pin to cause it to refresh its state. Just spitballing…

alexandrezia, 2015-09-06:
I have the same issue here, On boot it doesn't detect an already inserted SD card
on latest commit  b23d7659 
using board:  #define REPRAP_DISCOUNT_SMART_CONTROLLER
Played with all combinations of SDSUPPORT and SD_DETECT_INVERTED.

o-lampe, 2015-09-06:
@thinkyhead  enabling SDSUPPORT didn´t help. Don´t know where the other variables you mentioned are located. In the past, I never had to change anything outside the configuration.h to make a printer work.

thinkyhead, 2015-09-06:
@alexandrezia Try this as a test. Replace the line which reads `lcd_oldcardstatus = IS_SD_INSERTED;` with `lcd_oldcardstatus = false;` and see if it causes the initialization routine to do the right thing. (Namely, to see a change in status.)

o-lampe, 2015-09-07:
`<addr>`
#if ENABLED(SDSUPPORT) && PIN_EXISTS(SD_DETECT)
    pinMode(SD_DETECT_PIN, INPUT);
    WRITE(SD_DETECT_PIN, HIGH);
    lcd_oldcardstatus = false;
  #endif
`<addr>`

I´m not an expert, but is it common to define a pin as *input and then "write" to it, as if it was an *output?

o-lampe, 2015-09-07:
@thinkyhead changing it to `lcd_oldcardstatus = false` helps to recognize the SD-card at startup, but the status line is overwritten by "Card inserted" immediately, so I can´t see the "{user defined text} ready " anymore. I used the line, to see which FW-version I´m currently using

fmalpartida, 2015-09-07:
Yes that is usual in the AVR processor family to set the internal pull-ups.

o-lampe, 2015-09-08:
@fmalpartida I believe the way to define a pullup is to use `pinMode(SD_DETECT_PIN, **input_pullup**)

alexandrezia, 2015-09-09:
Well, I did test the 
```
lcd_oldcardstatus = false;
```
And now it recognizes the sdcard on boot and it's working fine.
The only thing I've noted is that as mentioned by @o-lampe it overwrites the last line since boot and I can't check the "user defined text" that usually sits on that last line.
I think this should be adressed before code freezes to avoid merging this into Release branch

thanks

thinkyhead, 2015-09-13:
Do we know if the code should definitely be `pinMode(SD_DETECT_PIN, INPUT_PULLUP)` instead of `pinMode(SD_DETECT_PIN, INPUT)`? Does that have any effect on the behavior (as opposed to using `lcd_oldcardstatus = false`)?

Wackerbarth, 2015-09-13:
The distinction between INPUT and INPUT_PULLUP is a hardware issue. Does the hardware provide the pull_up, or do we need to supply that from the MPU?

fmalpartida, 2015-09-13:
On AVR based processors and a range of PICs all IO lines have a pull-up.

Wackerbarth, 2015-09-13:
@fmalpartida -- They have them available. However, activation is a separate issue.

fmalpartida, 2015-09-13:
It very much depends on what you connect to the pin indeed. If you have an open collector/drain configuration or if it doesn’t.

thinkyhead, 2015-09-14:
> it overwrites the last line since boot

#180 should take care of that. Testing of SD status is deferred until the first LCD update.

o-lampe, 2015-09-14:
My concern about the code was more related to the second line
`pinMode(SD_DETECT_PIN, INPUT);
  WRITE(SD_DETECT_PIN, HIGH);`
Is it possible to send a write command to an input?

Wurstnase, 2015-09-14:
> Is it possible to send a write command to an input?

Called pullup.

Wackerbarth, 2015-09-14:
Yes, you can do so. However, it should not be allowed in that form.  That is the whole purpose of INPUT_PULLUP.

This points out an error in the previous code that assumes certain hardware characteristics rather than specifying them back in the hardware definition.  We really need to introduce a HAL layer and move things like this to that layer.

fmalpartida, 2015-09-14:
Fully agree.

Wurstnase, 2015-09-14:
Yes, that fully true. On most (if not all) AVR this works. But ARM has different registers for that.

Wackerbarth, 2015-09-14:
@Wurstnase -- All the more reason to move this to a HAL.

thinkyhead, 2016-04-18:
I assume we fixed this bug in `RCBugFix`…? If so, we can close this issue. Code from `RCBugFix` will be getting migrated to `MarlinDev` (and reorganized) right after 1.1.0 release.

closed 2016-05-25 by boelle