MarlinFirmware/MarlinDev #136: Toshiba AirFlash SD/WiFi card support?

o-lampe, 2015-09-04:
Hi, I´m sorry to flood the issue section so often lately, but I´m chin deep into configuring my first delta w. marlin.
Q: Is it possible to support the AirFlash SD-Cards? Make it an option, whether USB or WiFi is chosen.
It would greatly help to configure the printer without moving the laptop around and the well-known "melting USB" cable from bad ground loops are history, too.

ntoff, 2015-09-05:
Do you think by putting one of those wireless sd cards in there, that'll let you control the printer via wifi?

o-lampe, 2015-09-06:
@ntoff  It works for the AZSMZ mini, why not for other SD-Card-boards? Is there such a big difference between USB-control and WiFi?

closed 2016-05-04 by jbrazio