MarlinFirmware/MarlinDev #140: G29 (autolevel) usage? -Delta build-

otvald, 2015-09-05:
I assume that G29 is always preceded by G28. I.e. Do G28, then do G29.

Doing G28 will reset my home to 0,0,300.35. 

When I start G29, it seem like the SW want to move the probe UP.

--------------------------------
G29 Auto Bed Leveling
reset_bed_level
setup_for_endstop_move > enable_endstops(true)
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
z_before = (before) 15.00
probe_pt >>>
> ProbeAction:0
> current_position: (0.00, 0.00, 290.35)
Z Raise to z_before 15.00
> do_blocking_move_to_z 15.00
do_blocking_move_to: (0.00, 0.00, 15.00)
--------------------------------

As mentioned, the probe now wants to move UP, but it should move DOWN to 15.00.

Any further debugging I can do?

otvald, 2015-09-05:
Another test: It seem like it is assumed we start at 0,0,0, and the z axis is reversed
G28, then move to 0,0,200
Then issue G29:

2015-09-05 15:35:50,618 - RECV: G29 Auto Bed Leveling
2015-09-05 15:35:50,619 - RECV: reset_bed_level
2015-09-05 15:35:50,619 - RECV: setup_for_endstop_move > enable_endstops(true)
2015-09-05 15:35:50,620 - RECV: z_before = (before) 15.00
2015-09-05 15:35:50,622 - RECV: z_before = (before) 15.00
2015-09-05 15:35:50,623 - RECV: z_before = (before) 15.00
2015-09-05 15:35:50,623 - RECV: z_before = (before) 15.00
2015-09-05 15:35:50,627 - RECV: z_before = (before) 15.00
2015-09-05 15:35:50,627 - RECV: probe_pt >>>
2015-09-05 15:35:50,627 - RECV: > ProbeAction:0

2015-09-05 15:35:50,627 - RECV: > current_position: (0.00, 0.00, 200.00)

2015-09-05 15:35:50,631 - RECV: Z Raise to z_before 15.00
2015-09-05 15:35:50,631 - RECV: > do_blocking_move_to_z 15.00
2015-09-05 15:35:50,632 - RECV: do_blocking_move_to: (0.00, 0.00, 15.00)
2015-09-05 15:35:50,635 - RECV: prepare_move_raw > destination: (0.00, 0.00, 15.00)
2015-09-05 15:35:50,639 - RECV: > do_blocking_move_to_xy 0.00, -60.00
2015-09-05 15:35:50,643 - RECV: do_blocking_move_to: (0.00, -60.00, 15.00)
2015-09-05 15:35:50,643 - RECV: prepare_move_raw > destination: (0.00, -60.00, 15.00)
2015-09-05 15:35:50,647 - RECV: run_z_probe (DELTA) 1
2015-09-05 15:35:50,651 - RECV: prepare_move_raw > destination: (0.00, -60.00, -10.00)
2015-09-05 15:35:50,656 - RECV: run_z_probe (DELTA) 2 > current_position: (0.00, -60.00, 14.99)
2015-09-05 15:35:50,656 - RECV: <<< probe_pt
2015-09-05 15:35:50,659 - RECV: z_before = (between) 19.99
2015-09-05 15:35:50,660 - RECV: z_before = (between) 19.99
2015-09-05 15:35:50,660 - RECV: z_before = (between) 19.99
2015-09-05 15:35:50,661 - RECV: z_before = (between) 19.99
2015-09-05 15:35:50,664 - RECV: z_before = (between) 19.99
2015-09-05 15:35:50,664 - RECV: z_before = (between) 19.99
2015-09-05 15:35:50,664 - RECV: z_before = (between) 19.99
2015-09-05 15:35:50,667 - RECV: probe_pt >>>
2015-09-05 15:35:50,668 - RECV: > ProbeAction:0

2015-09-05 15:35:50,668 - RECV: > current_position: (0.00, -60.00, 14.99)

This is VERY wrong. The position is about 214.99!

clefranc, 2015-09-05:
@otvald Can you put your firmware/config on your GitHub account? I just began to use G29 properly, after A LOT of trial and error. Perhaps I could help...

otvald, 2015-09-05:
Forked, and added my config.

clefranc, 2015-09-05:
@otvald Can you give us some description of your delta, which kind of Z probe, deploy/stow method, etc.

otvald, 2015-09-05:
The printer is a K800 china 'kossel'. The Z-probe is integrated into the hotend spider. See ' Auto bed leveling not level properly #2413 ', where I have added an image and link to a video showing the behaviour. Basically, the hotend touching the bed will cause the z-endstop to trigger.

I am able to manually level out everything, and using manual calibration, it prints OK.

clefranc, 2015-09-06:
@otvald I notice you didn't define `Z_MIN_PROBE_ENDSTOP`. This must be defined in your `Configuration.h` file if you want the `Z_MIN_PROBE_PIN` to use pin 32 in your `pins_RAMPS_13.h` board.

clefranc, 2015-09-06:
Sanity check should have catched this.

otvald, 2015-09-06:
@clefranc Thanks for the input, but still no success :(

Same result: Still goes *UP*, instead of down.

clefranc, 2015-09-06:
@otvald Is the Z probe register correctly with M119?

otvald, 2015-09-06:
Ooops, No. It s always triggered... Hmm, how to change..

clefranc, 2015-09-06:
First, visual inspection of the wiring. Is this a simple microswitch, or a endstop like at the top of your printer? You can use a multimeter to check the signal.

otvald, 2015-09-06:
It seem that z probe is wired as z_min (I/O 18). I tried to change Z_MIN_PROBE_PIN to 18, but this did not affect z_probe, it still says 'TRIGGERED' (z_min correctly says open/TRIGGERED, when toggling the switch).

Turns out that Z_MIN_PROBE_PIN exists *twice* in pin_RAMPS_13.h (Why didn't the compile complain about the redefinition?).

Wouldn't it be better to enforce setting a value different from the default -1? The default setting (line 43) is Z_MIN_PROBE_PIN = -1. Then, if this pin is required to be configured, an error can be thrown if the default value still exists.. I have uploaded pins_RAMPS_13.h with changes on my branch.

Please comment on this approach.

Using M119, I see that z_probe triggers correct.

It still seem like the probe moves the wrong way.. It moves *up* when issuing G29.

G28, then M114:
X:0.00 Y:0.00 Z:300.35 E:0.00 Count X: 489.14 Y:489.14 Z:489.14

I then issue G29, and it just tries to move up, and hits end stops. M114 the returns:
X:0.00 Y:60.00 Z:254.51 E:0.00 Count X: 414.46 Y:414.46 Z:466.53

This is way off. The position is *still* 0 0 300.35

Any clues?

clefranc, 2015-09-06:
@otvald My `pins_RUMBA.h` have the same logic than your board. Have you define `Z_MIN_PROBE_ENDSTOP` like I said? I don't think your pin code correction is correct. And the sanity check file have already verification for these, perhaps it needs a closer look.

Anyway, I presume your printer is fully operational without G29?

otvald, 2015-09-06:
@clefranc First; Yes, the board is working without G29 (by manual calibration).

I did try defining Z_MIN_PROBE_PIN as you suggested (It was actually already set due to Z_MIN_PROBE_ENDSTOP was set. pins_RAMPS_13.h would explicitly set Z_MIN_PROBE_PIN if Z_MIN_PROBE_ENDSTOP is enabled). - That is why the sanity check passed.. it would be defined!

On my ramps, pin 32 is part of the AUX4 connector. Here, I have attached a display, so the pin is not available. As said, I can verify z_probe using M119, and see that it is open/TRIGGERED correctly (when I manually activate the switch).

Guess I have to check the code as of why the it is moving in the wrong direction..

Strange!

clefranc, 2015-09-06:
Indeed, will take another look at your config and code tonight. Good luck!

thinkyhead, 2015-09-06:
> Turns out that `Z_MIN_PROBE_PIN` exists twice in `pin_RAMPS_13.h` (Why didn't the compile complain about the redefinition?)

I've got a PR #144 that will patch that for RAMPS and RUMBA. It's a bug to redefine a `#define` without `#undef` even though the compiler will allow it.

Wackerbarth, 2015-09-09:
@thinkyhead -- Actually, I think that you may be incorrect. It is my understanding that ANSI-C allows duplicate #define statements provided that "the two definitions are lexically identical." Some compiler implementations permit even looser equivalence, issuing a warning, but not an error.

That's not to say that it is bad to remove the duplication.

And, I suggest that, unless they are being used to crete alternate codes during repeated inclusions of a single file, any use of #undef is a poor programming practice.

otvald, 2015-09-09:
@clefranc 
Hmm, failed to post my findings.. Had to write post again; I have uploaded a detailed log on http://pastebin.com/GSxNNbw9.

@Wackerbarth I quote you here too, since I could not reply in #2413 as it was closed.

It shows:
G28
M111 S32 (enables autolevel debug)
G29

I suspect the problem is in 'do_blocking_move_to'. It seem like prepare_move_raw() is OK, the line plan is not fulfilled. I expected st_synchronize() to wait for completion.

Looking at the log, I see:
2015-09-06 22:40:37,117 - RECV: > current_position: (0.00, 0.00, 300.35)
2015-09-06 22:40:37,120 - RECV: Z Raise to z_before 15.00
2015-09-06 22:40:37,121 - RECV: > do_blocking_move_to_z 15.00
2015-09-06 22:40:37,121 - RECV: do_blocking_move_to: (0.00, 0.00, 15.00)
2015-09-06 22:40:37,124 - RECV: prepare_move_raw > destination: (0.00, 0.00, 15.00)
2015-09-06 22:40:37,125 - RECV: cartesian x=0.00 y=0.00 z=15.00
2015-09-06 22:40:37,128 - RECV: delta a=203.79 b=203.79 c=203.79
2015-09-06 22:40:37,132 - RECV: echo:plan_buffer_line (dx): -28535.00, (dy):-28535.00, (dz):-28535.00
2015-09-06 22:40:37,137 - RECV: > do_blocking_move_to_xy 0.00, -60.00
2015-09-06 22:40:37,137 - RECV: do_blocking_move_to: (0.00, -60.00, 15.00)

This tells me that
* the starting point is known!
* Plan for moving to 0 0 15
* plan_buffer_line seem also OK (anyone can confirm it looks OK?)
* But as seen in the log, the movement (should take several seconds) is almost instantaneous. It seem like the st_synchronize() is not working, or the pan is wrong??

Any ideas?

otvald, 2015-09-13:
@Wackerbarth  Got it working.

I am not using X_MIN or Y_MIN endstops, and they are unconnected. When the X_MIN_ENDSTOP_INVERTING was not set to false, there would continously be triggeren an endstop.

This issue can be closed!

closed 2015-09-13 by otvald