MarlinFirmware/MarlinDev #141: One extruder but two nozzles in LCD menu

o-lampe, 2015-09-06:
I´m running arduino / ramps and have one extruder in configuration.h
But the RRD LCD 2004 tuning menue shows Nozzle and Nozzle1. Also shows three flow subitems: 
Flow ( general )
Flow1
Flow2

Latest dev-contribution

thinkyhead, 2015-09-07:
It will show those items if you set `TEMP_SENSOR_1`, `TEMP_SENSOR_2`, etc. to any non-zero value. For the moment we might fix that by constraining them by the number of extruders as in some other cases. In later iterations we may have multiple heaters or nozzles per extruder.

o-lampe, 2015-09-07:
My bad, I had temp1 configured as -1, because this usually indicates "not used" I should have read the thermistor table :/

thinkyhead, 2015-09-14:
@o-lampe It shouldn't be a problem any more, either way. I've patched up the menus to show everything as it should based on the number of extruders.

closed 2015-09-14 by thinkyhead