MarlinFirmware/MarlinDev #142: Move z-axis up before retracing servo

doomstrike, 2015-09-06:
I have auto-bed leveling enable in Marlin 1.02 for my Lautr3k printer. I have a problem in the z-home dance because after the probe makes its first contact with the bed it tries to retract the probe before continuing to the next probe location. The problem is that when it tries to retract the probe it can't complete the retracting because the tail end of the switch hits the bed and it gets stuck there.

As a solution I would need the printer to retract Z a little so that the switch arm has room to complete the retraction. I have the following set but it doesn't seem to be working or it doesn't try to raise the Z until the servo retracting is complete :(.

#define Z_RAISE_BEFORE_PROBING 13   
#define Z_RAISE_BETWEEN_PROBINGS 3

thinkyhead, 2015-09-07:
Marlin 1.1.0 dev has addressed this issue. It may be a long time before the fix is back-ported to 1.0.2, or it may not be back-ported to 1.0.2 at all depending on available resources. Are you able to use 1.1.0 dev?

doomstrike, 2015-09-07:
@thinkyhead Tried 1.1.0 Dev build as suggested but now I have a problem of the probe not extending when doing a Z home.

thinkyhead, 2015-09-13:
@doomstrike Could you enable the `DEBUG_LEVELING_FEATURE` option and re-flash? Then you'll be able to use `M115 S32` to get additional logging output when doing homing and leveling, and we'll be able to see where in the code the probe is supposed to get extended, but failing to do so.

heezy101, 2015-11-03:
I'm having this same issue.  Is there a fix at this time?

When I had my probe mounted poorly the servo was able to power through the contact with the bed and retract fully, however I was getting poor measurements.  I moved the servo so it is now solidly mounted and the Z end stop/probe hit the bed when retracting, which keeps the probe from retracting.  If the Z axis moved 1mm before retracting the probe I wouldn't have this issue.

Please let me know if there is a fix!  Thank you!

KiteLab, 2015-11-03:
@heezy101 
Please tell us a bit more about your branch and system.

heezy101, 2015-11-04:
I'm sorry, I'm new to this and I'm not sure what you're asking.  I'd be happy to provide any info you need!

In the mean time I swapped out the end stop for one with a longer "trigger lever" on it to get me going again.  I would still like to find a solution to my original issue as I feel that an end stop with the metal lever on it will give me a more accurate measurement.

Roxy-3DPrintBoard, 2015-11-04:
There is also a  Z_RAISE_AFTER_PROBING setting now.  (At least in RC2).   The truth is, if we were to redo this I think we would do it differently.   The Z_RAISE_?????   settings got added one after another to fix problems with retracting and deploying the probes.   It is best if you can have your probe clear such that it can always retract and engage.

But of course, that isn't the real world.   So first the Z_RAISE_BEFORE_PROBING #define happened.  That handled most of the cases.  But there were still issues with some probe designs.  So, Z_RAISE_BETWEEN_PROBINGS got added.  That fixed most of the remaining cases.  But there were still issues.  So, Z_RAISE_AFTER_PROBING got added.   And guess what?  We still have corner cases.

But anyway...   The root of the problem is some probe designs need more space to deploy and retract than where they can be in either final state.   Probably, the Z_RAISE_AFTER_PROBING can get you out of the ditch.   Please try RC2 and report back either way things go.

heezy101, 2015-11-04:
@Roxy-3DPrintBoard Thank you for the reply!  It seems that Z_RAISE_AFTER_PROBING would fix my issue.  I see Z_RAISE_BEFORE_PROBING and Z_RAISE_BETWEEN_PROBINGS but not Z_RAISE_AFTER_PROBING  in my firmware.

I'm not sure what you mean by RC2 but using it gives me the option for Z_RAISE_AFTER_PROBING then I'll use it for sure!  Thank you again for all of your help.

Roxy-3DPrintBoard, 2015-11-04:
It is just called RC but it is the 2nd Release Candidate.    If you go
here,you can grab it and see if it helps your problem.

https://github.com/MarlinFirmware/Marlin/tree/RC

heezy101, 2015-11-04:
Got it.  Thanks again!

What is the best procedure for bringing over settings from my current version to the new one when I download it?  Obviously I can do it manually but I wasn't sure if there is a better/easier/faster way that I'm not aware of.

Roxy-3DPrintBoard, 2015-11-04:
I typically use a visual diff program.   I like ExamDiff Pro.   There is a trial version you can use to give it a try.   It makes it very easy to see what is different and to cross changes over from one file to another.   Otherwise, it is very easy to miss a number here or there.

heezy101, 2015-11-04:
Thank you.  I will look into that!

nitewing76, 2015-11-18:
I use Notepad++  and the "Compare" plugin (both are free).

heezy101, 2015-11-18:
I ended up using Notepad to hold the "old" config file and transferred everything over.  The new firmware fixed my initial issue and everything has been working great since.  I tightened up my Y axis belt and printed an X axis tensioner.  I also found and fixed some slop between my bed plate and heated bed/glass.

The printer is producing the highest quality prints to date!  I still need to add a filament fan as right now I only have a fan on the electronics and hot end.  To do that I want to design an X axis carriage that integrates the hot end mount, hot end fan mount, filament fan mount, and lights.  I'm still working on that.

Thank you all for your help!

Roxy-3DPrintBoard, 2015-11-18:
> I use Notepad++ and the "Compare" plugin (both are free).

I just installed the pair to check them out.  This is a surprisingly good solution.  It looks like ExamDiff Pro is better...   But for a free solution, it really is good.  The one strange thing I noticed is they don't have an "Ignore Case" option for the comparison.  This is so easy to implement and so useful it is strange it is not in there.

thinkyhead, 2016-04-19:
As the code in `RCBugFix` is going to form the basis for MarlinDev going forward, let's close this and we can reopen if these issues haven't finally been wrangled there.

closed 2016-04-19 by thinkyhead