MarlinFirmware/MarlinDev #150: Temperature Fault error

RetireeJay46, 2015-09-07:
I am using the Marlinfirmware/Marlindev version that I downloaded on August 12.  It mostly works OK on my Printrbot Plus with a Printrboard Rev D.  But recently I've had problems with the firmware knocking me out on a supposed temperature error.  I believe that the problem is my printhead is heating up more slowly than the firmware expects.
I tried to fix this by tripling the tolerances in Configuration_Adv.h, but to no avail.  

#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 120        // Seconds (originally 40)
  #define THERMAL_PROTECTION_HYSTERESIS 12     // Degrees Celsius (originally 4)
  #define WATCH_TEMP_PERIOD 48                // Seconds (originally 16)
  #define WATCH_TEMP_INCREASE 12               // Degrees Celsius (originally 4)
#endif

The temperature still faults out.
[image]

It's ugly when it does this:  Repetier Host completely loses communication with the board, and I have to force R-H to shut down and I have to reboot the Printrboard manually.

I'm not a C programmer, so I can't quite untangle the logic of the program or the source of the error in temperature.cpp.

RetireeJay46, 2015-09-07:
By the way, I tried another version with the line 
//#define THERMAL_PROTECTION_HOTENDS  
commented out in Configuration.h, and the firmware allowed my hot end to reach operating temperature.  It took 6 minutes (I have a fan on the body of the extruder, which slows it down considerably from the rise time without a fan).

So the issue is definitely within the code that tries to do thermal protection, but it's not clear (to me) how to set up (modify) the operating parameters correctly.

I guess I would have to ask, what scenario is this code supposed to protect against?  If a MOSFET is shorted "ON" then nothing you do in the program is going to prevent thermal runaway.  But it would seem in that case that you would want to issue a warning to the user, and then keep the program running and the communication line open with the host.  The user might want to move the printhead away from the print, for example.  Making the entire program "die" doesn't seem like it solves anything.  (Exception:  in the case of boards which can shut down their own ATX supply, then doing so would be prudent.)

KiteLab, 2015-09-07:
THERMAL_PROTECTION_HOTENDS tries to detect if a temperature sensor failed. When the heater is full on but the temperature does not rise enough it will threw an error and shut down Marlin.

Your hotend is heating slow - the slope of the temperature curve was about 0,08° per second in the last minute before the error. I expect, until 230° are reached the slope will be well below .0.04° per minute
The predefined values in Marlin:
#define THERMAL_PROTECTION_PERIOD 40
#define THERMAL_PROTECTION_HYSTERES 4
describe a slope of 4/40 = 0.1° per second.
Your changes:
#define THERMAL_PROTECTION_PERIOD 120 // Seconds (originally 40)
#define THERMAL_PROTECTION_HYSTERESIS 12 // Degrees Celsius (originally 4)
are describing exactly the same slope. 12/120 = 0.1
You changed nothing.

If you __increase__ the __time__ and or __decrease__ the __temperature difference__ you will go in the right direction.

You don't want a difference below 1 because of noise of the temperature sensor and you don't wont a very long time because the time to react than will be very long.

My rough guess for your hotend would be about 120, 2.

RetireeJay46, 2015-09-08:
Perfect, now I understand.  Thank you.

closed 2015-09-08 by RetireeJay46