MarlinFirmware/MarlinDev #152: Weather Alert. A Freeze is coming! (Yes, I know the it is 100ºF outside today)

Wackerbarth, 2015-09-08:
OK, gang, things have been dragging on far too long. We have to draw the line somewhere.

Therefore, unless you have something that is "complete" and "ready", it needs to wait for the next cycle.

That doesn't mean that we are uninterested, but we DO need to "publish" the current accumulated changes and then introduce some structural changes that should improve the maintainability of our code base.

So, it your change hasn't been incorporated, advocate on its behalf that it is "ready" and won't cause problems.

Unless someone has a convincing argument to the contrary, the CODE which generates the firmware will be FROZEN this weekend.

clefranc, 2015-09-08:
@Wackerbarth Is this only mean that a branch will be promoted as release candidate in MarlinFirmware/Marlin, and only debug work on it? I think the dev branch here should go on, even if it's not your goal, there is already a good motion in contributors and users, don't let this go down.

I propose you start a new Marlin repository, called MarlinFirmware/MarlinNG, with clear design and clean new code, and build the next Marlin generation on it, eventually replacing MarlinFirmware/MarlinDev with a major release.

Sniffle, 2015-09-08:
No he is saying freeze the code to as it sits right now, promote to the
latest release and then start making large overhauls to the code base for
future maintainability.  Unless i am missing something... The thing is this
is the most development i have seen on marlin in quite some time, and for
it to progress further major changes need to occur and right now we're not
doing much but spinning wheels because until the major changes occur not
much else can change... After the sweeping changes that are planned soon it
could almost be considered marlin 2.0

Roxy-3DPrintBoard, 2015-09-08:
> No he is saying freeze the code to as it sits right now, promote to the
latest release a

Unless somebody has something very good that is really close...  The code is frozen.

fmalpartida, 2015-09-08:
I am very much in favor of RC1 and move on to bug fixing it for eventual release. 
Work in parallel on development to continue evolution for the next iteration. 
+1 Richard

CONSULitAS, 2015-09-08:
:+1: from me

And we need not fix every bug for a release! Important are the major/critical bugs. For the other ones a documentation of the bugs could be enough.

BTW: We will (perhaps) never have a bugfree codebase...

So waiting for a bugfree moment would lock us forever!

CONSULitAS, 2015-09-08:
if you ask only for verified bugs (https://github.com/MarlinFirmware/MarlinDev/labels/Verified%20bug!) we have **only 12 bugs**. All other issues are 'could be bugs'...

Of this are **only 3 not related to ABL**:
#103 #133 #141 

IMHO none of them (3) is critical for a safe and stable usage of the machine.

So we could have a release stating:
* take care for the known bugs #103 #133 #141 
* ABL is an experimental feature with known bugs, use with care and if you know what you are doing.
* For additional information see `https://github.com/MarlinFirmware/MarlinDev/labels/Verified%20bug!`

@Wackerbarth What do you think?

fmalpartida, 2015-09-08:
Couldn't agree more.

Wackerbarth, 2015-09-08:
Just for clarification.
I am suggesting that we freeze any and all changes that affect the "externals" of the build.
In particular, anything that changes the Configuration files, M- & G- codes handled, etc.

We produce a Release Candidate and let others start building against that.

As they find problems such as compilation errors for a particular configuration, we fix those, both in the ReleaseCandidate and, at an appropriate time, porting those corrections into the new development effort.
However, we do not add ANY additional features and, as others have alluded, we may find it appropriate to say "don't do that with this release" to any conditions which we cannot fix with only minimal changes to the code.

At the same time, we also pause on the development side to rearrange the files into a structure that will allow us to make the code more modular, including a hierarchy of header includes (basically, we turn the code into a library called by a trivial main program rather than the current huge main program).

Although it isn't necessary, I suggest that we adopt, and enforce, a coding style that is supported by the `astyle` program. Doing so will eliminate the need for the frequent changes that Scott has made to tweak the spacing on some particular edit.  I suggest that it be done at this point because it will cause a massive change in form (as it corrects spacing errors, eliminates trailing white space, etc.) which is not related to substance.

Then, I would like to see us BEGIN moving code into generic functional units rather than the current "spaghetti" of conditional compilations.  The key point is that we define some modules and their interactions and avoid adding "features" as patches to the existing code base rather than making them a part of the appropriate module.

Roxy-3DPrintBoard, 2015-09-08:
> At the same time, we also pause on the development side to rearrange the files into a structure that will allow us to make the code more modular, including a hierarchy of header includes (basically, [...]

I like this idea.  But let's talk about it.    We are at the tail end of a 6 month bug fix phase.  I don't think anybody would have felt comfortable if they had known it was going to take 6 months.  

How much time will it take to modularize the code?   I'm guessing that is going to require a similar amount of time.  Something in the 6 month area.

Can a few other people offer an opinion on how long it will take to restructure the code?

Can we speed up the effort by compartmentalizing the code at a high level and then assign the subsections to volunteers to further clean up?  What I'm hoping is we can get the clean up effort broken up into small enough pieces that we don't have to eat the whole elephant in one sitting.

Wackerbarth, 2015-09-08:
@Roxy-3DPrintBoard -- Note that I said "BEGIN moving code". The really disruptive items are the repositioning of the code into a library, (hopefully) reformatting as needed to conform to some `astyle` and selecting our initial cut at defining some modules.  Those operations can be performed rather quickly.

I don't think that we need to "eat the whole elephant" by any means.  However, we do need to identify areas where a more generic solution, then customized, is likely to improve things and avoid adding new patches that violate those interfaces.

A good example of this is the storage of "tuning adjustments". I believe that we can move from a single struct definition of the parameters to a key/value dictionary and thereby reduce the complexity involved in adding additional parameters (such as those needed for better kinematics models). So, rather than adding code to handle some particular set of parameters, and additional M-codes, we should specify a new interface and utilize it to add these parameters. Initially, this could be nothing more than an alternate interface and wrapper for the existing operations. Later, it can be enhanced. However, this activity is something which can be performed in a module separate from the main code and thus, with minimal interference, developed independently.

Roxy-3DPrintBoard, 2015-09-08:
OK!   And your example of "Tuning adjustments" really does bring the point home.   

As for your statement that the really disruptive items are the repositioning of the code into a library....

Have you given much thought to how many different libraries might make sense and what they might be?   Just hearing speculation on that question would help me understand things better.

Wackerbarth, 2015-09-08:
The "disruption" is, initially, that we introduce some changes to virtually every file and those who are modifying the present code will need to port their modifications to the new files. 

To get the desired flexibility, we need only one library. That library would have multiple source modules.

At some later time, we have the option to move some of these modules into separate libraries if that makes sense. However, we gain most of the value from the creation of modules, not libraries. It's just that we need the library structure to get around the way that the Arduino IDE behaves. (In the "library" world, things are just the standard C/C++ code structure that we find everywhere.)

jhsandell, 2015-09-08:
I'm currently working on a new machine kinematics, but its *nowhere* near mature for release. 
**So just go on with the freeze.**

My immediate observations to the DEV branch and suggestions to future improvements are:
- The code readability has improved a lot with the ENABLED and DISABLED macros.
- Kinematics are a mess. Not all functions support both Cartesian, Delta, Scara, CoreXY ... and my Inchworm. The whole thing about transformations should go into a library of itself. I give a further description below.
- Can anybody make a diagram over the code structure. All these global variables and side-effect usage of them is giving me a headache (whining). At least describe the calling hierachy.

A model of kinematic transformations could be like going from "g-code cartesian space" through a kinematic transfer function to "rail position in mm" to a "step-position"
- The first should be tightly coupled to the main code, 
- the second in the planner.cpp, and 
- the third in the stepper.cpp (where it already is today).
With a stacking like this you could make subroutines and functions for supporting two-way communication between the modules.

The CoreXY should be taken out of the Planner.cpp.

Wackerbarth, 2015-09-08:
Jens, I fully agree. However, I don't understand why you think that we need three layers rather than just two.  Fundamentally, we need a function (and it should be a function common to all printer designs, but with alternate implementations for the distinct kinematics) that takes as input a parameter vector which describes the machine, including any "tuning" and a motion in the cartesian space of the object. The output is the necessary actuator values to perform the motion.  There may also need to be an inverse function which takes the actuator positions as input, and the same parameter vector, and provides the equivalent position in the object space.

Tracking of the location of a nozzle should be done in the actuator space because changes in position correspond directly to step pulses (or equivalent) delivered to the stepping motors. It also eliminates all of the compensations for the deployment and stowing of the probe, etc.

Changes to the parameter vector because of bed leveling calculations, don't really care where the nozzle is presently located. Those changes will be reflected in the next movement because the calculated actuator position will reflect the changed parameters.

jhsandell, 2015-09-09:
@Wackerbarth 
There is three coordinate spaces in the firmware today. : XYZ from the gcode in millimeters, ABC for the position along the axes in millimeters (just taking DELTA as an example), and at last the same postion in steps. 

You are right, when you say that we only need one function. Every mathematician can give you a series of matrix operations and then collapse them into one transfer function. But I assess that you would then have to build a lot of code around each transferfunction for each machine. 

If you keep the two transformations separated (mm(XYZ) -> mm(ABC) -> steps(ABC) ) then we can make motion, homing and ABL generic for all types of kinematics; regardless of the user choosing point, grid or mesh ABL.

By the way:
My Inchworm machine has a lot of challenges when reading the endstop collisions. I am a newbie to programming the MARLIN firmware, so I am missing (or more likely cannot find) the functions to expose each individual endstop pin/state. And I am also trying to make the XY homing do an autocalibration of the build volume. So I have a all hands full. :sweat: 
Have a peek at the mess I have made in marlin_main.cpp in the Homing subroutine. https://github.com/jhsandell/MarlinDev_Inchworm

jhsandell, 2015-09-09:
@Wackerbarth 
Please read my contributions for what they are. I am not trying to change everything or make a big mess. My project will move on as isolated as possible, and when it reaches an acceptable maturity level, I will incorporate it into the current DEV baseline and make a pull request. 
Please give me a heads up, if there is any spill-over into the mainstream.

Wackerbarth, 2015-09-09:
Jens,

I appreciate your comments and totally agree with the idea of attempting to unify the kinematics.

However, I still fail to see the value in separating the mm(ABC) and steps(ABC). Where do you suggest that we NEED to access mm(ABC)? As far as I can tell, the units for each axis are arbitrary. You can choose them however you wish. Although it may be “friendly” for the user to specify distances in mm rather than steps, the unit conversion can be performed on input (either at compile time or as a function of some command that affects a setting.) However, the actual control of the machine need not go through this step during the kinematic transformation.

I would also like to point out that I believe that you, as well as others are unduly restricting the transformations when you speak of them in terms of matrix operations.  Although it is easy to think in terms of matrix multiplication for linear transformations, and it is certainly easier to determine inverse transformations when you restrict them to the linear ones, the underlying concepts equally apply to a larger class of transformations. In fact, the kinematics of delta printers are a good example of the use of a non-linear mapping.

In looking at your repository, I have a couple of suggestions. First, because your repository is closely tied to the MarlinDev repository, rather than doing your development on “master”, I suggest that you name your branch something like “inchworm”. That way, there will not be conflicts in the branch names when you wish to move material from one repository to another.

On the issue of end stops, I didn’t quite follow. Do I understand that you have three sensors that operate along the axis of your AB carriages? (left limit, right limit, and minimal carriage separation)?

One of the problems we encounter is the ambiguity in the use of the term “homing”. It can mean an operation to establish the location of a carriage in relation to the frame, or it is also used to designate movement to some pre-defined position.

For the first operation, you need one sensor per degree of freedom.

You might find it easier to do the math if you change the way that you approach the operation.

First, to avoid issues with collision with the bed, I would raise the C carriage to its maximum position. This establishes the location of that carriage.
Then, in concert, I would move both carriages to the left until the left end stop is reached. This establishes the location of the left carriage.
Finally, I would move the two carriages toward each other until the minimum distance is achieved. At this point, the location of the right carriage can be set because it is at a defined amount greater than the location of the left carriage.

Although it isn’t needed for homing, the position of right end stop can be determined as a part of the calibration sequence once you have established the locations for the carriages.

Richard

jhsandell, 2015-09-09:
> ...that you, as well as others are unduly restricting the transformations when you speak of them in terms of matrix operations...

Apologies, I made an unnecessary generalisation. :sweat_smile: To many years have gone by since  I took the "Advanced Control Algorithms" course. (open loop, closed loop, polar plots, laplace transform, etc. not longing for it.)

I actually assumed that getting rid of one of the tranformations would be too big a mouthfull for the community. Again, apologies. Lets not press the case further.

Thanks, for the advice on changing the branch name. I will look around and try to change it.

Cheers.

Wackerbarth, 2015-09-09:
"Too many years have gone by since I took" -- I hear that! For me, this came from my graduate course in "Linear Algebra", where we started from the fundamentals of relations and spent at least half of the semester before we got to anything which you and I would call "linear". But that was, literally, half a century ago. I've forgotten many of the nuances.

thinkyhead, 2015-09-12:
I have a number of bug fixes pending in the PR queue. Hopefully some of these can be merged for the RC.

Wackerbarth, 2015-09-12:
I incorporated most of Scott's changes into #166. Please evaluate it for both content (some feature we really NEED to include in the candidate) and functionality (does it compile/run acceptably) to be "good enough" to call it our 1.1 Release Candidate.

Also, we need to address the accompanying documentation.

shampine1, 2015-09-13:
I'm trying out the latest commit.  On the GREAT side is that my REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER works!  Thanks SOOOO much!
I like being able to tune feed rate on the home screen; gold star!
I'm running a print tethered right now, and things look good.

On the bad side, my SD card doesn't show up on the menu at all.

closed 2015-09-28 by Wackerbarth

KiteLab, 2015-09-28:
@shampine1 
Please activate 
#define SDSUPPORT
explicit..