MarlinFirmware/MarlinDev #156: Reprap discount smart controller doesn't compile with 2 extruders

shampine1, 2015-09-10:
I have commit A35669982
I am compiling with Arduino 1.0.6, and I have the u8glib in the libraries area.
With these configuration changes it compiles:
#define MOTHERBOARD BOARD_RAMPS_13_EFB
#define CUSTOM_MACHINE_NAME "WeldingRodBot"
#define EXTRUDERS 1
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 1
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1
#define DEFAULT_AXIS_STEPS_PER_UNIT   {160,160,800,191} 
#define REPRAP_DISCOUNT_SMART_CONTROLLER

When I set it to 2 extruders it fails:
This line is high lighted:
   MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_e);
and I get a bajillion lines of errors, the last few of which are:

ultralcd.cpp:1301: warning: 'void menu_action_setting_edit_long5(const char*, long unsigned int*, long unsigned int, long unsigned int)' defined but not used
ultralcd.cpp:1294: warning: 'void menu_action_setting_edit_callback_int3(const char*, int*, int, int, void (*)())' defined but not used
ultralcd.cpp:1295: warning: 'void menu_action_setting_edit_callback_float3(const char*, float*, float, float, void (*)())' defined but not used
ultralcd.cpp:1296: warning: 'void menu_action_setting_edit_callback_float32(const char*, float*, float, float, void (*)())' defined but not used
ultralcd.cpp:1298: warning: 'void menu_action_setting_edit_callback_float5(const char*, float*, float, float, void (*)())' defined but not used
ultralcd.cpp:1299: warning: 'void menu_action_setting_edit_callback_float51(const char*, float*, float, float, void (*)())' defined but not used

clefranc, 2015-09-10:
@shampine1 There is already a PR https://github.com/MarlinFirmware/MarlinDev/pull/154 to resolve this. Waiting for a good soul to merge it.

shampine1, 2015-09-10:
Sorry!  I searched for the pull request I created and forgot to search when I made this issue!

closed 2015-09-10 by shampine1