MarlinFirmware/MarlinDev #157: LCD_UPDATE_INTERVAL in Ultralcd seams to be troublesome

justmyopinion, 2015-09-10:
I am testing my Azteeg x3 based printer with latest dev download but i cannot have smooth printing with the default setting of LCD_UPDATE_INTERVAL = 100 in Ultralcd.h.
I guess it is somehow the update rate (100ms) of LCD (mine is Viki).
if i change this value to something higher 200-500 my problem with stuttering printing disappears.
This value was not used in earlier versions like rel 1.02.
Could this value be far too optimistic selected. I assume this is a general problem.

clefranc, 2015-09-10:
@justmyopinion I presume you're using a `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`?

There isn't much you can do. Increasing the LCD refresh can become annoying when navigating in menu. I've added a switch in my [fork](https://github.com/clefranc/Marlin4Delta/tree/optimise_sps) to disable the automatic return of the status screen (where updates causes stuttering). A small workaround for deltas pushing the limit of 8bit controllers.

See also https://github.com/MarlinFirmware/Marlin/issues/1739 with suggestion from @thinkyhead to set Arduino optimization to -O3, it helps.

https://github.com/MarlinFirmware/Marlin/issues/1404

justmyopinion, 2015-09-10:
@clefranc Thankyou for your attention.
I do not have any graphics LCD just a plain 4 lines LCD (LCD_I2C_VIKI).
I made an upgrade to Arduino 1.65 and downloaded dev version and the result was a degraded version. i do not run very fast 20-30 mm/s on a standard XY-printer and do not make any playing with LCD buttons and yet it is stuttering mainly during printing round corners (arcs).
I made it look normal by changing LCD_UPDATE_INTERVAL to 200 or more but it is not very comfortable that resources suddenly has become low without knowing why. Motherboard is 2560 (Azteeg X3)

justmyopinion, 2015-09-10:
I have checked that ver 1.02 rel does not have this problem so some extension to menu updating must have been introduced.since 1.02 taking resources from cpu. A first look indicates that x,y positions has been added which is really a novalue addition as it only has display value for static positions due to update rate.
Is it possible to have a clue of where to look for these additions to display, I assume no documentation is present?

o-lampe, 2015-09-11:
I agree with @justmyopinion . The XYZ-coordinates are obsolete info while printing.
Can´t we make the refresh ratio dynamic with low priority? I better like to have blobfree prints, than smooth LCD navigation.
Just my 2 cents
-Olaf

ntoff, 2015-09-12:
Exactly. I've said it before and I'll say it again. I would rather have the X and Y info just not be there. The Z axis height is useful for attempting to resume a failed print later on by figuring out which gcode line it failed at by comparing the z height but X and Y are useless.

Yes, it's useful while moving the printer around by hand, but while printing? There's no point to it at all.

thinkyhead, 2015-09-13:
While it would be cool to hide XY during printing, this could currently only be available for SD printing. Marlin can't currently tell that it's doing a print job when a host is doing the printing.

ntoff, 2015-09-13:
sounds like a job for bidirectional communications in Marlin 2.0

clefranc, 2015-09-13:
Perhaps a new start/end g-code for enabling/disabling XY (even Z) during host printing?

justmyopinion, 2015-09-13:
@clefranc
Sounds like a good idea.
I have no evidence that XY updating is the reason for missing resources during print, It may need some investigation to find out, but I find it worrying that the problems with graphic display update rate is also now a problem with simple char displays with current code.

justmyopinion, 2015-09-13:
I made one more test with LCD updating.
I enabled two extruders which generates a different status LCD.. 
XY pos updating is replaced with second extruder temperature which maybe has a different load to updating.
However result was the same: stuttering printer if LCD_UPDATE is 100 or less and no stuttering if update is 150-200 or more.
It generates a feeling that problem may be hided elsewhere, but where?.

fmalpartida, 2015-09-13:
This one is funny, I've been trying with my SR LCD which uses the New LiquidCrystal library as opposed to the stock LCD driver and haven't stumbled with this issue. Having said this, the library is 4x faster than the stock one in the SR configuration.

justmyopinion, 2015-09-15:
I find it hard to believe that I am the only one having this problem
i have a fairly standard setup to my FW running Viki I2C display with no SD card enabled.
problems are very easy to generate , I just need to do some curved printing and the stuttering is present as explained and is as easily removed by lowering LCD_UPDATE:.

o-lampe, 2015-09-16:
@justmyopinion  Maybe the I2C display is blocking the ISR, messing up the timing?

justmyopinion, 2015-09-16:
I took a few measures to get some feedback and it is a bit worrying.
If updatefrequency is set to 10 , 100ms is available per display update. The measure I took was for status line only (last line of 4x20) and it tells me that last line need 20ms to update and for 4 lines this would be around 80ms out of hundred :-( so maybe it is understandable that printing will be starved with these conditions or how should I interpret this? What happens if 100ms update is exceeded?
are all LCD's that slow?

Roxy-3DPrintBoard, 2015-09-16:
The ISR's do eat up a lot of the processor time.  But I'm kind of skeptical the LCD Panel Updates are eating 99% of the processor.   How are you measuring this time?   It might be good to toggle a GPIO port at the start of the LCD Panel ISR code and then clear it at the end.  That way it would be very clear with an oscilloscope EXACTLY how much the ISR code is using.

justmyopinion, 2015-09-16:
I made a GPIO signal at start and end of lcd_implementation_status_screen() and triggered a scope and the result is as above 99ms (app) out of default 100.
I could do the same with ISR but nothing but SPI is stopping interrupt so it should not interfere also lcd_update may a bit suspecious as idle turnaround is app 100us and no resouces needs an update rate close to this.

fmalpartida, 2015-09-16:
One comment, at best most I2C panels using the standard Arduino wire library can only get a refresh rate of about 13FPS on a 20x4 display. 

Wire also use interrupts and nesting them is not well looked after.

justmyopinion, 2015-09-16:
@fmalpartida 
Thankyou for supplying info of refresh rates.
i really do not hope we have nested interrupts, that could really be a nasty thing if not taken care of.

justmyopinion, 2015-09-17:
I am still trying to get into understanding of this problem.
I tried to look into cmd buffer but buffer never gets empty when problem is present more likely it seems to be full so processor is behind somehow(3).
Two parameters have influence to provoke the problem:
1: if LCD_UPDATE is 100 or lower the problem is present and disappear if update frequency is lowered (higher LCD_UPDATE >> 100)
2: if print speed is lowered same thing happens, problem disappear if speed is lowered
So still processor resources seems to be lacking if update frequency is too high, but I cannot explain how this is happening.
Any suggestions to narrow this problem further?

Roxy-3DPrintBoard, 2015-09-17:
> This is really no good and needs another implementation for sure, maybe cmd reading should also be interrupt based?.

The whole purpose of having a #define'able LCD_UPDATE value is so it can be set at a number that makes sense.   Why doesn't it just make sense to lower the number?  And for sure, the LCD code could be modified to be more efficient.  For example, it probably does not need to update each line every time.  Some lines are more important than other lines.   

I did give some thought to making the firmware be able to measure how much time is being spent on different items.   But it is enough work that the time would be better spent some where else.   Even with good, accurate measurements, it doesn't actually fix the problem.  It just gives you more information about it.

fmalpartida, 2015-09-17:
If you are interested in the performance and bench marking:
https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home#!performance-and-benchmakrs <https://bitbucket.org/fmalpartida/new-liquidcrystal/wiki/Home#!performance-and-benchmakrs>

And divide the fps by 2.5 to get the performance out of an I2C LCD.

I use the SR_2W in my SAV 3D LCD and haven’t experienced any problems.

I2C devices (specially slow ones) are particularly annoying as the use ISRs, to make things get worse on “data intensive” devices such
as LCDs and the fact that we share the only preemtive mechanism we have (the timer interrupt).

For the time being, I do see very fit to use the LCD_UPDATE “feature” to cater for slow LCDs.

Looking into the future the I2C interface should be revisited for speed. Its on my ToDo things for the New LiquidCrystal library but here
the first thing that should happen is to start putting HAL in place.

justmyopinion, 2015-09-17:
"Why doesn't it just make sense to lower the number?" 
The number must be increased to do any help as verified :-).

Well, increasing LCD_UPDATE time removes the symptoms but is no real cure as different hw and conditions influence the result. How do you think Graphic LCD's are performing? But for a start it certainly helps.

Measurements of performance are vital for tuning of FW and for changing vital coding to the better.
System parameters should be validated to understand where resources are best used.
As input, cmds are feeded with a speed of 25kbytes ore maybe even more depending on interface and in the other end printer is trying to execute cmds with a speed that could be close to input. Any processing in between these two ends take resources from this feedtrough.
Take an example of a command for moving one axe with 100mm/s. in 100ms this would be about 10mm.If display processing steels 100ms for updating it would mean that if cmd only takes 50ms to execute 5mm, it had to wait 50ms to get the next cmd for the next 5mm. That is what i would call "stuttering" and is typical behaveour for curved movements. 
So you are right it is necessary to make measurements to fully understand how to avoid bottlenecks.
For now, changing LCD_UPDATE would do it for me, but on the long run i would prefer to see a better solution.

Roxy-3DPrintBoard, 2015-09-17:
> Looking into the future the I2C interface should be revisited for speed. Its on my ToDo things for the New LiquidCrystal library but here the first thing that should happen is to start putting HAL in place.

Sort of along those lines...  If a HAL was actually in place, it would not be very difficult to put a Real Time OS  on top of it.  And then these kind of issues just go away.  The LCD Update could be partitioned as a fairly low priority task and as long as something more important was not demanding the processor, it could consume 100% of it.   But...   That would be almost a total re-write of the code.

fmalpartida, 2015-09-17:
Spot on.

justmyopinion, 2015-09-17:
What is needed is that reading cMD buffer should have higher priority than display. If sw interrupt were available it could be issued when a cmd had executed and let intr get next command from buffer. As this is not possible it could may be be embedded in one of the existing ISR timer routine with a little overhead asking if any cmd in buffer and then flagging this overriding display updating. but still display update needs to be revised to more efficiency. Just some thoughts.

Roxy-3DPrintBoard, 2015-09-17:
> What is needed is that reading cMD buffer should have higher priority than display. 

Please help me understand.   I'm thinking it is more difficult than this.  The reason is the USB Serial interface (from Windows) is always trying to jam characters at the 'serial' line.  It tries really hard to keep the buffer full.  Perhaps what you are saying is:   As long as the command buffer has enough stuff in it, instead of 'Busy Waiting', use the time to update the LCD Information.  But once again, you run into a race condition (at least without a re-write) because some commands could fill the buffer but take almost no time to execute.

justmyopinion, 2015-09-17:
Correct, with existing code it will always be a race condition and hard to prove safe. I think it is vital to improve, as stuttering really affects quality of printing in a negative way as extruder is stopped brutally.
With the test run i made with an example of curved printing I tested cmd buffer echoing how many cmds in buffer when cmd was read during print and it never got empty but had always 2-3 commands waiting so host serial connection has no problems of keeping up a full buffer. It is printing that is the botleneck and short command executing that makes the problems here.

Wackerbarth, 2015-09-17:
Unfortunately, at some point, the "best" solution is KISS. (Quit trying to do everything with such a small processor) If you really need to do so much, get a faster/more complex processor that can handle the workload and quit trying to make your silk purse from a sow's ear.

Roxy-3DPrintBoard, 2015-09-17:
> Unfortunately, at some point, the "best" solution is KISS. (Quit trying to do everything with such a small processor) 

Yeah....  I was just looking through the LCD code.   There is a fair amount of busy work going on there.  Everything from plotting lines from one coordinate to another to formatting floating point numbers.  No wonder the LCD Code is so big and slow.

There certainly would be an opportunity to shrink the complexity of what the LCD displays.   That would free up a lot of processor cycles and code space.

justmyopinion, 2015-09-17:
There is still a lot of resources hidden in this "small" processor but it may take some work to find them and get to the abuse made somewhere.
Maybe it is time to make a reevaluation of resources used in the programming of Marlin?
I have some personal experience that influence my thinking that may not cover any aspect.
1: personaly i do not think printer display needs updating more than once per sec (?) it is not a CNC
2: buttons should typically be debounced in the area of 200ms and not repeated more than 3-5 times/s and therfore seldom sampled.
3: flashing leds and animated displays should not be allowed in a serious printer
these demands hardly needs any complex processor just some good thinking when programming.
a lot of things in Marlin could be evaluated for resources. PID is an exampl which neither needs updating but a few times/sec with the time constants of extruder/bed in mind ......zzzzzz

Roxy-3DPrintBoard, 2015-09-18:
on #3: If the machine is paused, we can burn the processor to do pretty lights and things that make the user feel good!   :stuck_out_tongue_closed_eyes:

justmyopinion, 2015-09-18:
Today I made a short hack to get around this display problem and it seems to work fine.
In lcd_implementation_status where status is printed to display i split the update in 4 lines (i have a four line 20x4 display) . i established a counter of 4 and for every update I increase the counter to next line.So it takes 4 updates to print whole display. In this way one line is using 25 ms each and the rest of  LCD_UPDATE time can be used for processing. This takes the load out of processing and I have tested it down to LCD_UPDATE values down to 50 ms and it works fine even with very small circles at high speeds. 
It is not rocket science but it works :-)

Roxy-3DPrintBoard, 2015-09-18:
At the front end of lcd_implementation_status_screen(), it draws all the boxes and lines every time through the code.   You might try calling that a 'line' and only have it draw the graphics stuff once in a while.   If it works like I think it does, that would chop the amount of time it takes as it does each of your 'lines'.

justmyopinion, 2015-09-18:
No "graphics stuff" here, it is a simple 20x4 char display....graphics displays have their own problems
 :-(
Seems to be a bit of "crosstalk" here...Maybe it is time to close the issue as apparently no one seems to care anyway.

o-lampe, 2015-09-18:
@justmyopinion 
Sure there is graphic in the 2004 lcd. The thermometer symbol and probably the bed symbol are non standard ASCII.
I´m very interested in code- improvement for deltas with Ramps/Mega and 2004 LCD.
As you wrote, the split refresh is no rocket science, Just provide the code you wrote and give´m time to test and implement it one day.

closed 2015-09-18 by justmyopinion