MarlinFirmware/MarlinDev #162: FIX: Homing after MANUAL_BED_LEVELING issue.

drbogger, 2015-09-11:
I have been testing the latest MarlinDev (master) firmware, and I found an issue related to homing (G28) that only occurred after doing the "Manual / Mesh Bed Leveling" process.

**PROBLEM:**
When homing (G28), it would home to the endstops, then it would move the X & Y to 10 mm, which was displayed on the LCD correctly, but not in Repetier Host, which said the X & Y were at 0.

Secondly, the LCD stated that the Z-Axis was at 4mm, when it clearly was still at 0mm.

**SOLUTION:**
I modified the code to fix the problem. Below is the code snippets.

**FILE: Marlin_main.cpp**

**REPLACE:**

```
  // For manual leveling move back to 0,0
  #if ENABLED(MESH_BED_LEVELING)
    if (mbl_was_active) {
      current_position[X_AXIS] = mbl.get_x(0);
      current_position[Y_AXIS] = mbl.get_y(0);
      set_destination_to_current();
      feedrate = homing_feedrate[X_AXIS];
      line_to_destination();
      st_synchronize();
      current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
      sync_plan_position();
      mbl.active = 1;
      #if ENABLED(DEBUG_LEVELING_FEATURE)
        if (marlin_debug_flags & DEBUG_LEVELING) {
          print_xyz("mbl_was_active > current_position", current_position);
        }
      #endif
    }
  #endif
```

**WITH:**

```
  // For manual leveling move back to 0,0
  #if ENABLED(MESH_BED_LEVELING)
    if (mbl_was_active) {
      current_position[X_AXIS] = mbl.get_x(0);
      current_position[Y_AXIS] = mbl.get_y(0);
      set_destination_to_current();
      feedrate = homing_feedrate[X_AXIS];
      line_to_destination();
      st_synchronize();
      current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
      set_destination_to_current();
      feedrate = homing_feedrate[Z_AXIS];
      line_to_destination();
      st_synchronize();
      sync_plan_position();
      mbl.active = 1;
      #if ENABLED(DEBUG_LEVELING_FEATURE)
        if (marlin_debug_flags & DEBUG_LEVELING) {
          print_xyz("mbl_was_active > current_position", current_position);
        }
      #endif
    }
  #endif
  ```
  
  
**FILE: Marlin_main.cpp**

  **REPLACE:**

        case 28: // G28: Home all axes, one at a time
        gcode_G28();
        break;
  
  
  **WITH:**

        case 28: // G28: Home all axes, one at a time
        gcode_G28();
        gcode_M114(); // Added to send current position to host.
        break;

eboston, 2015-09-11:
Why?  If the user needs the position, they can just send the M114 
command.  No need to force it on everyone whether they need it or not.

drbogger, 2015-09-11:
Because when printing from Repetier Host, when it homes, it thinks its at 0 position, but when the print starts, it starts 10mm past where it should.

eboston, 2015-09-11:
So how does sending the position back when you home effect anything?

drbogger, 2015-09-11:
It lets repetier host know where it's at.

eboston, 2015-09-11:
but you already said that Repeiter host thinks it at 0,0 so all you would be doing is telling it what it thinks.

drbogger, 2015-09-11:
Your wrong. With manual_bed_leveling enabled, it homes, then immediately after it homes, it moves X & Y +10mm. But Repetier Host still thinks its at 0,0, when its actually at 10,10.

eboston, 2015-09-12:
You stated, "When homing (G28), it would home to the endstops, then it would move the X & Y to 10 mm, which was displayed on the LCD correctly, but not in Repetier Host, which said the X & Y were at 0."  So at this point, Repetier thinks it is at 0,0 when the hot end is actually at 10, 10.

Your fix to manual leveling moves the hot end back to 0,0 where Repetier thinks it is, correct?  Or is it just resetting the home position from 0,0 to 10,10 in which case, that is totally wrong.

drbogger, 2015-09-12:
it's just resetting  the home position to 10,10... but can you tell me how that is wrong?
Should I make the fix move back to 0,0 instead?

eboston, 2015-09-12:
G28 is suppose to leave the hot end the home position, not somewhere else.  Normally this is at the min or max of the axis.  If G28 puts the hot end in one place before doing the manual bed leveling, and in a different place after, there is something wrong.  G28 should always put the hot end at the home location.  From looking at the Configuration.h file, it looks like it is leaving the hot end at one of the test points since by default, it tests 10mm in from each end of the axis.

Unless I am totally misunderstand how Manual/Mesh bed leveling is suppose to work.

drbogger, 2015-09-12:
You are correct. G28 puts the hotend at the first test point. The code I posted above is what is causing it. Any idea why that code is even there to begin with? I am not very familiar with the coding of any type of auto leveling (or manual leveling in this case).

eboston, 2015-09-12:
I'm not that familiar with the actual code so can't really help you there.

drbogger, 2015-09-12:
Okay, I'll just leave it as a bug report for now. If I find a fix, I'll post it.

thinkyhead, 2015-09-13:
@epatel Is this a bug? It kind of looks like it. At the end of manual leveling there's a `G28` command to re-home the printer. Then we have custom code (as shown above) within `gcode_G28` to set XY to the first mesh point, and Z to `MESH_HOME_SEARCH_Z`. But this code will actually affect homing whenever `mbl.active` is set to 1, not just during the leveling process. So it looks like this code just doesn't belong in `G28` and maybe some of it should be in your custom `G29` instead.

epatel, 2015-09-13:
@thinkyhead I am happy to see someone actually testing it! :)

No, it is very intentional. The rational is that `G28` is made when one do not know the real position of the hotend. So while doing the `G28` it turns off the mesh calculations, then move to the closest best known measured position and turn the mesh calculations on. These steps I actually copied from another guy that did a manual bed levelling feature a couple years ago (doing it without the mesh that I added).

If its a big problem I can see if I can calculate the levelled `Z` for `[0,0]` and turn it on while in the homing position.

thinkyhead, 2015-09-13:
@epatel Hmm, well you can see why moving the nozzle at the end of a `G28` operation is problematic. In this case Repetier Host is actually being presumptuous (since homing doesn't leave the position at 0,0,0 — it leaves the position set to the home position plus offsets) so that's not the main issue. The fact that the current Z position is being set to `MESH_HOME_SEARCH_Z` at the end of `G28` (when the nozzle is actually touching the bed, or may even be at the other end with a Z_MAX endstop) does seem to be a mistake.

You will note that for Automatic Bed Leveling there is a different behavior: `G28` completely turns off the leveling compensation. Also for Automatic Bed Leveling, no homing is done at the end. It just stays where it ends up after it puts the probe away, and then the very last thing that Automatic Bed Leveling `G29` does is to apply the rotation matrix to the current (raw, uncorrected) position so it becomes corrected. We might want to do that with MBL as well, to make it behave consistently with the way that ABL works. Since your MBL implementation saves its results it can be turned on/off any time. So you can still do a `G28` at the end (turning off MBL) but then enable it afterward, which will immediately apply the Z correction.

epatel, 2015-09-13:
@thinkyhead If one has followed the instructions I wrote for the MBL the hotend should not touch the bed at `[0,0,0]`. This is a safety measure so one don't regularly run the hotend into the bed.

I hope you don't mean one should do the measuring every time after a `G28`? This is not automatic so its a little tedious. I maybe do it once per month or so (values being saved in EEPROM between usage).

Auto homing after a measure session is mainly for convenience. Sure the activation of MBL in `G28` can be separated but feels unnecessary as I guess it should be done every time and seperating it opens for forgetting it and can failed prints.

drbogger, 2015-09-13:
The problem I was having, is that after a G28, it offsets the nozzle to 10,10, and it throws off repetier host, because it's assuming the nozzle is at 0,0.

Why exactly must the nozzle move after a G28? I'd much rather it stay at 0,0,0

epatel, 2015-09-13:
@drbogger `G28` travels to the end stops. Having `X=0` and `Y=0` from there is fine but it isn't necessary for `Z` to be touching the bed at that point. Having the end stop for `Z` to be slightly above the bed makes it easy to setup and calibrate.

The movement to the closest best known measured point is to move to a known measured point and turn on the levelling calculations, it is a position where all the values are known. We have just reached the end stop and we have a measured point, activating levelling there is easy and safe. I could activate the levelling calculations at the end stops, but to do that I (think) will need to add a reverse calculation at `[0,0]` from the mesh, doable but will need testing etc. I can look into it, maybe add a final move to `[0,0,0]`.

drbogger, 2015-09-13:
Adding a final move to '''' 0,0,0 '''' is fine by me. My biggest problem is it screws with repetier host thinking it's at '''' 0,0 '''' when in reality it's at '''' 10,10 ''''

thinkyhead, 2015-09-14:
> I hope you don't mean one should do the measuring every time after a G28?

No that would be annoying. However, when MBL is turned on it shouldn't continue to have an effect on `G28`. I suggest that instead, if MBL is turned on during a `G28` turn it off temporarily, then just turn it back on when `G28` is all done. But without moving the nozzle away from `0,0,0`.

Meanwhile, it seems like a bug if Sean's machine thinks Z=4 when Z is actually touching the bed. It's this pair of lines at the end of `G28` that seems incorrect, but correct me if I'm misunderstanding. (I haven't actually tested MBL on my machine, but I guess I should!):
```cpp
current_position[Z_AXIS] = MESH_HOME_SEARCH_Z;
sync_plan_position();
```

thinkyhead, 2015-09-14:
> BuildTak

I've not tried it, but I will in the future!

drbogger, 2015-09-17:
So what is the suggested fix for this? Could this code be removed all together? Or is it required for something specific?

epatel, 2015-09-17:
I guess the suggested fix is to calculate the real Z value at `[0.0]`. Activate the calculation and move to `Z=0` at the end of a `G28`. I'll see if I can add (and test) it this weekend.

epatel, 2015-09-17:
Yeah, the code is needed at the moment. It's about turning on the calculations at a known point.

drbogger, 2015-09-17:
Okay, I'll wait for your update.

epatel, 2015-09-19:
Just created a PR. Actually this thing was easier than I thought. Hope it solves your problem.

nitewing76, 2015-11-18:
@drbogger In the Repitier Host Printer Settings--->Printer Shape....is your X-Min, Y-Min, Bed Left, and Bed Front all set to 0? Having X/Y-Min=0 and Bed Front/Left=10 will cause the behavior.

This setting is confusing....because it has nothing to do with the dimensions of the structure supporting the bed, but the relative position of the nozzle to the bed after "G28 X Y".

I am assuming this discussion about Z=0, is only before the nozzle begins to heat up.  If not, then this a bad idea because in addition to the safety factor, there will be an transfer of heat from the nozzle to the bed. This means in the area where the nozzle is touching the bed, the bed will physically expand...then contract as that area of the bed cools while printing. Slightly changing the topology of the bed...the degree of which depends on the material of the bed and the coefficient of thermal expansion.

thinkyhead, 2016-04-22:
I presume this issue is long fixed by now…?

jbrazio, 2016-04-22:
@thinkyhead I would suggest to define a threshold and close "without questions" all issues not updated and not feature requests in the last couple of months. The code has change so much that they have lost their importance.

thinkyhead, 2016-04-23:
I suppose you are right, João. Life is too short…

closed 2016-04-23 by thinkyhead

brainscan, 2016-05-17:
@epatel just wondering if I undo these changes will mbl work again? Sorry Git really confuses me so I'm trying to work out how to put mbl back to its original state.

epatel, 2016-05-17:
@brainscan Just for you I can see if I later today can setup my system again to do this fix for you. And you will be my tester, is that OK?

brainscan, 2016-05-17:
That would be amazing!! Thank you so much. Happy to test, pretty sure if it can be reverted to the original code it will work perfectly, as it did from the start.

epatel, 2016-05-17:
@brainscan Please try this https://github.com/epatel/Marlin/tree/marlin/g28_all_restpos

I added a new option `MESH_G28_REST_ORIGIN` which is off by default. Enabled it will move the nozzle to `Z` = 0 as requested by @spaelectronics 

Could you please test both cases that if enabled moves to Z=0 and not when off.

I based this on a commit before https://github.com/MarlinFirmware/Marlin/commit/bc5a547d55722a87d41d2e4ee6ad32ed9db736ba as I believe that commit broke MBL.

gddeen, 2016-05-18:
Hmmm. When I do a 

    G29 S4 Z-0.3
    G28                 ; Homes XY, Probes [100,100,Z], Resets Z=Z-Probeoffsetfrom Nozzle-0.3,Homes Z, Sets Z=0

- Will your change add that BedZ (-0.3 in the case above) since there won't be a probe? I would assume not. (A few dozen G28s in a row wouldn't be too good.)  So, does that mean BEDZ can never be applied?

- Is there a G28 option to use the probe?

- Will M401 and M402 be in there??? (Raise/Lower the probe.)

gddeen, 2016-05-18:
Never mind, I guess G28 will still do the probe. (A prior mesh bed level had G28 with no probe.)

brainscan, 2016-05-18:
@epatel thank you, I will test tonight and report back.

brainscan, 2016-05-19:
@epatel Success!! Works perfectly for both cases. The 0,0,0 setting seems to work fine for those that would want/need it that way. The original version works great and just as expected. 

Thank you so much!! I've not made a PR before so I will have to check out what to do.

thinkyhead, 2016-05-19:
> If you want you may fork and make a PR for this into RCBugFix.

I can take care of that, no problem. If any issues introduced by https://github.com/MarlinFirmware/Marlin/commit/bc5a547d55722a87d41d2e4ee6ad32ed9db736ba still remain, this will be a good chance to debug them, as I will have access my 3D printer for a few hours.