MarlinFirmware/MarlinDev #169: Fashion Designer Needed

Wackerbarth, 2015-09-12:
With the approach of fall, it's time for a new wardrobe. So, what's going to be in style this season?

Actually, the issue is that we need to change from an ad hoc manual formatting procedure to an automated "pretty printer" that can handle all of these spacing, indention, bracket style issues rather than allowing code that does not conform to the standard to be merged into our code base.

There is a fine program for the purpose, Artistic Style [http://astyle.sourceforge.net/astyle.html].

The only problem is that I haven't figured out how to make it accept the exact style that has been in recent use here. And, frankly, there are a few things about our current style which I feel fail to really help with readability.

As a result, I'd rather make some minor changes to the style in order to gain the consistency provided by automatic checking which we can add to our integration testing.

But the problem is to select a style. There are many reasonable styles.

For example,
```
--style=google

--indent=spaces=2
--indent-preproc-block
--indent-preproc-define
--indent-col1-comments

--keep-one-line-blocks
--remove-brackets 
```
produces reasonable formatting.

But, I not that particular about the actual style, as long as we have, and follow SOME standard.

Suggestions for alternate style definitions are requested.
Hopefully we can pick a style that is acceptable to most of the developers and users.

eboston, 2015-09-12:
Personally, I prefer what they call allman/bsd/break style.  The style that is currently used, putting the bracket at the end of the line, makes seeing the block structure harder.  When the previous and following line are long, the bracket gets lost in the clutter.  If you want to see where putting the bracket at the end of the line gets real messy, try looking at some of the OpenSCAD files people produce.  Scary.

But I guess if we used an automatic formatter, then we could use whatever style we like and it will get changed when merging.

fmalpartida, 2015-09-12:
+1 on the Allman/bsd style I find it a lot clearer too.

Wackerbarth, 2015-09-12:
"But I guess if we used an automatic formatter, then we could use whatever style we like" -- But, the important thing is "WE LIKE". Although i"m sure everyone and "live with" any (reasonable) format, and you can always translate between styles, we don't want to be using something that is unpopular.

More discussion requested!

Roxy-3DPrintBoard, 2015-09-12:
>  The style that is currently used, putting the bracket at the end of the line, makes seeing the block structure harder.

I definitely prefer the opening brace be on the same line as the corresponding if/for/while/do.   Otherwise I lose a line of information per block.  I can live with what ever makes sense.   But I like the opening brace on the same line!

eboston, 2015-09-12:
How do you feel are you losing a line of information?

Wackerbarth, 2015-09-12:
Thanks  for all of the responses.  Let's try to make this a bit more concrete.  I ask that each of you, and anyone else interested,  examine the multitude of options available in astyle and explicitly present the selection of options that you would prefer.

Roxy-3DPrintBoard, 2015-09-12:
Because the opening brace is sitting on a line all by itself.   I much prefer K&R style where the opening brace is up on the line that starts the block.  I can see the whole block and match it to what ever started the block because of the indenting of the block.  

By wasting a line for a single brace, I can see less on the scree.

Roxy-3DPrintBoard, 2015-09-12:
style=kr   will work for me.

eboston, 2015-09-12:
I went through all the options, and it looks like these are the ones that I would vote for.

--style=bsd
--indent=spaces=2
--indent-preproc-block
--indent-preproc-define
--indent-col1-comments
--pad-oper
--pad-header
--delete-empty-lines
--align-pointer=type
--align-reference=type
--remove-brackets
--break-after-logical

And reasonable values for this...
--max-code-length=#

dposea, 2015-09-13:
I'm still trying to learn the code before contributing any changes. but if want my opinion I agree with the post above except I like empty lines , between methods or functions at least.

Wackerbarth, 2015-09-13:
@eoston -- Thanks for a concrete proposal. It provides an additional viewpoint for discussion. Perhaps you would comment on the differences that you consider significant and why you feel that your proposal is superior.

Please understand that I consider it a reasonable style, but would like to see us involve many in the discussion of the relative merits of the various styles.

eboston, 2015-09-13:
@Wackerbarth  All of this is really going to get down to personal style and likes.  For example, brackets and indentation.  Roxy prefers to have them on the same line so more lines of code can be displayed at once.  While that is fine for them, to me, it forces too many lines together in a smaller space, making it more difficult to read.  Back when I started programming and displays were smaller (anyone remember when an 80x24 line display was the hottest new option?), Getting more information visible was a concern because of the limited space.  But with today's displays, it is easy to get 10x that amount of data on the screen at once.  For me, it makes the code easier to read when it is spread out more.  Having the bracket on the line following the conditional sets the block apart.  Again, it is all personal preferences.  What I may see as a merit, others may see as a waste of space.  Doesn't mean either person is right or wrong.

We just need to pick a style and stick with it.  The options I have proposed are basically the style I have developed during my time doing code development.  It is what I like and feel comfortable with and thus why I proposed it.

Roxy-3DPrintBoard, 2015-09-13:
> @Wackerbarth All of this is really going to get down to personal style and likes. For example, brackets and indentation. Roxy prefers to have them on the same line so 

At a higher level....   It almost doesn't matter.   What Wakerbarth is doing is good.  He could pick the absolute worst combination of options and it would be better than what is there today.   My vote is in the morning he tells us what he has decided.

thinkyhead, 2015-09-13:
I've explained my reasoning for preferring the ["One True Brace"](https://en.wikipedia.org/wiki/Indent_style) style over having the opening brace on a separate line, and this is formalized in the [Coding Standards](http://www.marlinfirmware.org/index.php/Coding_Standards#Bracket-style) for some time now. I have a real problem with automatic formatters, especially for code that teams are working on. It allows more flexibility and the use of whitespace and indentation for emphasis.

fmalpartida, 2015-09-13:
This suggestion works for me. 
I do prefer 3 indent spaces though, but will go with the trend. 

> --style=bsd
> --indent=spaces=2
> --indent-preproc-block
> --indent-preproc-define
> --indent-col1-comments
> --pad-oper
> --pad-header
> --delete-empty-lines
> --align-pointer=type
> --align-reference=type
> --remove-brackets
> --break-after-logical
> 
> And reasonable values for this...
> --max-code-length=#

eboston, 2015-09-13:
@thinkyhead Thanks for the link.  It reminded me of an advantage of having the brace on it's own line.  When debugging, if you want to force the block to be executed, you can just comment out the line with the conditional and not worry about have mis-matched braces.  In the case of if statements, the original line can be commented out and replaces with either if (0) or if (1) to force which block is executed.

fmalpartida, 2015-09-13:
Talking about fashion, can someone with write permissions to the wiki update the logo by the new ones?
I don't think I have access to update it.

Roxy-3DPrintBoard, 2015-09-13:
@Wackerbarth   Can we switch gears for a second?   

What is the Group's thinking about timing on this?    Regardless of which style is used, when does it get cut in?   I'm thinking it would make sense to do it prior to promoting the branch to 'Stable Version' status.   It may make sense to do it immediately because it looks like most important last minute bug fixes have been folded in.

Is there a reason to hold off?

fmalpartida, 2015-09-13:
I am with you on this one, between rc1 and release should be a good timeframe.

Wackerbarth, 2015-09-13:
@fmalpartida -- I disagree. We should either do it IMMEDIATELY (before RC1) if it is going into the 1.1 stable branch or, which I see more practical, as the first step of the reorganization for 1.2

@thinkyhead -- In reference to PR#170, you found it necessary to write "Oops, a tab slipped through. Should be 2 spaces." To me that just weakens your argument against using an automated style enforcer.

I am somewhat confused about "One True Brace" styling.  The examples seem to show it as a variant of K&R with the opening brace for a function on a separate line. Yet the present standard seems to have that bracket on the opening line.

As I have stated previously, although I have no objection to the elements in the present style, I was unable to reproduce the exact style with astyle options.  At least to me, and apparently others here, having a style that we can automatically test is more important than the actual style chosen.

Therefore, I again request that everyone reduce their proposal to a specific set of options so that anyone can run that style definition against our current code base and see the resulting restyled version that it would produce.

Roxy-3DPrintBoard, 2015-09-13:
> I am somewhat confused about "One True Brace" styling. The examples seem to show it as a variant of K&R with the opening brace for a function on a separate line. Yet the present standard seems to have that bracket on the opening line.

From:  http://programmers.stackexchange.com/questions/99543/what-is-the-difference-between-kr-and-one-true-brace-style-1tbs-styles

It says:   The biggest difference between K&R and the One True Brace Style (1TBS) is that in the 1TBS, all if, else, while, and for statements have opening and closing braces, even if they aren't necessary. The purpose is to make it easy to insert new statements and know exactly how they will be grouped.

As an example:

K&R:

int i;
for (i = 0; i < 10; i++)
  printf("Hi.");
1TBS:

int i;
for (i = 0; i < 10; i++) {
  printf("Hi");
}

Wackerbarth, 2015-09-13:
@Roxy-3DPrintBoard -- Yes. But I was referring to the other part of the positioning of braces; namely those associated with function specification:

OUR:
void myFunction() {
  if (myCondition == 0) {
    #ifdef PETER_PARKER
      slingWeb(100);
    #else
      findPhoneBooth();
    #endif
  }
}

1TBS:
void myFunction()
{
  if (myCondition == 0) {
#ifdef PETER_PARKER
    slingWeb(100);
#else
    findPhoneBooth();
#endif
  }
}

GOOGLE:
void myFunction() {
  if (myCondition == 0) {
#ifdef PETER_PARKER
    slingWeb(100);
#else
    findPhoneBooth();
#endif
  }
}

So, which do we want?

Roxy-3DPrintBoard, 2015-09-13:
I like the opening brace up on the same line as the function declaration.   But most people don't like that.   I can live with what ever.    The bigger deal is to cut in the change at a point where it makes sense because it is probably going to make it harder for people to merge code back in.

fmalpartida, 2015-09-13:
I still prefer bsd/Allman.

fmalpartida, 2015-09-13:
Richard, after a second thought I think that you are completely right. If we change it now or as we transition from RC to release all hell is going to break loose as it will be almost impossible to do a merger. All the files will inevitably change. 

I will be running a configuration of the stile in a few minutes in a separate branch to see how it looks. But I am still in favor of the one mentioned before.

fmalpartida, 2015-09-13:
This is more closely related to may style preference. After doing a couple of tests these are the options that match nicely to what may coding preferences are:
--style=bsd
--indent=spaces=3
--indent-preproc-block
--indent-preproc-define
--indent-col1-comments
--pad-oper
--pad-header
--delete-empty-lines
--align-pointer=type
--align-reference=type
--add-brackets
--break-after-logical

I am still not too sure about deleting empty lines as it give the notion of paragraphs and code sections.
Padding operators is the other one I am not too fuzzy about.

thinkyhead, 2015-09-14:
The main reason I had avoided using an automatic formatter when I jumped into Marlin-world was that submitting PRs with sweeping changes screws with people's forks and pull requests. Now that we don't seem to care about that, I'll be curious to see how applying a formatter globally affects things, especially to see how badly it breaks outstanding PRs and how effectively it frustrates the maintainers of other forks.

For my part, I prefer this style (apparently not purely "1TBS"):
```cpp
void myFunction() {
  if (myCondition == 0) {
    #ifdef PETER_PARKER
      slingWeb(100);
    #else
      findPhoneBooth();
    #endif
  }
}
```
Note that I have indented the preprocessor block statements too. That's important for naïve text editors that con't parse C++, but which can parse the general syntax using braces or blocks of indented lines as cues for where to allow code-folding.

The braces around a single line `if` statement should be there if the developer feels like more lines might be added in the future, but in many obvious cases they can be left out. (In other words, we should mix freely and not be such pedants.) I also don't mind (actually very much like) lines like this:
```cpp
if (condition) doSomething();
```

Roxy-3DPrintBoard, 2015-09-14:
> The main reason I had avoided using an automatic formatter when I jumped into Marlin-world was that submitting PRs with sweeping changes screws with people's forks and pull requests. Now that we [...]

No....  I don't think that is quite accurate.  We all care about that.  But given the fact we are going to be starting 'Clean' code bases at the RC1 and 'Stable Release' level...  This is a chance to clean up a lot of dirty stuff.   It looks to me like all good bug fix's and/or features are being folded in.   There shouldn't be much stuff that is outstanding.

And I'm not sure that this is true, but if there is something really good that didn't make it in, the author could just run the Astyle program with the exact same settings that the code is migrating to and their stuff should 'diff & merge' OK?   Right?

But the intent is to do this at a time when it disrupts the fewest number of people....

Wackerbarth, 2015-09-14:
@thinkyhead -- By your thinking, we would still require that someone walk in front of each horseless carriage. We cannot, and should not, live in the past forever. For the common good, we must make changes to modernize our practices. People have been warned, and many are holding off submitting changes so that we can make a change. Unfortunately, you have been a major force in delaying that because you insist on presenting a constant stream of "improvements" rather than pausing long enough to get a code base stable for long enough to even consider it for public release.

I have repeatedly requested that you assist in bringing documentation to an appropriate state for a release. Thus far, I haven't seen any progress in that area.

As for formatting, "user's choice" doesn't work. The repository needs to have a standard ENFORCED by automated tests. Since I know of no automated tested that will exactly match your personal preferences, you need to decide, for yourself, which are the important aspects and compromise on the rest.

In your comments, above, you note "always including braces makes for cleaner diffs".
That is certainly a consideration.

You make an argument about cascaded indention of preprocessor blocks, in the interest of "naïve text editors". To that I suggest, "get a better text editor" or, even better, change our coding practices to reduce the cause. IMHO, having highly nested conditionals is one of the major issues that makes it very difficult to "read" the code.

"For my part, I prefer this style (apparently not purely "1TBS"):"

I, again, request that you present a well defined set of options that most closely matches your preferences.

Wackerbarth, 2015-09-15:
@AnHardt @Wurstnase @CONSULitAS @maverikou @nophead @paclema @epatel @ErikZalm @daid @bkubicek -- AND ANYONE else who is interested in contributing to the project.

We need additional input on this issue.

Wurstnase, 2015-09-15:
I can live with 1TBS or the modified version. But anyhow you make it, it's fine for me. I doesn't make that much work, so I would do my best to use this or that standard.

closed 2015-09-23 by Wackerbarth