MarlinFirmware/MarlinDev #171: dup lines

ruggb, 2015-09-13:
u have had these duplicate lines for at least a month in configuration.h

// Uncomment this option to enable CoreXZ kinematics
//#define COREXZ

// Uncomment this option to enable CoreXZ kinematics
//#define COREXZ

thinkyhead, 2015-09-13:
Are you sure about that? I just did a global search on the project for `define COREX` and I only see `COREXZ` defined once in each configuration file. Are you referring to a different branch other than MarlinDev/master?
```cpp
// Uncomment this option to enable CoreXY kinematics
//#define COREXY

// Uncomment this option to enable CoreXZ kinematics
//#define COREXZ
```

ruggb, 2015-09-13:
yes - I just dnld that last night
line 308 to 312
MarlinDev-master.zip
configuration.h

thinkyhead, 2015-09-14:
@ruggb Look more carefully. One of those is COREXY and the other one is COREXZ. They are separate options.

closed 2015-09-14 by thinkyhead

ruggb, 2015-09-15:
sorry to bother u for my bad eyes