MarlinFirmware/MarlinDev #173: Option to Shutdown Power Supply on MAXTEMP

1-23, 2015-09-13:
It would be an excellent safety feature to have Marlin shutdown an external power supply (same behavior as an M81) if MAXTEMP is reached. This would cause heating to stop, even if a MOSFET fails and becomes suck in the on position. Can this be added as a configuration option?

KiteLab, 2015-09-13:
Since all the temperature errors use kill() now - that's already implemented but not configurable.

1-23, 2015-09-13:
Hmm, that's now how mine behaves. I'll pull the latest version, play with it a little, and report back. Thanks!

thinkyhead, 2015-09-14:
@1-23 Currently `kill()` does this:
```cpp
pinMode(PS_ON_PIN, INPUT);
```
…but maybe on your machine it needs to do this:
```cpp
OUT_WRITE(PS_ON_PIN, PS_ON_ASLEEP);
```

ntoff, 2015-09-17:
I use an xbox 360 PSU on one printer which requires the PSU's ps_on pin go high. Does that affect the way kill() works?

1-23, 2015-09-18:
Updating to the latest version fixed it. Sorry for the false alarm!

closed 2015-09-18 by Wackerbarth