MarlinFirmware/MarlinDev #178: Full graphic LCD doesn't recognize SD card

shampine1, 2015-09-13:
I'm running release 42E... on a Mega/Ramps/full graphic LCD.  It doesn't recognize the SD card at all; I don't see any SD card menu items.  I have tried inserting it, and also powering up with it inserted.

I know the system works because I can print off the SD card with the current release of Marlin.

thinkyhead, 2015-09-14:
I just pushed #180 which is supposed to improve things. But even without that you should have a working SD slot. So make sure you have uncommented `#define SDSUPPORT` in your configuration — this is now required for all controllers, otherwise SD support is disabled by default.

shampine1, 2015-09-14:
Ah, now I understand!  There used to be logic to select when SD support was used.  It would be nice to add a note to that effect in configuration.h.
I've tested it and that's the fix!

closed 2015-09-14 by shampine1

thinkyhead, 2015-09-14:
@shampine1 The problem was that you couldn't disable SD support on certain controllers because `Conditionals.h` would always set it for them. So it was a toss-up between adding a `DISABLE_SDCARD` configuration setting or changing the configuration so that `SDSUPPORT` must be enabled separately. It's "better" but we'll probably get a lot of support requests about this. One option might be to add a check in `SanityCheck.h` for those controllers, and throw an error if `SDSUPPORT` is not enabled. The error message would say something like `comment this line if you want to continue without SD card support`.