MarlinFirmware/MarlinDev #190: Marlin Will Not Compile Using G3D LCD Controller

AlexDoran, 2015-09-16:
Hi there, first of all, i am a totally new to GitHub and this whole environment of reporting problems, second of all, my coding knowledge is basic at best. I have two gMax 1.5 Printers, one is a dual extruder, the other is a single extruder.

The dual extruder printer has a G3D LCD Panel and the Single Extruder has a RepRapDiscount Smart Controller (Not that i think this matters i'm just trying to provide as much info as possible).

 Yesterday i decided to flash the latest Dev Release onto both machines, the single extruder compiles and runs fine (However "Ready." Is most of time printed where the extruder temp should be?), but as soon as i try and compile the dual extruder firmware, it fails with the following error:

ultralcd.cpp: In function 'void lcd_move_menu_axis()':
ultralcd.cpp:165: error: invalid conversion from 'void (*)(uint8_t) {aka void (*)(unsigned char)}' to 'menuFunc_t {aka void (*)()}' [-fpermissive]
         menu_action_ ## type(args); \
                                  ^
ultralcd.cpp:798:5: note: in expansion of macro 'MENU_ITEM'
     MENU_ITEM(submenu, MSG_MOVE_E, lcd_move_e);
     ^
ultralcd.cpp:79: error: initializing argument 1 of 'void menu_action_submenu(menuFunc_t)' [-fpermissive]
   static void menu_action_submenu(menuFunc_t data);
               ^
invalid conversion from 'void (*)(uint8_t) {aka void (*)(unsigned char)}' to 'menuFunc_t {aka void (*)()}' [-fpermissive]

As soon as i revert back to using 1 extruder it compiles perfectly.. Am i just being really thick with something here? 

Thanks

Alex

killpop, 2015-09-29:
I had an issue like this a month or so ago and I think it had to do with formatting of the temp tables that was confusing the compiler. I got around it by using and older version of the arduino ide to compile. I know, not a real solution but it got my machine going.

Wackerbarth, 2015-09-29:
@killpop -- Would you please post an example of a table that compiles under an older compiler, but not the current one.  A gist or a reference to some repo is the preferred way for you to post it.

ruggb, 2015-10-05:
sounds like the thermal runaway issue.
won't compile with both bed and hotend runaway enabled
with 2 hotends
I think it is fixed at least in RC2

thinkyhead, 2016-04-18:
Fixed in `RCBugFix` — which will be getting migrated over here shortly after the 1.1.0 release.

closed 2016-04-18 by thinkyhead