MarlinFirmware/MarlinDev #191: Ramps_13_EFF issues

o-lampe, 2015-09-16:
I try to run my delta without heated bed and therefor defined `MOTHERBOARD BOARD_RAMPS_13_EFF`
1st problem:
 There is no comment in boards.h about HEATER_0 is set to D8 instead of D10
Maybe rename it to **FFE**, to give a hint about the new connection or add a comment like" Extruder is wired to D8, when no bed or second extruder is required"

2nd problem:
I disabled the bed temp probe pin in RAMPS_13.h
Even when I set TEMP_1_PIN to -1, I get an ERR: MINTEMP BED in Display
I´ve tried it this way:
`#if MB(RAMPS_13_EFF)          //add more setups without bed temp`
`  #define TEMP_1_PIN     -1  // No Bed temp`
`#else`
`  #define TEMP_1_PIN         15   // ANALOG NUMBERING`
`#endif`

o-lampe, 2015-09-16:
I solved the 2nd problem with also defining
`#define TEMP_SENSOR_BED 0`
Sometimes the obvious things are the hardest to find ;)

thinkyhead, 2016-04-18:
I assume we fixed this issue in `RCBugFix`…? If so, we can close this issue. Code from `RCBugFix` will be getting migrated to `MarlinDev` (and reorganized) right after 1.1.0 release.

closed 2016-05-25 by boelle