MarlinFirmware/MarlinDev #205: New system

MagoKimbra, 2015-09-21:
Excuse me, could explain better how the whole thing now? If you launch the file MarlinDev.ino me by mistake because not find the file configurations / UUID / e9cbe275-AAEE-4383-9352-01c1bac200ac / Customization.h...
I have Arduino Ide 1.6.5 r5

Wackerbarth, 2015-09-21:
I'm still have a bit to fix up. So you may need to "work around" some things for a few days.

The idea is that you should get a UUID(s) for your own printer(s) and put the top level configuration file under the configurations/uuid/ directory.(I've created a directory for each one so that we can have multiple files should we need them.)

This will allow us to archive particular configurations and be able to reproduce EXACT builds from data stored in git repositories (No editing required).

The intention of the configuration scheme is to treat it like C++ classes. You specify exceptions and then inherit from a more general case.

To flesh this out, we will need to turn some entries into a "defaults_to" syntax.
If we don't use a true macro preprocessor for that purpose, we will make the entries that we might change in the form
```
#ifndef SOMETHING
  #define SOMETHING DEFAULT_VALUE
#endif
```

That way, the user can either define the term, or he will inherit the default value.

As for the assigning of the ```CUSTOMIZATION``` parameter, I'm hoping to work out a good way to handle that in the IDE.

MagoKimbra, 2015-09-21:
It was pretty clear the idea, but what I do not understand how you can include files in subfolders that Arduino IDE does not allow to do.

Wackerbarth, 2015-09-21:
The difference is that I have moved all of the Marlin code into an Arduino library configuration where you can include subfolders.

BuzzBumbleBee, 2015-09-22:
Woow, @Wackerbarth i love this idea. but it seems to me a lot of this is missing. Like the default configs for a device to inherit from :/

Is there a like to whatever discussion happened that lead to this system ? (the new layout isnt exactly 100% obvious to where things should / do live)

KiteLab, 2015-09-22:
Let's wait and see where he will end up.
If we don't want it, we always have the option to fork from RC and do it an other way.

BuzzBumbleBee, 2015-09-22:
@KiteLab i like the idea as a whole (needing to update all X many configurations every few days is a royal pain)

Its just this seems unintuitive, personally i would have done something like

configurations/kinematics/    || all base "descriptions of the machine types" 
configurations/machines/      ||  inherits form kinematics for a description of the machine type eg. i3/3DR
configurations/modules/        ||  descriptions of common optional modules (heated bed / level probe ect) and all build specific dimensions including trim ect
configurations/build_specs/<uid>/    || your configeration file (would normally inherit from a machine and one or more optional modules)
configurations/current_config.h || just a file to include the current build_specs

But i may be missing a major goal of this restructure

Wackerbarth, 2015-09-22:
@KiteLab -- Thanks. You are correct that we can always revert to RC if it doesn't work out. In the short run, I'm not changing any of the functionality.

@BuzzBumbleBee - With respect to the configurations, I intentionally left the path to permit someone to use their 1.1 Configuration and then, slowly, migrate to the revised form. In the interest of getting back to a functional code base, I tackled these existing files first.

Your description (above) seems to be very much in line with my destination.  The only difference is a minor naming (where you used "build_specs" and I have used "uuid").

Note that with this approach, we can start "factoring" the existing configurations.  When we build one of the descriptions as mentioned, we can add the #include for it and then delete the then redundant entries in the particular specification file.

clefranc, 2015-09-22:
@Wackerbarth I personally don't find this UUID very human friendly. What does it mean actually, that I can PR my custom config? If so, I prefer something like biv25_clefranc.

Can you give us a pragmatic usage of your new configuration scheme?

Wackerbarth, 2015-09-22:
@clefranc -- Although UUIDs are not "human friendly", they do provide a consistent nomenclature which is generally accepted in the realm of electronic devices. Personally, I alias my few entries just as I use DNS to alias IP addresses.

I don't really expect us to maintain custom versions via the PR process. But the mechanism is consistent with other distribution schemes and does provide a way that we can communicate EXACT build configurations for the purpose of debugging.

I think that I have the initial reorganization cut in place and welcome assistance in fleshing out the particulars. So if someone would like to start factoring out some of the configurations, please me know.

clefranc, 2015-09-22:
@Wackerbarth Have you finished?

If yes, can you explain to me how can I compile Marlin now? I'm using Window 7 x64. Thank you!

Wackerbarth, 2015-09-22:
(Temporarily) The easiest way would be:
1) Download and install the Arduino IDE 1.6.5 from https://www.arduino.cc/en/Main/Software
2) Download our IDE_1.6.5 branch and install it as an alternate platform in the hardware folder
3) Download our dev branch as a library
4) Install your customizations in an appropriate Customization.h file in the configurations/uuid/ folder
5) Reference that file in FirmwareCustomization.h  (I'm working on an alternative that, instead, will allow you to do this from the IDE menus)
4) Compile the MarlinFirmware.ino found in examples/MarlinFirmware

BuzzBumbleBee, 2015-09-23:
@Wackerbarth although i welcome the changes (the new configuration splits will help a lot in the long run), whats the advantage to building marlin as a lib ? (to me all it does is complicate the build system)

Wackerbarth, 2015-09-23:
Building Marlin as a lib is necessary to allow us to get around the way that the Arduino IDE performs its builds.  If the files are in a lib, they get compiled "in place" rather than being copied, rearranged, etc. before to build. This will allow us to restore structure to the source and create a more modular layout. GRBL, Cosa, and a number of other projects are following the same approach.

thinkyhead, 2015-09-24:
> Let's wait and see where he will end up.

If this is all experimentation and "not ready" and "not documented" — could trigger-happy developers please limit such changes to their own forks, submit them as PRs, and go through the same waiting and approval process as everyone else?

I am saddened to see @eboston losing faith, but I understand his anger and frustration, and I am distressed myself. I am not happy to see broken, not-ready changes of such huge import being pushed onto MarlinDev/master that everyone's forks depend upon. And I am very unhappy to see a lack of documentation and more forthright engagement with the community before pushing things through.

As Marlin is losing its community focus in favor of the "better way" I think I will take a vacation until this coup is over.

clefranc, 2015-09-24:
I give @Wackerbarth the benefit of the doubt, that something "user friendly" will emerge from this.

I didn't give up yet, I was waiting for @thinkyhead reaction, because, honestly, he's the reason I'm still around.

I'm also disappointed that @eboston jumped ship, hope nobody will stop bring new ideas, imaginative concepts, unconventionals thinking, etc... These are the core of Open Source.

BuzzBumbleBee, 2015-09-24:
@Wackerbarth @clefranc 
Would it not be preferable to move this change into MarlinDev/libformat and keep MarlinDev/master in the "old" format until the change is stable and documented, I can see advantages to having the new system in the main  dev repo but in its own branch (so people can have more input, this doesn't really happen much in user forks). But at the moment this lib format change is causing more bad than good.

@thinkyhead 
I 100% agree a change of this type / impact to the main dev branch should be somewhat documented AND tested via multiple devs. From what I can see this lib change wasn't really talked about within the "core" devs :/ 

50% side note have you guys thought of a "review" system such as gerrit ?

Wurstnase, 2015-09-24:
Moving the RC to Marlin/Marlin/RC was not a good idea. We have actually some fixes in MarlinDev which are now adressed to MarlinDev/Master and not the RC. 

That's not fun, that's not discussed. 
That's nothing which help us.

BuzzBumbleBee, 2015-09-24:
@Wurstnase humm a better documented / thought out branching process would help ?

** I can help out with this stuff if needed **

Wackerbarth, 2015-09-24:
@BuzzBumbleBee -- "Would it not be preferable to move this change into MarlinDev/libformat and keep MarlinDev/master in the "old" format until the change is stable and documented" -- No, that won't work.

You cannot make changes to a moving target in that manner.  That is why I waited until I  moved RC back from MarlinDev to Marlin. CORRECTIONS to RC1 are now proceeding in the Marlin while NEW FEATURES are proceeding in MarlinDev.  You will note that I haven't really changed any of the CODE with this change. It's mostly a shuffling of files so that we can utilize better coding practices (such as modules and structured libraries) that were not possible in the previous organization.

By making this change, I have opened the door to permit parallel development of new aspects in the same PR style that we have been using.

@Wurstnase -- Since Marlin and MarlinDev are really the same repository, it is trivial to move items from one to the other locally.  I have my repository set up with both Marlin and MarlinDev as remote repositories and push and pull from either at will.

As for changes, that are not in RC, we had to draw a line somewhere. there has been a call for the Release Candidate that has gone on for months. It was generally agreed that what we had was "good enough", and better than waiting for ever.

If you have some corrections that are "in the pipe" and were not included in RC1, they can be ported to RC2 without much effort.  If they are new additions, they need to be considered "after deadline" and folded into a subsequent release.

BuzzBumbleBee, 2015-09-24:
@Wackerbarth i was more thinking, if it was on another branch for a few weeks people could have provided feedback / corrections / improvements before the "big jump" and merging into Dev master. As iv said above i agree that this change is a good thing and will help marlin develop. As far as i can tell this change is not complete AND not very well documented to the point even core developers of the project don't really know what to do with the dev branch now (@Wurstnase @clefranc @thinkyhead correct me if im wrong here)

I fully understand how a change like this will almost always conflict against other changes added BUT im my head the following works best.

1) Branch made for lib version
2) All changes pushed to lib branch
3) Other review / commend / improve
4) Dev is frozen for a few days while lib branch is merged in
5) Profit ?

Wackerbarth, 2015-09-24:
@BuzzBumbleBee -- Sorry, you are leaving out step 4a) Totally rewrite the lib branch to reincorporate the changes made to dev between (1) and (4).  That's way too much repeated work.  And I tried it, multiple times, without success.

But, if you want to think of it that way, Marlin/RC is one branch and MarlinDev/dev is the other.
Rather than adding more half-baked features, we NEED to be concentrating on POLISHING the existing ones.

We are now at a point where the first step of the change has taken place. If that change is acceptable, we can proceed with making (relatively minor) code changes. Or we can make adjustments to the scheme. If it is really a bad idea, we can even revert back to RC1.

As for major changes, I only caution that the current "build something, then discuss it" approach is a major cause of the mess that modern code finds itself. It leaves out what I consider the most important aspect "DESIGN".  We are leaving out the entire design/review part of development.

As for documentation, it appears that we need to change what we are doing. Scott holds the keys to our present documentation and he isn't cooperating. He won't even document his own changes.

BuzzBumbleBee, 2015-09-24:
@Wackerbarth 

"we NEED to be concentrating on POLISHING the existing ones."

Fair point but this is what the PR review system is for

"As for documentation, it appears that we need to change what we are doing. Scott holds the keys to our present documentation and he isn't cooperating. He won't even document his own changes."

For new features its best for (where needed) the documentation needs to go into the PR request.

Wackerbarth, 2015-09-24:
The PR process has not worked for this project.  Someone writes some code and presents it. They consider the purpose "obvious" or "self explanatory" Then, in most cases, there has been little, or no discussion, before it gets merged into the master branch. Proper documentation NEVER gets written.

Part of the reason for this is that the user level documentation is in some separate repository.

I would like to see a different workflow.
(1) A feature is proposed as an issue. The function and interface are discussed. If it is decided that the concept is a viable addition to the project, a separate dev branch is created for the code.
(2) Through a series of Pull Requests, the code is developed AND Tested. The documentation is a part of the source tree and included in the pull requests.
(3) Periodically, and immediately before it is merged, the code is rebased to the head of the current development branch. (Note: One of the things that makes this difficult in the pre-RC1 structure is that everything is part of one single module. Creating some structure to the code will allow us to actually have some parallel development where we don't have every change "stepping on" someone else.)
(4) NOTHING gets merged into the master branch unless XXX reviewers "sign off" on it as complete.

Yes, I recognize that this will slow down the process. But, IMHO, we have been operating in a "haste makes waste" mode and, as someone else noted, just "spinning our wheels".

BuzzBumbleBee, 2015-09-24:
@Wackerbarth if thats whats needed then look at https://en.wikipedia.org/wiki/Gerrit_(software)

Through the repo tool you push your changes then it gets reviewed commented on +1 / -1 and then merged. The tool itself is the git "master" no one pushed through or invalidates. This would enforce proper code review in an easy to manage way. 

HOWEVER it could be a little heavy handed for marlin.

Also we are digressing from the point what this issue was filed for ;) so ill probably call it a day and wait for other to respond.

Wackerbarth, 2015-09-24:
Yes, there are tools such as Gerrit that automatically enforce such discipline. However, I think we can operate a little more "manually" if EVERYONE will agree to follow some workflow procedure.
The problem exists when even one user starts circumventing such "rules".

I get the feeling that there are some who want to utilize a less disciplined approach.

BuzzBumbleBee, 2015-09-24:
@Wackerbarth fwiw to me it seems that you where "circumventing such "rules"" without sending a PR (unless i missed something :/ )

clefranc, 2015-09-24:
@ErikZalm @daid @Roxy-3DPrintBoard As owners of this repo, I'm curious to know your opinion on this "new system" and all alleged comments stated here. Thank you!

daid, 2015-09-24:
I think the idea of having as many "stock" configurations as possible is good, however, I don't think UUID's are they way to go. UUIDs are in generally something computers are using internally, not something users should be bothered with.
Would make more sense to have "sensible human names" instead of UUIDs.

As for the subdirectories. I do think it's important for the code to work "out of the box" on the latest Arduino environment. No idea if they finally are joining the 1900 and support directories...

Finally, might be interesting to know you can use defines in #include statements:
http://git.savannah.gnu.org/cgit/freetype/freetype2.git/tree/include/freetype/freetype.h?h=STABLE-2-1-0#n41

Wackerbarth, 2015-09-24:
@daid -"Finally, might be interesting to know you can use defines in #include statements:"

Yes, in a restricted sense. And I do so.  Unfortunately, AFAIKT you can

```
#define MY_HEADER "some/path/to/a/header.h"
...
#include MY_HEADER
```
but you cannot
```
#define VARIANT "a"
```
and generate an include from a template
```
#include ""some/path/to/•••VARIANT•••/header.h"
```

Wackerbarth, 2015-09-24:
@daid -- "I don't think UUID's are they way to go"

Perhaps not. UUIDs do provide a mechanism to avoid namespace conflict (as well as disagreements over "ownership" of particular portions of the namespace)

Personally, just as I use DNS to manage access to IP addresses, I alias the UUIDs for my personal devices in my own platform.local.txt and use names of my choice to locally select the particular item that I wish.

And I'm not going to insist that UUIDs be used. If someone presents a better solution that works reasonably for a larger audience, we can always migrate to it and drop the UUID namespace.

However, in defining "better", please take into consideration that I would like to be able to place YOUR configuration, as well as those of others discussing firmware issues, in my repository in a manner that I can easily manage without constantly having to change schemes to re-reference a particular instance.

BuzzBumbleBee, 2015-09-24:
@Wackerbarth you can use macros to generate include names :/

Like ...

```
#define DEVICE delta
#define TYPE my_first_delta

#define BLANK(M)       #M
#define INCLUDE_FILE(N,M)  BLANK(includes/N/M/cfg.h)
#include INCLUDE_FILE(DEVICE,TYPE)
```

Tries to include "includes/delta/my_first_delta/cfg.h" in the Arduino IDE :)

Roxy-3DPrintBoard, 2015-09-24:
That is clever BuzzBumbleBee!   I was trying different approaches with the preprocessor and wasn't getting anywhere.

BuzzBumbleBee, 2015-09-24:
@Roxy-3DPrintBoard only annoying thing is that it does (at least in the IDE) needs to be in a library for the include to works afaik :/

Wackerbarth, 2015-09-24:
Great to see that it works!  Now we can use it to provide some needed structure to configurations.

"needs to be in a library" -- Exactly WHY I made the massive change.

I was trying to do something a bit fancier  

```
#define DEVICE delta
#define WHICH my_first
```
and, from that,

```
#include "includes/delta/my_first_delta.h"
```
Never got it to work.

But we can live with restricting the variables to directory names if necessary.

BuzzBumbleBee, 2015-09-24:
@Wackerbarth 

```
#define DEVICE delta
#define TYPE my_first_delta

#define BLANK(M)       #M
#define INCLUDE_FILE(N,M)  BLANK(includes/N/M.h)
#include INCLUDE_FILE(DEVICE,TYPE)
```

Results "fatal error: includes/delta/my_first_delta.h: No such file or directory" so yea :D

clefranc, 2015-09-25:
@daid Thank you for your guideline. I agree with you that Marlin must preserve its "user friendly" approach. I think it's what that made it so appealing. The community here is very supportive and tolerant and I believe that with the help of @BuzzBumbleBee, @Wackerbarth will eventually provide a practical and powerful solution.

clefranc, 2015-09-25:
That said, I'm still very curious to hear from @ErikZalm and @Roxy-3DPrintBoard.

Roxy-3DPrintBoard, 2015-09-25:
> That said, I'm still very curious to hear from...

This work is going to prove valuable to Marlin's future development.   It allows us to break out and simplify functionality from some very big pieces of the code.  Marlin_main.cpp has grown to the point of being ridiculous.   This re-organization of the code allows this complexity to be cleaned up.   (In particular, think of the simplification it enables for my favorite area: the G29 code.)

With that said, as a community we did a bad job of building consensus and factoring in everybody's concerns.   I do believe cutting in the changes at a 'Release Candidate' stage does make sense.  But even so the community could have benefited from more warning on the direction the code base was going.  We need to **ALL** work better together to make sure we are doing what is Marlin's best interest.

closed 2016-05-04 by jbrazio