MarlinFirmware/MarlinDev #206: Delta servo probe deployment not working

chris-bo, 2015-09-21:
G29 on my delta printer does not deploy (and stow) the servo probe.

I think the issue is located in this part of G29

```c
#if ENABLED(DELTA)
      // Avoid probing the corners (outside the round or hexagon print surface) on a delta printer.
      float distance_from_center = sqrt(xProbe * xProbe + yProbe * yProbe);
      if (distance_from_center > DELTA_PROBABLE_RADIUS) continue;
#endif //DELTA
      ProbeAction act;
      if (deploy_probe_for_each_reading) // G29 E - Stow between probes
        act = ProbeDeployAndStow;
      else if (yCount == 0 && xCount == xStart)
        act = ProbeDeploy;
      else if (yCount == auto_bed_leveling_grid_points - 1 && xCount == xStop - xInc)
        act = ProbeStow;
      else
        act = ProbeStay;
      measured_z = probe_pt(xProbe, yProbe, z_before, act, verbose_level);
```
the skipped corners are causing `xCount==xStart` and `xCount==xStop - xInc` to be always false.

Maybe the ProbeAction for deltas should be handled before and after the probing loop, like it is done with allen key probes? So we could leave the code above as it is, and the G29 E option should then still work as expected.

chris-bo, 2015-09-21:
I could create a PR, but maybe I'll wait until the reorganization of the code is finished.

chris-bo, 2015-09-24:
#50

closed 2015-09-24 by chris-bo

KiteLab, 2015-09-24:
As a workaround you could try to shrink the the probed area until all points are probable.