MarlinFirmware/MarlinDev #209: Manual Offset Value

sarf2k4, 2015-09-23:
Is it possible to manually change the axis offset just like how we manually move the axis?

KiteLab, 2015-09-23:
What do you mean with manually?

Move the head to the position you want with the move menu.
Then select "Set home offsets", what executes a "G92 X0 Y0 Z0".

sarf2k4, 2015-09-23:
The current "set home offset" will take the current head position as its value to save it as home offsets in eeprom, I mean is to manually dial in the values except taking in the current head position to save it in eeprom

Is this possible?

drbogger, 2015-09-23:
But when you start a print, it'll execute a G28, and therefore defeats the purpose of setting the offsets.

The only way around this, is to remove the G28 from your slicer program.

drbogger, 2015-09-23:
However, this is not something I would implement into a menu on LCD. This is something that should be set once, then forgotten.

sarf2k4, 2015-09-23:
I'm not sure why, but during printing the bed seems to be varies in height. By having the head parked outside of the bed, lower than the bed physically when homed, set the z offset by about -0.4, this is to have the z axis raised. By having offset to positive value e.g z= 0.5, marlin won't go lower than 0.5, discussed in another issue

My bed varies from print to print, so I had to use baby steps. Currently I'm trying to work on the auto bed leveling on my printer with z probe enabled with z min. But having the offset set manually is pretty much good I think

drbogger, 2015-09-23:
Sounds to me like you're endstop is not very accurate, or your bed is not very stable.

I set my endstop once, and it's good for over a month. I think you should to look at your printer for the issue, and resolve this issue at hardware level. That's just my advise, hope it helps!

sarf2k4, 2015-09-23:
I try to clean the glass bed first and do the auto bed leveling too, but how about the manual offset input? Is it possible?

thinkyhead, 2016-04-18:
Housecleaning…

closed 2016-04-18 by thinkyhead