MarlinFirmware/MarlinDev #210: How about a wiki  here

KiteLab, 2015-09-23:
http://www.marlinfirmware.org/index.php/Main_Page fails the main criteria for a wiki - to be editable by everyone interested.
And it's going to be outdated.

Especially for MarlinDev we need some essential info like coding style, how to set up and compile.

@Roxy-3DPrintBoard 
Please activate the wiki.

drbogger, 2015-09-23:
I could help out keep the wiki updated, if you would like. I just need access to be able to.

KiteLab, 2015-09-23:
Every github user has write axes to the github-wikis. But an owner has to activate it.

Roxy-3DPrintBoard, 2015-09-23:
Let me check with a few people and see if there are any concerns....

AnHardt, 2015-09-23:
:+1: 
I don't see any advantage in Marlinfirmware.org.
Mainly it lacks on write-accessibility. Yes, it was spamed - but github-wikis rarely are.

Hope to return to the desk soon. Still wrestling with finishing a shi..y Sintron-Mini_Kossel.

drbogger, 2015-09-23:
Why not enable write access, but make it so the posts have to be approved? Or is that not an option?

fmalpartida, 2015-09-23:
I tried to get write access to the wiki, prior to RC1.1 to add a bit of documentation but the request is in the limbo with the write access request I did a few months back. 

If the wiki is read only there is no much point in it. I guess. There are options: git or Reprap.

Roxy-3DPrintBoard, 2015-09-23:
@fmalpartida   Can you tell us the history and why it wasn't just done at GitHub?    

Everybody else:   Would a GitHub Wiki work for you?

clefranc, 2015-09-23:
@Roxy-3DPrintBoard The domain was created originally for the configurator (March 2015). @thinkyhead did the main (all) documentations. A few (me included) asked for write permission, but no one never had granted access. I've lost interest...

https://github.com/MarlinFirmware/Marlin/issues/1592

Roxy-3DPrintBoard, 2015-09-23:
Thank You Clefranc!   

@scotty1024   Are you managing this domain?   If not, should we be discussing alternatives?

fmalpartida, 2015-09-23:
> MarlinFirmware/Marlin#1592

Me too. 
At the time it looked as a good idea to have a separate documentation repo so that we could mimic what we have done in clone wars, document the public interface for alignment with the host's apps. 
A place to publish releases, features, configuration, expected behaviors and translations to other languages. 
Now we have a RC with little documentation on configuration, setups, where we are going,...  
I made a call to the Clone Wars community to try the RC out and it's turning into a nightmare (getting flooded with questions all over). 
From my point of very I lost complete interest last week when I gave it a last shot (a few days before Richard made he's call for documentation prior to RC release).

thinkyhead, 2015-09-24:
Some history: I was writing documentation and posting on the wiki here at Github (on the main Marlin project, not the MarlinDev project), but the wiki here isn't that great. It was suggested by @boelle that Marlin should have its own domain as a central point and clearinghouse for all things Marlin. Scott Turner @Scotty1024 (you can use the same handle with @me.com for his email address) created the domain and set up MediaWiki. I was able to get an account and start editing the documentation there, expecting more editors to jump in. But it was hard to get an account because the server is locked down and only @Scotty1024 can grant access. (He may respond to email better than Github.) The site isn't trafficked enough to open up write access to everyone. There simply aren't enough site police to fix every graffiti and spam post.

- [web configurator](https://github.com/MarlinFirmware/Marlin/issues/1592) - Bo: "i wonder if marlin should have its own domain name?"
- [Move documentation](https://github.com/MarlinFirmware/Marlin/issues/2002) - Bo: "I got this idea of having all documentation in wiki format on the site and use github for the code only…" "the goal is to have all documentation in one place and not a mix between github and reprap.org"

> Now we have a RC with little documentation on configuration, setups, where we are going

Seems there was a rush to get it out. I was surprised to see it announced. I have been moving so not much time for Marlin involvement. I see MarlinDev/master has also diverged from RC1. Hmm…

Wackerbarth, 2015-09-24:
Since we don't have, and I don't expect us to have, reasonable access to the existing documentation site,
and, because we need to be able to tie documentation to particular versions of the Firmware,
if we can export information from a wiki into revisioned markdown documents, I suggest that we open the wiki as a collaborative workspace FOR A LIMITED TIME. At the end of that TRIAL period, we can export the results to static files in a Documentation folder or continue with the wiki.

So, can someone tell us if there is an easy way to perform such an export.
If not, I suggest we simply open a DOCUMENTATION branch and update it via Pull requests.
I know that we can check out those files and add them to our dev branch.

AnHardt, 2015-09-30:
@Roxy-3DPrintBoard 
I interpreted 'here' as 'in MarlinDev'. We need a place to describe the differences.

clefranc, 2015-09-30:
I still want to contribute to the domain, if @scotty1024 can grant me access. And I hope others will follow if they eventually can get access too. I think MediaWiki website is way more versatile than the GitHub one.

Also, since we have 1.1 or 1.2 (or whatever) freezed, perhaps @thinkyhead (or anyone interested) will resume works on the configurator.

Wackerbarth, 2015-09-30:
@AnHardt -- How will we be able to handle versioning of the information in the wiki?
We are certain to be in a case where the various versions of the firmware have differences that need documenting. Further, we need to keep older documentation available for those users who have not, or perhaps even cannot, upgrade to a more recent version. We also need to be able to move the documentation readily.

I'm not suggesting doing so, but one option is to have a Documentation folder that is checked in as a part of the branch to which it applies. I suggest that some equivalent capability be used.

KiteLab, 2015-09-30:
At least 'system setup' and 'compiling process' are totally different in the repositorys and desperately need findable documentation. And if the differences are not blinking red that will confuse the Marlin-branch users.

I'd love a Mediawiki. With emphasis on wiki. Without write access it's not a wiki.

Ciao then.

Wackerbarth, 2015-10-05:
@KiteLab -- Well, we have a wiki on MarlinFirmware/Marlin. That would seem to put us closer to your desired situation.  If it still isn't sufficient, please make a SPECIFIC recommendation about how we might continue to change things.  Unlike the situation when you opened this issue, we have at least "gotten the keys back" and there is someone, active, who can make changes to the permissions.

closed 2015-10-06 by KiteLab