MarlinFirmware/MarlinDev #211: Bug: Change Filament Feature

drbogger, 2015-09-23:
Okay, I decided to test out the "Change Filament" feature, and before it continues the print, it pulls the filament completely back out again.

When I commented out ```#define AUTO_FILAMENT_CHANGE```, it doesn't have this issue.

The issue here I believe is the current position of the extruder is not being restored from the remembered position.

**FILE: Configuration_adv.h**
```
// Add support for experimental filament exchange support M600; requires display
#if ENABLED(ULTIPANEL)
  #define FILAMENTCHANGEENABLE
  #if ENABLED(FILAMENTCHANGEENABLE)
    #define FILAMENTCHANGE_XPOS 10
    #define FILAMENTCHANGE_YPOS 3
    #define FILAMENTCHANGE_ZADD 10
    #define FILAMENTCHANGE_FIRSTRETRACT -2
    #define FILAMENTCHANGE_FINALRETRACT -100
    #define AUTO_FILAMENT_CHANGE                //This extrude filament until you press the button on LCD
    #define AUTO_FILAMENT_CHANGE_LENGTH 0.04    //Extrusion length on automatic extrusion loop
    #define AUTO_FILAMENT_CHANGE_FEEDRATE 300   //Extrusion feedrate (mm/min) on automatic extrusion loop
  #endif
#endif
```

drbogger, 2015-09-28:
Does anybody else have this issue?

juanramos1, 2015-10-15:
Hi, 

I have exactly the same problem.

I think the problem is:

The posicion is saved, as example, X100 Y100 Z100 E100
The head goes to X0 Y0 Z110 
The filament goes out of the extruder
The extruder moves the filament inside the hotend adding values to E100 (AUTO_FILAMENT_CHANGE)
You press the boton when the filament is okay, now E150
The extruder goes back to X100 Y100 Z100
The extruder goes back to E100 which means -50 E distance. The filament is out again.

The logic needs to ignore the changes made to E values during filament change, or delete it before continue printing again.

antonkopa, 2016-01-23:
Good day, I managed to solve this problem?

Roxy-3DPrintBoard, 2016-01-23:
> The logic needs to ignore the changes made to E values during filament change, or delete it before continue printing again.

Yes, I sort of agree with your analysis.   But remember, this code was written with an assumption.  This snippet is from the M600 code: 
~~~~
    current_position[E_AXIS] = destination[E_AXIS]; //the long retract of L is compensated by manual filament feeding
    plan_set_e_position(current_position[E_AXIS]);
~~~~
We should add a couple lines of code to do just as you say so even non-manual movement of the extruder feed is handled.

antonkopa, 2016-01-23:
I am weak in programming but this function extremely needs me. Can eat the prepared decision of changing of plastic in manual ?
even as be before with the manual serve of new plastic but for sewing of Marlin RC3

Roxy-3DPrintBoard, 2016-01-23:
What you typed doesn't really make sense.   Maybe you can try typing it in your native language and I'll run it through Google Translate.

But if you can manually advance the new filament into your extruder and get it so it is ooozing out of the nozzle, I think you won't have a problem.

With that said, we should probably be tolerant of what ever method a person uses to get the new filament feed into the extruder.

antonkopa, 2016-01-23:
I apologize for translation quality ((, can it is possible realization to get from sewing of Witbox https://www.youtube.com/watch?v=rHaLD2r-s68

Roxy-3DPrintBoard, 2016-01-23:
Please type a detailed message of what you need and the problems you are seeing in your native language.   I'll try to use that to understand what you need.   But, if you turn off the #define AUTO_FILAMENT_CHANGE   probably it is going to work for you.  Especially if you manually advance the new filament into the extruder.

antonkopa, 2016-01-23:
Проблема у меня такая же как описана в теме, пытаюсь узнать как можно в прошивки Marlin RC3 сделать замену пластика который заканчивается во время печати на новый. Раньше данная функция работа таким образом, экструдер отъезжал выдавливал старый пластик, и принтер издавал звуковой сигнал на смену пластика, после его замены нужно было в ручную его продавить и печать продолжалась. Можно ли данную схему вернуть в новую прошивку и если да то как.

Roxy-3DPrintBoard, 2016-01-23:
Yes, I think you can return to the way it was.   But there are a couple of things that are not clear to me.   Are just trying to change the filament in the middle of a print because you want to change colors or some other reason?   Or are you trying to have the firmware automatically detect the spool of filament has run out of material and have it start the switch to a new spool automatically.

If it is the first case, that is pretty easy.   You can use the LCD Panel or you can can put an M600 into your GCode file at the right level.   In this case you would just switch the filament and continue.   In this case you would want to leave AUTO_FILAMENT_CHANGE undefined (commented out).

If it is the second case, we might need to get somebody else to answer the question.  But lets get a little more detail.  If it is the second case where you want to let a spool run out of material and have it start doing the swap to a new spool of plastic, please answer some questions:

Can you manually feed the new filament into the extruder?  Does that help with the problem?  

What was the version of Marlin firmware where things worked the way you needed them to be?
Can you post the Configuration.h file from that version?

What are your current values for these values?  Which ones are enabled or have numerical values?
FILAMENTCHANGEENABLE
FILAMENTCHANGE_FIRSTRETRACT
FILAMENTCHANGE_ZADD
FILAMENTCHANGE_XPOS
FILAMENTCHANGE_YPOS
FILAMENTCHANGE_FINALRETRACT;
AUTO_FILAMENT_CHANGE
AUTO_FILAMENT_CHANGE_FEEDRATE
FILAMENT_RUNOUT_SENSOR

antonkopa, 2016-01-24:
What I want - I just trying to change the filament in the middle of a print because i want to change colors or some other reason.

"You can use the LCD Panel" - I can not no item in marline rc3 it worked in Marlin-Development. (в том то и дело что нет в меню как раньше функции смены пластика, ее можно включить только активировав define FILAMENTCHANGEENABLE )

And current values in my marlin rc3

// Add support for experimental filament exchange support M600; requires display
#if ENABLED(ULTIPANEL)
 #define FILAMENTCHANGEENABLE
#if ENABLED(FILAMENTCHANGEENABLE)
  #define FILAMENTCHANGE_XPOS 3
   #define FILAMENTCHANGE_YPOS 3
    #define FILAMENTCHANGE_ZADD 10
   #define FILAMENTCHANGE_FIRSTRETRACT -2
  #define FILAMENTCHANGE_FINALRETRACT -110
   #define AUTO_FILAMENT_CHANGE                //This extrude filament until you press the button on LCD
    #define AUTO_FILAMENT_CHANGE_LENGTH 0.04    //Extrusion length on automatic extrusion loop
  #define AUTO_FILAMENT_CHANGE_FEEDRATE 300   //Extrusion feedrate (mm/min) on automatic extrusion loop
 #endif
 #endif

antonkopa, 2016-01-24:
Товарищ http://roboforum.ru/member20805/ помог с данной проблемой  :) 

Итак в файле configuration_adv.h нужно только расскоментировать  #define FILAMENTCHANGEENABLE
А в файле marlin_main.cpp  нужно заменить кусок  функции М600 

[code]
#if ENABLED(FILAMENTCHANGEENABLE)  // Смена нити

  /**
   * M600: Pause for filament change
   *
   *  E[distance] - Retract the filament this far (negative value)
   *  Z[distance] - Move the Z axis by this distance
   *  X[position] - Move to this X position, with Y
   *  Y[position] - Move to this Y position, with X
   *  L[distance] - Retract distance for removal (manual reload)
   *
   *  Default values are used for omitted arguments.
   *
   */

inline void gcode_M600() {

    if (degHotend(active_extruder) < extrude_min_temp) {  // Проверяем температуру, если она ниже минимальной то выводим ошибку.
      SERIAL_ERROR_START;
      SERIAL_ERRORLNPGM(MSG_TOO_COLD_FOR_M600);
      return;
    }

    float lastpos[NUM_AXIS], fr60 = feedrate / 60;  //запоминаем позицию и тукущую подачу.

    for (int i = 0; i < NUM_AXIS; i++)
      lastpos[i] = destination[i] = current_position[i];

    #if ENABLED(DELTA)
      #define RUNPLAN calculate_delta(destination); \
                      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], destination[E_AXIS], fr60, active_extruder);
    #else
      #define RUNPLAN line_to_destination();
    #endif

    //retract by E
    if (code_seen('E')) destination[E_AXIS] += code_value();
    #ifdef FILAMENTCHANGE_FIRSTRETRACT
      else destination[E_AXIS] += FILAMENTCHANGE_FIRSTRETRACT;
    #endif

    RUNPLAN;

    //lift Z
    if (code_seen('Z')) destination[Z_AXIS] += code_value();
    #ifdef FILAMENTCHANGE_ZADD
      else destination[Z_AXIS] += FILAMENTCHANGE_ZADD;
    #endif

    RUNPLAN;

    //move xy
    if (code_seen('X')) destination[X_AXIS] = code_value();
    #ifdef FILAMENTCHANGE_XPOS
      else destination[X_AXIS] = FILAMENTCHANGE_XPOS;
    #endif

    if (code_seen('Y')) destination[Y_AXIS] = code_value();
    #ifdef FILAMENTCHANGE_YPOS
      else destination[Y_AXIS] = FILAMENTCHANGE_YPOS;
    #endif

    RUNPLAN;

    LCD_ALERTMESSAGEPGM("Выгрузка прутка !");
    if (code_seen('L')) destination[E_AXIS] += code_value();
    #ifdef FILAMENTCHANGE_FINALRETRACT
      else destination[E_AXIS] += FILAMENTCHANGE_FINALRETRACT;
    #endif

    RUNPLAN;

    //finish moves
    st_synchronize();

    
 LCD_ALERTMESSAGEPGM("Пруток вставлен ?");
    millis_t next_tick = 0;
    while (!lcd_clicked()) { 
      millis_t ms = millis();
      if (ms >= next_tick) {
      lcd_quick_feedback();
      next_tick = ms + 1000; //
      }
        manage_heater();
        manage_inactivity(true);
        lcd_update(); 
    }
  
    //disable extruder steppers so filament can be removed
    disable_e0();
    disable_e1();
    disable_e2();
    disable_e3();
    delay(300);
    
    #if ENABLED(AUTO_FILAMENT_CHANGE)
      current_position[E_AXIS] = 0;
      st_synchronize();
    #endif
    
    LCD_ALERTMESSAGEPGM("Загузка прутка !"); 
    //return to normal
    if (code_seen('L')) destination[E_AXIS] -= code_value();
    #ifdef FILAMENTCHANGE_FINALRETRACT
      else destination[E_AXIS] -= FILAMENTCHANGE_FINALRETRACT;
    #endif

    next_tick = 0; 
    while (!lcd_clicked()) {  
        current_position[E_AXIS] += AUTO_FILAMENT_CHANGE_LENGTH;
        destination[E_AXIS] = current_position[E_AXIS];
        line_to_destination(AUTO_FILAMENT_CHANGE_FEEDRATE);
        st_synchronize();
    }
    lcd_quick_feedback(); // click sound feedback
    
  //  lcd_reset_alert_level();

    #if ENABLED(DELTA)
      // Move XYZ to starting position, then E
      calculate_delta(lastpos);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], destination[E_AXIS], fr60, active_extruder);
      plan_buffer_line(delta[X_AXIS], delta[Y_AXIS], delta[Z_AXIS], lastpos[E_AXIS], fr60, active_extruder);
    #else

    LCD_ALERTMESSAGEPGM("Продолжаем печатать");
    
      // Move XY to starting position, then Z, then E
      destination[X_AXIS] = lastpos[X_AXIS];
      destination[Y_AXIS] = lastpos[Y_AXIS];
      line_to_destination();
      destination[Z_AXIS] = lastpos[Z_AXIS];
      line_to_destination();
      current_position[E_AXIS] = lastpos[E_AXIS];
      st_synchronize();
    #endif

    #if ENABLED(FILAMENT_RUNOUT_SENSOR)
      filrunoutEnqueued = false;
    #endif

  }
[/code]

Вот получилось оочень удобно.[/quote]

Roxy-3DPrintBoard, 2016-01-24:
You said: Вот получилось оочень удобно.   Which translates to "That turned out very convenient."  That kind of sounds like you got it going.   But If so, can you confirm that?

I think you realize from the two sets of #define's that you should not have #define AUTO_FILAMENT_CHANGE enabled if you are just trying to manually change the filament.  The Auto Filament Change stuff is to enable the printer to watch for the filament being used up and to automatically start the change process to a new roll.

I think what you are saying in these messages is you don't have a Filament Change command on the LCD Panel's menu.    First, you need ULTIPANEL defined.  But you also need FILAMENTCHANGEENABLE defined to get that to happen.   One thing to check, with these things defined, is can you give the printer a M600 command from ProterFace?   If so, the code is built into the firmware, and it is just the LCD Panel that can not get to it.

There have been some changes to the LCD Panel code near where the Filament Change menu item used to be.  Here is something to try.   Find this code.  It is at the end of a function:
~~~~
  //
  // Change filament
  //
  #if ENABLED(FILAMENTCHANGEENABLE)
     MENU_ITEM(gcode, MSG_FILAMENTCHANGE, PSTR("M600"));
  #endif
  END_MENU();
}
~~~~
Leave the END_MENU() where it is.   Make the start and end of the function look like this:
~~~~
/**
 *
 * "Tune" submenu
 *
 */
static void lcd_tune_menu() {
  START_MENU();
  //
  // Change filament
  //
  #if ENABLED(FILAMENTCHANGEENABLE)
     MENU_ITEM(gcode, MSG_FILAMENTCHANGE, PSTR("M600"));
  #endif

  //
  // ^ Main
  //
  MENU_ITEM(back, MSG_MAIN, lcd_main_menu);
...
  #if ENABLED(BABYSTEPPING)
    #if ENABLED(BABYSTEP_XY)
      MENU_ITEM(submenu, MSG_BABYSTEP_X, lcd_babystep_x);
      MENU_ITEM(submenu, MSG_BABYSTEP_Y, lcd_babystep_y);
    #endif //BABYSTEP_XY
    MENU_ITEM(submenu, MSG_BABYSTEP_Z, lcd_babystep_z);
  #endif
  END_MENU();
}
~~~~
This should put the Filament Change menu item at the front of the list.   That just might cure your problem.   Please let us know what happens....

antonkopa, 2016-01-24:
На пост выше я написал полное решение проблемы автора ветки и мое, все работает отлично. Основные изменения описанны и проводятся в файле marlin_main.cpp

thinkyhead, 2016-05-03:
New work is being done on this feature. Keep an eye on the PRs MarlinFirmware/Marlin#3601 MarlinFirmware/Marlin#3662

closed 2016-05-03 by thinkyhead