MarlinFirmware/MarlinDev #212: New Layout Idea

BuzzBumbleBee, 2015-09-24:
Seeing as the new layout is up in the air please see

https://github.com/BuzzBumbleBee/Marlin/tree/LibDev
https://github.com/BuzzBumbleBee/MarlinLib

Marlin in this case gets all machine configuration from MarlinLib, all the user needs to do is download MarlinLib as a zip and add to the IDE

This has the following advantages in my eyes :- 
    1) Configurations can now inherit via the use of proper include paths
    2) Common code can be slowly moved over to MarlinLib to clean up the code
    3) Only adds 1 more stage to the build system (2 clicks in the IDE)

Thoughts ?

BuzzBumbleBee, 2015-09-24:
More information on how this would work 

https://github.com/BuzzBumbleBee/MarlinLib/blob/master/readme.md

Wackerbarth, 2015-09-24:
Either you are I are missing something. I NEVER download zip files to obtain the source with which I am working. The idea is to be able to embed a git repository directly into my build environment. And, for those who do work from zip files, have it structured to take advantage of the automatic library update mechanism recently added to the IDE.

Except for the addition of the macros (which I certainly want to pick up and utilize as a means of merging alternates without explicit individual paths) you have neither addressed the need either to embed that git repository as the library or utilize an automatically generated zip file based on that repository.

Nor, which is an essential maintenance objective, have you done any factoring of the configuration options into manageable, sharable, subunits.

In particular, since we both seem to have Kossel Pro systems, I would like to get to the position where I simply add a small file that describes the differences between our machines. For example, I'm sure that your limit switch positions will differ from mine by some very small, but significant, amount.
However, since we both have the same structure, stepping motors, wiring harnesses, etc., most of our "configuration" is otherwise the same. So, to get all of those values, we should both be including the configuration for a typical Kossel Pro which has its limit switch positions set to the design value.

Further, a typical Kossel Pro and a Kossel Mini share much of the same configuration (other than the specific dimensions) and so they should utilize a common configuration for a typical delta printer, etc.

Now, it isn't a tree, but a directed acyclic graph. Cartesian systems which use a RAMPS card will share a lot of characteristics whether they are CORE-XY, CORE-XZ, CARTESIAN w/2 Extruders or DELTA.

The idea is to generate a HAL abstraction and use it to factor the parts of the configuration.

BuzzBumbleBee, 2015-09-24:
> Either you are I are missing something. I NEVER download zip files to obtain the source with which I am working. The idea is to be able to embed a git repository directly into my build environment. [...]

> Except for the addition of the macros (which I certainly want to pick up and utilize as a means of merging alternates without explicit individual paths) you have neither addressed the need either to [...]

Kinda missed the point, the mentioning of the zip is the quickest way to set marlin up for those who just want to use it, ofc I use a git repo to work in (in this case inside ~/Arduino/libraries/)

>Nor, which is an essential maintenance objective, have you done any factoring of the configuration options into manageable, sharable, subunits.

True, but this can quickly be done now that the configurations have been move into a library, along with that parts of the system outside of config can slowly me moved over.

This is just a less intrusive option to your change for people to look over.

Wackerbarth, 2015-09-24:
I believe that we are thinking along the same lines. However, perhaps all you are doing with this examble is using a different name for the (now library) version of things and pushing stuff down a level.

For distribution purposes, I'm trying to make the code a library AT THE TOP LEVEL so it can be automatically redistributed.  To install you just install your IDE and (once we are ready, click to install the platform. I think that doing so will then automatically tell you that your library is out of date and offer to update it for you.    In any case, that is the direction that I am trying to reach.

BuzzBumbleBee, 2015-09-25:
@Wackerbarth so we are aiming for something close to https://learn.adafruit.com/adafruit-arduino-ide-setup/arduino-1-dot-6-x-ide WRT setup ?

Wackerbarth, 2015-09-25:
Yes, that looks good to me. However, I will admit that I would be just as happy to simply provide the URL to our platform and refer the users to the Adafruit (or some other) explanation of how to set up to use it.

BuzzBumbleBee, 2015-09-25:
OK I think i get how this is looking, how far are you along with modifying the build system to properly handle inherited configs & getting able to "just add" the platform to the IDE via URL ?

Also please note i found with the current RC code that the configs call includes in a very strange way

BuzzBumbleBee, 2015-09-25:
humm looking at 

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x---package_index.json-format-specification

we can easy push hardware/marlin/avr from IDE_1.6.5 but not the library :/ its almost as if we need 3 different repos :- 

* Marlin
* MarlinLib
* MarlinPlatform

Wackerbarth, 2015-09-25:
I don't think so. The json files are "static" and we can serve them directly from a file buried in any repository. Each of the "repos" you suggest need be only branches of a common repository.

As I see your items, your "MarlinPlatform" is the IDE_1.6.6 branch; Your "MarlinLib" is "dev", "RC", "1.2.x", etc. And your "Marlin" is just the READMEs/wiki.

BuzzBumbleBee, 2015-09-25:
@Wackerbarth yeah, at least splitting out IDE_1.6.6 would make it a little more manageable as far as branches go. Also im 99% we cant serve the librarys in the same JSON way as we can the platform. So the repos will need a rework to get the tar files correct.

>Also "I don't think so. The json files are "static" and we can serve them directly from a file buried in any repository. Each of the "repos" you suggest need be only branches of a common repository."

What I think is the best method for this is within MarlinPlatform repo we would have a "manifest" branch that only contains the json file. This JSON then refers back to the different MarlinPlatform tags/branch zips, whenever we tag a new stable version we add it to the JSON. That way the user can select any version of the platform code within the IDE.

Wackerbarth, 2015-09-25:
I think the libraries come a .zip files -- look at the way that [Cosa](https://github.com/mikaelpatel/Cosa) Is going. Mikael seems to have a handle on this.

BuzzBumbleBee, 2015-09-25:
@Wackerbarth yeah look at https://github.com/mikaelpatel/Cosa/archive/v1.0.3.zip

Our "IDE" branch / repo needs to follow that directory wise so

*bootloaders
*boards
*libraries
*ect.

Need to be top level with all of the .txt files

Wackerbarth, 2015-09-25:
@BuzzBumbleBee -- "the build system to properly handle inherited configs"
The build system can handle that now. But I am sure that it can be improved.
As to HOW we split up the configs, that is a separate issue.

You seem to have some good ideas on the subject and are a bit more up-to-date than I on writing macros to build include paths.

Let's take "Inherited Configs" over to https://github.com/Wackerbarth/MarlinDev/tree/Configs and bring it back in a much more complete, clean, form.

Wackerbarth, 2015-09-25:
No, I think that Cosa is off by one layer on that. I embed his repository as ```.../Documents/Arduino/hardware/Cosa/avr``` and I think that it should come in one level higher.

BuzzBumbleBee, 2015-09-25:
@Wackerbarth ill try and add my macro stuff to your dev tree over the weekend, i need to get the new layout building and running. I would suggest a structure like 

* configurations/{machine_type}/base.h **Base config for all {machine types} (Delta ect.)**
* configurations/{machine_type}/{machine}/machine.h **Inherits from base config from that machine type (only defines changes / additions to the base)**
* configurations/{machine_type}/{machine}/{variation}/config.h **Inherits from machine config, should only need minor adjustments defined for build error / component changes, these would be what the users normally add/modify**

Also I would have

* configurations/extras/{module_name}/extra.h **inherit this from either / variation OR machine to add extra's to the system like bed level / lights / LCD ect**

I also think we select the configs via defines like 

>#devine MACHINE_TYPE delta
#define MACHINE 3dr
#define VARIATION bumblebot10000

Wackerbarth, 2015-09-25:
"configurations/{machine_type}/{machine}/{variation}/config.h Inherits from machine config, should only need minor adjustments defined for build error / component changes, these would be what the users normally add/modify" ---

This is what I have in the uuid area.  Although we certainly need the functionality, I think that we should keep the individual user's part high in the tree.  In particular, I want to have a scheme that allows the user to add his machines to an IDE menu. For ease of use, I suggest that in doing so, we reduce individual machines to a flat namespace. Within that customized configuration file, the user can specify the elements such as {machine_type} which we then use to access the appropriate element down in the tree.

BTW, earlier, I had used "options.h" for the same purpose that as your "extras". So we are certainly thinking along parallel lines.

closed 2016-05-04 by jbrazio