MarlinFirmware/MarlinDev #213: Duplicate "Tune > Flow" menu when using single extruder.

drbogger, 2015-09-24:
I am using a Full Graphic LCD, and during a print, the menu options "Tune > Flow" is duplicated.

Here is the fix I found that works:

**FILE: ultralcd.cpp**
```
 //
  // Flow:
  //
  MENU_ITEM_EDIT(int3, MSG_FLOW, &extruder_multiplier[active_extruder], 10, 999);
  //
  // Flow:
  // Flow 1:
  // Flow 2:
  // Flow 3:
  // Flow 4:
  //
#if EXTRUDERS == 1
 // MENU_ITEM_EDIT(int3, MSG_FLOW, &extruder_multiplier[0], 10, 999); // Disabled to prevent duplicate "Tune > Flow" menu options when using 1 extruder.
#else // EXTRUDERS > 1
  MENU_ITEM_EDIT(int3, MSG_FLOW MSG_N1, &extruder_multiplier[0], 10, 999);
  MENU_ITEM_EDIT(int3, MSG_FLOW MSG_N2, &extruder_multiplier[1], 10, 999);
#if EXTRUDERS > 2
  MENU_ITEM_EDIT(int3, MSG_FLOW MSG_N3, &extruder_multiplier[2], 10, 999);
#if EXTRUDERS > 3
  MENU_ITEM_EDIT(int3, MSG_FLOW MSG_N4, &extruder_multiplier[3], 10, 999);
#endif //EXTRUDERS > 3
#endif //EXTRUDERS > 2
#endif //EXTRUDERS > 1
```

drbogger, 2015-09-28:
Can somebody verify this?

thinkyhead, 2016-04-18:
This is patched over at `RCBugFix` and that code will be getting migrated over here after the 1.1.0 release. So if `RCBugFix` is behaving properly then we can close this issue.

closed 2016-04-18 by thinkyhead