MarlinFirmware/MarlinDev #214: A new update race?

justmyopinion, 2015-09-25:
@mikenz Wackerbarth 
Arduino IDE has already passed latest introduction as latest version is now 1.77.
How are you going to keep up on this new problem introduced?

KiteLab, 2015-09-25:
Until now we support only the Arduino.cc versions. (https://www.arduino.cc/en/Main/Software). Their latest release is 1.6.5 and 1.6.6 is the version of the HOURLY BUILDS.
Adruino IDE 1.1.7 is from Arduino.org (http://www.arduino.org/downloads)

Never tried the second.

clefranc, 2015-09-25:
Can I also add that since Marlin is going with its own hardware libraries, using 1.7.7 is basically only switching for its editor.

If you're a developper/debugger/tester, I highly recommand to use the HOURLY BUILDS, not only have the latest bug fix (if any), but also a great new feature find in its Window title.

justmyopinion, 2015-09-25:
Had no idea about Arduino being a split organization :-(, I am using 1.77 with no problem sofar.

Wackerbarth, 2015-09-25:
There are two aspects of this. One is that, for the less knowledgable user, I feel that we need to support some system that makes it reasonably easy for them to make a few minor changes and update their firmware. Thus far, the Arduino IDE seems to best fit that need.

But the IDE is really just a wrapper around a toolset. Those who know how to use those same, or similar, tools within a different IDE are free to do so. Ultimately, they produce effectively the same code.

But the other aspect of the issue is that I don't think that it should fall upon us, the Marlin community, to support the IDE community. So I adopt the attitude that we should have ONE, and only one, OFFICIAL build environment. Anyone who chooses to use another should be expected to verify that their "Marlin" issue is actually an issue with Marlin and not in their build procedure. I think that many developers are capable of meeting that expectation.

I cannot quite support the comment that @clefranc just added. Using the HOURLY BUILD is a bit too cutting edge (and sometimes it is broken) (Disclaimer: I make my own "HOURLY BUILD" directly from their git repository every few days.) to recommend to everyone. But I like the general direction that they are going (although their mashup of changes to the IDE vs the MCU cores is getting a bit confusing)

I changed the structure of the Marlin source files for two main reasons. (1) By making it a library we regain the #include capability of standard C++ that was "broken" by their handling of the .pde/.ino files, and (2) when it gets suffiently stable, I want to be able to use their automated update capabilities.

Wackerbarth, 2015-09-25:
As to having our own hardware platform, that is mostly to get around a problem in the Marlin code. To gain greater actuator speed, it is spending far too long in the ISR. In order to get away with that, it has messed with the serial I/O handler. As such, Marlin will not work (at full speed) using the current Arduino cores. The cores that I provide are older ones that I would like to update. But that is another sub-project.

But, by providing our own hardware platform, we are able to "control" all of the code that goes into our firmware builds.  Again, another reason to support only one method of building.

clefranc, 2015-09-25:
@Wackerbarth
>to recommend to everyone

Which words didn't pass your filter?

Wackerbarth, 2015-09-25:
"tester"

closed 2016-05-04 by jbrazio