MarlinFirmware/MarlinDev #215: Not a real issue, but a question(s)

TheSnarfster, 2015-09-25:
Good day all!

Point me in the right direction if this post needs to go elsewhere.

Header Need another serial port.

I have a ton of old 4X20 VFD modules with the ability of either serial or parellel port comms. Now I've already added the RepRap discount LCD controller, and it works great (After I had to figure out where the code was). But, I also want to add the VFD display. Specificly to display the X,Y,Z data.

I kinda wanted to use 4X7 segment displays for each axis (Like the big CNC machines do), but I would have to design the 7 Seg hardware, mux/de-mux and serial interface first. Where-as the nice big VFD display would look cool and less design first.

In order to do this, I want to add another serial port on the arduino (Or more appropriately, the atmega2560). I've downloaded and installed Marlin 1.1 and with a bit of tweaking, the printer seems to be running fine.

Now looking at the code, I see the ability to add another serial for wifi, bluetooth ???. But, I need to specify which pins I will be using for simple serial communication for the VFD.

Arduino has a library called "SoftwareSerial" which will allow me to use ...... Pretty much any pins I want. (There are a few I can use on the atmega2560). But, I could have some issues.

1) SoftwareSerial looks like it uses an interrupt. Not sure if that would make a difference in the PD sampling of Marlin.
2) Could I use MarlinSerial.h (Etc) and just have another instance of that to use as a serial port. And if so, where can I define the pins to use.
3) Could I put it (The subroutines) in the LCD update call ???
4) or should I just mod Repieter host software ???

Any thoughts, Ideas, Comments, suggestions would be appreciated.

Cheers!

Roxy-3DPrintBoard, 2015-09-25:
> I have a ton of old 4X20 VFD modules with the ability of either serial or parellel port comms.

What is a ton?   Because if you donated a few of these to different people that do not have LCD Displays on all of their printers, the code would probably magically appear.

a4jp-com, 2015-10-22:
Did you figure out everything @TheSnarfster ?

closed 2016-04-18 by thinkyhead