MarlinFirmware/MarlinDev #216: PlatformIO fail when settings use U8glib

Nandox7, 2015-09-25:
Hey,

When trying to run platformio with settings that require the usage of the U8glib, for example the Full Graphics LCD controller, it fails to compile missing to find the U8glib.h file.

---
Marlin.ino:44:101: fatal error: U8glib.h: No such file or directory
compilation terminated.
scons: *** [.pioenvs/mega2560/src/tmp_ino_to.o] Error 1
---

I did tried to manually install the lib for platformio using: platformio lib install 9
(as seen here: https://libraries.io/platformio/U8glib)

It install correctly but still doesn't work.

Wackerbarth, 2015-09-25:
The start of PlatformIO support was entered at the request of one of our users who had promised to "flesh it out". However, he never provided any updates. Unless someone fixes it, we will soon remove it entirely.

Nandox7, 2015-09-25:
I can look into this then as I'd really like to avoid using the junky Arduino UI.

clefranc, 2015-09-25:
@thawkins https://github.com/MarlinFirmware/Marlin/issues/2472

thawkins, 2015-09-25:
Platformio lib install 7 

Works for me for a ramps card. 

Im not sure what extra is needed to "flesh it out", a number of folks over on printrbottalk are already using it to build marlin for printrboards. It works. Does it need any more. 

If its docs that are needed then i will try and write something  this week end.

Nandox7, 2015-09-25:
Got it.
With lib install 7 it works.

Maybe just add that to the docs/readme. As more like me may just find first
the lib install 9 and it will fail.

-Nando

thawkins, 2015-09-26:
I will start a wiki page on the release repo to document using platformio
with marlin.

Nandox7, 2015-09-26:
Perfect! If i can help in any way please let me know.

PS: I've been searching but haven't found answer for thi yet. Is it
possible to just compile one environment?
I just need the atmega2560 but ifi  do platformio run it will compile all
the ones defined. Solution so far isto edit the ini file and remove the
entries for the ones i dont want.

thawkins, 2015-09-27:
Platformio run -e mega2560

thawkins, 2015-09-27:
http://docs.platformio.org/en/latest/userguide/cmd_run.html

thinkyhead, 2016-04-18:
Sorry this issue is lingering for so long. As soon as 1.1.0 is released we will be migrating code into the new file layout (as in the MarlinKimbra fork), and we'll be paying much more attention to issues in MarlinDev to see what still applies with the latest code.

closed 2016-05-25 by boelle