MarlinFirmware/MarlinDev #217: Marlin is seriously screwed up – please rescue it!

thinkyhead, 2015-09-28:
Congratulations on making the codebase worse for everyone.

1. The new "astyle" formatting has made the code completely human-unreadable, destroyed the careful indentation of `#if` blocks that I spent months improving, stripped out the lines and whitespace that made the code easier to understand and to work with. I can no longer make out the cascade of `#if` blocks at a glance, and due to the screwed-up indentation the code's cascade is now completely un-parseable by naïve text editors, so we cannot use code-folding at all. In case you didn't notice, this codebase uses `#if` blocks _everywhere_, and it is _vital_ that the cascade be easy to see! In short, this auto-formatting has made the code far worse, just as I explained that it would! The "astyle" formatting has also screwed up Doxygen-style comments. All in all the code looks, basically, just stupid. Of course the contributor who pushed this change through doesn't even write code, so why should he care?
2. In the middle of the release cycle someone decided it was a brilliant idea to completely alter the layout of the files in the `MarlinDev/master` branch, the tip of the tree. During a release candidate evaluation, the `master` branch should keep parity with the RC branch and continue to be used for "current work." Instead an RC-Bugfix branch was added to the Marlin (not MarlinDev??), so -um- developers are apparently expected to work on bug fixes there. This is completely, utterly, and ridiculously the wrong approach, and along with the mass changes in formatting, this only makes it harder to work on this codebase!

I feel that these changes have been rushed through without due consideration for the community and without enough discussion and careful thought. I completely agree with @eboston that this RC was rushed out in a _fit of pique_, that the groundwork which has been laid down in cleaning up the codebase for almost a year has been completely trashed, and that a new tyrannical system is being impatiently and inconsiderately applied by certain contributors, screwing up the codebase, the coherence of the project, and the morale of contributors.

I am honestly horrified by what I am seeing, and as the most active and invested contributor to this project, I feel personally affronted by it. It feels like the ghost of Kim Jong Il has possessed the project. As I have stated before, I'm not going to be able to work on this codebase in its current form, or under these conditions. My head hurts from face-palming and head-desking every time I look at the code. Seriously, someone who has a clue needs to kick out this new regime and get the project back on track.

@Roxy-3DPrintBoard ? @daid ? @ErikZalm ? Are you really impressed by this RC procedure over the last couple of weeks? Do you like the way the code looks now? Are you happy with the way that contributors such as @eboston and others are being treated by our new Glorious Leader? Please, someone, save Marlin!

emartinez167, 2015-09-28:
Scotty,

I have been watching, and from comments from a few people, it looks like
nobody was consulted, and the decision was made by apparently only one
person. What happened to the way changes used to be treated? This is just
thrashing the good work by a lot of people. and it feels like a tyrannical
way of treating contributors. I am sorry if I missed the announcement of a
new overlord but his is just a poor way of treating a great bunch of people
who were only trying to make things better. And I am just repressing myself
from saying things as I feel them, out of respect for pretty much everybody
else.

GIVE US OUR MARLIN BACK!!!!!

-- 

"This e-mail and any attachments thereto, are intended only for use by the
addressee(s) named herein and may contain confidential or privileged
information. If you are not the intended recipient of this e-mail, you are
hereby notified that any dissemination, distribution or copying which
amounts to misappropriation of this e-mail and any attachments thereto, is
strictly prohibited. If you have received this e-mail in error, please
immediately notify me and permanently delete the original and any copy of
any e-mail and any printout thereof."
P Please consider the environment before printing this email

Wackerbarth, 2015-09-28:
> On Sep 28, 2015, at 6:18 AM, Scott Lahteine <notifications@github.com> wrote:

> Congratulations on making the codebase worse for everyone.
> 
> 	• The new "astyle" formatting 

> In case you didn't notice, this codebase uses #if blocks everywhere

And that is just one of the things that makes the code almost impossible to follow.

> 	• In the middle of the release cycle someone decided it was a brilliant idea to completely alter the layout of the files in the MarlinDev/master branch, the tip of the tree.

That was not my decision. I had suggested that it be done at the start of the 1.2 series. And it will not be difficult to “roll back” the formatting of the RC branch.

> The master branch should keep parity with the RC branch and continue to be used for "current work.”

No! That assumes that this “current work” will FOREVER be an extension of the current methodology.

You are placing far too much importance on the NAMES of branches. And since both Marlin and MarlinDev are really the same base repository, we can publish any branch in either, or both, repositories.  I am surprised that you find it difficult to operate out of multiple repositories. Perhaps you need to increase your understanding of “git”.

The entire purpose in creating the MarlinDev repository was to separate the discussions concerning the problems in the 1.1 branch from changes being proposed for 1.2 (and later). In particular, the desire was to remove the “customer service” issues from the “design discussion and experimental features"
You were a part of that decision. So, don’t complain about it happening. We haven’t even given it a real chance.

IMHO, we do need to “break” from the 1.1 development stream. The code base has grown to a point that it is unmanageable and, as someone else commented, “we are spinning our wheels” because of it. As the code currently exists, it is virtually impossible for more than one person to work on it at a time (other than minor patches). You seem 

> that this RC was rushed out

No, it wasn’t “rushed out”. It was done without your “stamp of approval” because YOU refused to participate in a community decision. For quite some time, it was requested that you stop adding “new” features, and release “something, even if it is not perfect”. Instead, you kept adding more and more and absented yourself from, and blocked, anything that wasn’t on your agenda.

> Seriously, someone who has a clue needs to kick out this new regime and get the project back on track.

As much as you criticize me, you seem to have adopted the attitude of a dictator. I understand that you want to retain all of the power.
But don’t claim to represent “the community”. For all of the individuals you mention, I can point to others who are equally “fed up” with your actions.

> @Roxy-3DPrintBoard ? @daid ? @ErikZalm ? Are you really impressed by this RC procedure over the last couple of weeks? Do you like the way the code looks now? Are you happy with the way that [...]

Save Czar Scott?

daid, 2015-09-28:
I'm only commenting on "consulting bases". So the code is to modify as the maintainers wish.

However, taking a quick glance at it, I agree with @thinkyhead here. The new styling is a horrible to read. It was not easy to read before. But now it's just worse. And astyle is known to mess up styling completely with ifdefs (did not look for these details. But I have seen it happening before)

Other things, I think the move to a library is bad, as it makes it harder for people to compile it themselves. It's also horribly documented, as the documentation assumes you know a LOT of stuff. Making the entry bar really high.

clefranc, 2015-09-28:
The king is dead, long live the king!

@Wackerbarth I wish you good luck with YOUR new Marlin, hope you'll find followers meeting YOUR specifications. For the rest of us, we can always go fork ourselves.

Wackerbarth, 2015-09-28:
@daid -- "he move to a library is bad" ... "horribly documented"

Just like all the other changes that have happened over the past months. :)

I agree that the documentation is missing. However, except for the past few days when documentation after documentation was forcibly moved, Scott seemed to be the only one who had access rights to the documentation.

But since you cannot ever catch a moving target, I have been trying to get the file reorganization cleaned up so that others could resume adding features.  Unless the idea is completely abandoned, the documentation will follow.

Moving to a library format is NECESSARY if you ever want to stop being constrained by coding styles that were abandoned decades ago. Those styles were "OK" when the code was originated because it was MUCH smaller. But as more and more features keep getting tacked on, and more people want to work on additions at the same time, we really need a different approach.

The use of astyle will be less problematic as the overall code base gets transformed. However, from decades of experience, I really think having SOME automated styling program really helps in the long run.  Without ABSOLUTE style compliance, automatically enforced, a lot of time is spent attempting to merge multiple changes that were developed in parallel.

emartinez167, 2015-09-28:
Well said Christian... Let's go fork ourselves.

Regards,

Ernesto

daid, 2015-09-28:
From decades of experience from my side, code style hardly matters shit. People will produce horrible code with perfect style, and people will produce the most wonderful code with bad styling. Saying "automatic code styling will IMPROVE THE WORLD!" is short sighted and wrong IMHO. (I do love python tough, with the "you do proper indenting or your code won't work")

Doing file moves and code styles on a moving and developing project also introduced another issue. "git blame" is completely broken now. Making it impossible to track what commit introduced certain features.

(Once again, you do as you wish. I maintain my own fork, which is years older then the current Marlin)

daid, 2015-09-28:
Also, no matter what you say, with the current #ifdefs and the new code style, it's impossible to read certain parts of the code and follow the logic flow.

clefranc, 2015-09-28:
> with the current #ifdefs and the new code style, it's impossible to read certain parts of the code and follow the logic flow.

It's because Marlin have no logic YET!

Wackerbarth, 2015-09-28:
"with the current #ifdefs and the new code style" -- I don't disagree. As I said, it was not MY decision to introduce it to the RC branch. I do expect various developers to rewrite significant sections of the code in order to eliminate some of the current difficulties (particularly with "bed leveling" and the related probing)
Perhaps we can come up with some scheme that will allow us to partition the files into "legacy" (without styling enforced) and "nuevo" (with automated styling). Then, over time, everything can get converted.

Wackerbarth, 2015-09-28:
@clefranc -- "because Marlin have no logic YET!"  And if Scott has his way, it NEVER will! At some time, you have to decide to make sweeping changes so you are not stuck back in the last century.

daid, 2015-09-28:
But getting proper code structure requires a lot of work, and a strong leader that can spend a lot of time on said code. So ether commit to it, and do the shitload of rework. Or keep the old structure. Don't do some restructuring and tell everyone else to clean up the mess you made.

"I do expect various developers to rewrite significant sections of the code"
Why would they? Which incentive do you offer them?

Roxy-3DPrintBoard, 2015-09-28:
It is clear this 'Release Candidate' does not have the required consensus to be promoted.  I think we should roll back and start fresh.    We do need to be moving forward on a 'Release Candidate'.  But it needs to be something that does not have controversy and that everybody can feel good about it being an improvement.

I am in favor of making the changes that allow us to start breaking out parts of the code into separate modules.   But that is with the understanding that any change to the source code in those separated areas would automatically get rebuilt and linked in without the user jumping through hoops.   

Perhaps we should close this thread and start a new discussion about 'What and When' is going to be the next attempt at a 'Release Candidate' ???

Wackerbarth, 2015-09-28:
"Why would they?" -- Because some parts of the code really "suck" and they have expressed a desire to work on PARTS of the code. I don't think anyone (except perhaps Scott) wants to do it by them self. We want to have an environment where we can all, reasonably, work on some part of the whole which will then be brought into the finished product without it having to be redone continually.

What we have now is like a kitchen with one stove. One cook can prepare some, but at only a limited rate, product. In contrast, a commercial kitchen can combine the work of many cooks and produce more, and often better, product because each cook specializes on only a portion of the whole.

clefranc, 2015-09-28:
As pointed out by @daid, the problem I see here is leadership. The biggest qualities of great leaders are their empathy and respect. Something apparently @Wackerbarth have total lack of. You don't impose your vision, you let others embrace it. It's not just the way the 'new system' was presented that bothers me, this is an Open Source community, not some big corporate project with strict deadlines and budgets.

All in all, I (too) think the new Marlin ruler should have done the work in its own fork, then present the result to collaborators so we can discuss it. Instead, we got dictatorship with no respect on the considerable work accomplished during the past year. I still believe user experience (compilation, usage and support) is more important than coding style or libraries. But one does not preclude the other. It just needs more planning.

Stop this nonsense before it became a one man band.

daid, 2015-09-28:
@clefranc It's not just that it needs a strong leader. It needs strong examples. Saying that the current auto-leveling&probing code is a mess does not solve anything unless you structure it better yourself, or provide a good idea on how to re-structure it. Just saying that it's a mess does not fix it, saying that it's a mess anyhow and then auto-format it also does not fix it.

Saying (or doing) a split in better functions, better seperation of concerns, better use if pre-compiler&optimizer without making tons of IFDEFs. (I don't know the details of the code. So there are just general ideas, you have to be more specific then this)

And IMHO, order of importance:
1) Stability (A printer that crashes is the worst experience ever)
2) Responsive user experience (this means printing without stuttering in our case. Not high quality UI)
3) Cross hardware compatibility (We got tons and tons of users, with lots and lots of different hardware setups. This includes ease of compiling for those different setups)
4) Readability of the code.

Roxy-3DPrintBoard, 2015-09-28:
>  Saying that the current auto-leveling&probing code is a mess does not solve anything unless you structure it better yourself, or provide a good idea on how to re-structure it. Just saying that it's a mess does not fix it, saying that it's a mess anyhow and then auto-format it also does not fix it.

To be fair...   That is a big part of why Wackerbarth is trying to get us to a 'library module' structure.  RIght now, a number of people have tried to jump in and clean up some of the G29 mess.  Nobody has made progress because things stop compiling and linking.

BuzzBumbleBee, 2015-09-29:
Hello all,

I am a bit of an outsider here (followed marlin for a quite a while, and have been trying to contribute to the project where I can)

> Saying that the current auto-leveling&probing code is a mess does not solve anything unless you structure it better yourself, or provide a good idea on how to re-structure it. Just saying that it's a mess does not fix it, saying that it's a mess anyhow and then auto-format it also does not fix it.

I have been trying to come up with fixes to the auto level problems, the issue I had was never knowing what the exact expected behavior was. (At some points it was working 100% to what I expected but others where filing issues saying that it was broken in some way). Maybe we need better documentation on whats expected within this sub system

> All in all, I (too) think the new Marlin ruler should have done the work in its own fork

Iv worked in many OSS projects and non of them had a "ruler" as such, there were always good talks between devs (regardless of if they have pushed 10000 commits or 1 commit to the system they should all be treated the same) commits where pushed when multiple devs agreed that they are needed / clean / safe

>That is a big part of why Wackerbarth is trying to get us to a 'library module' structure. RIght now, a number of people have tried to jump in and clean up some of the G29 mess. Nobody has made progress because things stop compiling and linking.

The library layout will help for code style and making marlin more maintainable (split the code up a little), that being said you cant achieve what this needs in a day, i proposed a less heavy handed change where I added a marlinlib repo (for now just moving the configs over). This would allow a gradual move to the library setup. see

https://github.com/MarlinFirmware/MarlinDev/issues/212

daid, 2015-09-29:
Don't think moving the config files to a lib is the best move. It's the first thing that people want to start with hacking Marlin will start to modify. And in a zip-lib it's one of the hardest things to modify...

BuzzBumbleBee, 2015-09-29:
@daid indeed its not a "nice" solution, however issue is with how the IDE handles includes in project vs library. In a library you can have includes under multiple folder layers, so you can then split machine configurations down. This would result in the users needing to have say a 10 line config over needing a full config per machine. This would also mean that patches submitted to fix delta core components would only be needed on one file.

Wackerbarth, 2015-09-29:
@BuzzBumbleBee -- You are "spot on" about the reasons for the change. I had hoped that you would have found the time to provide some feedback over the past weekend. Perhaps you can do so soon.

@daid -- "config files to a lib" ... "the first thing that people want to start with" 
I fully agree that the configs ARE the access point for everything.  But look at it differently.
Consider that the entirety of Marlin is just one big lib. Libraries have their configurations also. I assume that you have written C or C++ "programs" where main() is just a trivial call to the real implementation. (In fact, that's the way the Arduino  cores are implemented)  Now, because the Arduino IDE is so "helpful", we lose the ability to organize our C source files, as we have done for decades, when they are in a "sketch".

But by making a trivial "sketch" that calls a "library", and putting ALL of the files in the library packaging format, we can regain the functions that are normally expected in C programs. As I have suggested previously, that a look at grbl, among others. That is the structure.

"And in a zip-lib it's one of the hardest things to modify" -- who said anything about how the source is distributed? It appears that many users get their source of Marlin in a zip file anyway.

Please take a look at https://github.com/Wackerbarth/MarlinDev/tree/Configs  There I have a STRUCTURE that can be refined incrementally. It supports the existing configuration files (with only a trivial change to them) as well as the kind of structure that we envision.

To flesh it out, we will need to identify common subsets of the configuration and more them to a separate, shared file. (Much as Configuration_adv.h currently treats some of the configuration elements.)

As that refactoring occurs, the number of duplicated lines will decrease. Although the "ideal" of defining each thing in only one place is, perhaps, too extreme, moving toward that concept is certainly much better than the current scheme that is the diametrically opposite approach. The important thing is that the structure being advocated does not dictate the contents. Rather, it enables the ability for the project to choose the details of the partitioning.

BuzzBumbleBee, 2015-09-29:
@Wackerbarth 

> You are "spot on" about the reasons for the change. I had hoped that you would have found the time to provide some feedback over the past weekend. Perhaps you can do so soon.

Iv been looking over the changes, but seeing as everything is "up in the air" with how people are receiving the changes its hard to know what to do / recommend. I would say that taking a staged approach to moving things over to the library model would be best, this being said i totally understand the argument that marlin needs to be super easy to build and configure (although the library system is by no means rocket science, it does add a totally new level of complexity to installing and setting up marlin)

Its a painful situation, either stay with the current model : 
+ Easy to build
+ 1 repo / branch for all files
- Needs massive config files for all machines
- Can not become modeler

Move to library hybrid : 
+ Can make configs more maintainable through inherited files
+ Can split subsystems into their own directories and overall clean up the code
- Its more complex to setup that just 1 ino file
- Users would have to keep both the ino and library in sync

Wackerbarth, 2015-09-29:
Is it REALLY more complex? I contend that the build complexity is equal it's just that the files are located in a slightly different place. There is still only one ino file and, next to it, a file specifying the configuration to access.  Further, it provides a mechanism for users to keep their personal configurations (emphasis plural) in the tree without having to swap files around or directly edit any of the distribution files.

Further, as you have recognized, by using the IDE distribution system, we will be able to move toward an even easier way for users to fetch and update the code.

BTW, I just got a test version of the Arduino IDE (1.6.6 -- pull request) made available that allows the user to add their printers to a menu under tools.

Build successful. Please test this code using one of the following:
http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-3878-BUILD-405-linux32.tar.xz
http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-3878-BUILD-405-linux64.tar.xz
http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-3878-BUILD-405-windows.zip
http://downloads.arduino.cc/javaide/pull_requests/arduino-PR-3878-BUILD-405-macosx.zip

Use that with the [IDE_1.6.6 branch](https://github.com/Wackerbarth/MarlinDev/tree/IDE_1.6.6) for a platform  (Not yet properly arranged. But you will find the pieces that you need to install a platform)
and the [Configs branch](https://github.com/Wackerbarth/MarlinDev/tree/Configs).

That will provide a menu with a number of our test configurations.

You can add a Boards.local.txt with similar entries to access a configuration that you add to the configurations/custom.

shaggythesheep, 2015-09-29:
If you took advantage of the Arduino Library Manager could you not potentially make installation even easier... Users could search Marlin, select the version then want and download it all within the IDE.

Then have some presets utilising the examples folder so that novice users can get their printer up and running without writing a single line of code!

Just a thought...

Wackerbarth, 2015-09-29:
@shaggythesheep -- That is EXACTLY what I am trying to accomplish.
Unfortunately, some around here seem to think that you have to completely implement everything before they can understand the destination.

daid, 2015-09-29:
@Wackerbarth if you really want to know if it's more complex, then have someone with no experience in this stuff try both methods. Quite some R&D people at Ultimaker know how to build our Marlin firmware. But I'm quite sure they will get confused by this new method.

They work by:
* downloading the source zip
* extracting it somewhere
* opening Arduino environment
* Opening INO file in Arduino
* Open configuration.h tab, and modify it
* Compile&run

Wackerbarth, 2015-09-29:
And just what do you think I will be changing?

The ino file is still there, in the standard Arduino position as examples/MarlinFirmware/MarlinFirmware.ino

My 4th and 5th steps will be

```
• Open the Examples->marlin->MarlinFirmware menu in Arduino
• Open "FirmwareCustomization.h" tab and modify it.
```
and as @shaggythesheep mentions, we will soon be able to simplify the first three so that they will not even have to more than add a URL (or perhaps just click on an entry in the IDE)

justmyopinion, 2015-09-29:
@Wackerbarth 
This was obviously your first attempt to make some sort of userguidance, maybe you should spend some more time to make this guidance first.
I really cannot understand why this trial and error attempt for a new environment had to live in an open forum. This has extreme costs to the continuing development of Marlin which at the moment suffers from many problems to be called a stable release version.

Wackerbarth, 2015-09-29:
@justmyopinion --
1) The changes in MarlinDev do not really affect the 1.1.0 Release. Those issues are, intentionally, moved to Marlin. They can, and should, be addressed there. On the issue of the style change, I was instructed to apply it just prior to RC1. Because of the blowback, I have offered an RC2 with those changes reversed. Although it has been available on my github repository, and the developers informed of it,  for more than a day, I am awaiting approval to "publish" it on Marlin. I presume that the delay is to permit some of those who are complaining to actually look at the proposed material before it is posted.
If they will, I have my doubts.

2) I have been, for some time, advocating three significant changes. Each of them, by their very nature, need to be made during a "factory shutdown" because, if everyone needs to make the shift at the same time. To do otherwise causes a MASSIVE duplicative effort in an attempt to catch a moving target. Therefore, the transition from 1.1 to 1.2 is an appropriate time for such changes.

Things have been rearranged, but the files that contain the logic are still present, and should compile to the same result. These changes provide the altered infrastructure to permit us to MIGRATE to a better organized, and maintainable, system.

In an effort to minimize the shutdown, I had done the minimal amount necessary to allow parts of the source tree to be opened for continued work.

It is presumed that "developers" are more knowledgable about code and their coding environment. Thus, they should be expected to be able to adjust to a rearrangement of the files without having all of the "polish" that we will need before 1.2 is made available to common users.

Roxy-3DPrintBoard, 2015-09-29:
> Wackerbarth commented 4 hours ago
> @shaggythesheep -- That is EXACTLY what I am trying to accomplish.

You did not need the next sentence...

closed 2015-10-04 by Wackerbarth