MarlinFirmware/MarlinDev #219: 1.1 RC2 -- When, and what?

Wackerbarth, 2015-09-28:
So far, we've had only a few fixes and some cosmetic changes to RC1.
If we want to roll back the formatting, it would not be difficult.

So, should we immediately have RC2 and plan for RC3 in the near future?

However, we really need to get most of the "players" to support getting a release shipped.
An important part of that effort is to just concentrate on "what's essential?" and setting deadlines and responsibilities to make it happen.

Roxy-3DPrintBoard, 2015-09-28:
> So, should we immediately have RC2 and plan for RC3 in the near future?

I would like to hear a little discussion about moving towards a library structure.   I think I know the good side.   I would like to hear all the negative stuff and see if we can agree that it is still worth move.

We do need to get a 'Release Candidate' done.   But let's get everybody's concerns factored in to it.

Wackerbarth, 2015-09-28:
The "library structure" has nothing to do with the 1.1.0 Release. It is, and only would be, in the 1.2 (dev) future development.

BTW, I just pushed an "RC2" branch to my repo  https://github.com/Wackerbarth/MarlinDev/tree/RC2.

If that's the way we want to go, I think that it might be ready. Please take a look at it and make any suggestions.

Roxy-3DPrintBoard, 2015-09-28:
Collaborators:   Please weigh in on any concerns you have about his branch:

 https://github.com/Wackerbarth/MarlinDev/tree/RC2

Or for that matter, any other concerns you have for 'Release Candidate # 2'.

thinkyhead, 2015-09-29:
Try and tag the version correctly. `1.1.0-RC1`, `1.1.0-RC2` with the `.0` and the `-` (dash) included.

Wackerbarth, 2015-09-29:
Did you even look at the code? I already did the rollback that you are suggesting before I posted it this morning.
```
....
        #endif
      }
    #endif

    // Home Z last if homing towards the bed
    #if Z_HOME_DIR < 0

      if (home_all_axis || homeZ) {

        #if ENABLED(Z_SAFE_HOMING)

          #if ENABLED(DEBUG_LEVELING_FEATURE)
            if (marlin_debug_flags & DEBUG_LEVELING) {
              SERIAL_ECHOLNPGM("> Z_SAFE_HOMING >>>");
            }
          #endif
...
```

Sniffle, 2015-09-29:
From the outside looking in you two look like 2 year olds bickering about
who can piss in each others cherios more...

Marlin needs both of you not just one... It needs sweeping changes with
restraint... It needs an RC NOW not 6 months from now which means code
freeze that didnt happen... It needs you to put aside different
methodologies and experience and find a common ground to work from...
Otherwise all you have done is come in ruin peoples trust in marlin and
looked like a couple idiots and that helps the community about as much as
pissing in cherios in the first place...

And dont come back with some well he started it bs or some other excuse
because its just childish... Just find a common ground and move forward
leave past where it is... In the past

Roxy-3DPrintBoard, 2015-09-29:
Thank You Sniffle!    But the honest truth is I meant it when I said:

~~~With that said, as a community we did a bad job of building consensus and factoring in everybody's concerns. 
...  
We need to ALL work better together to make sure we are doing what is in Marlin's best interest.
~~~

We are starting fresh...   Right now...

AnHardt, 2015-09-30:
@Wackerbarth 
Go for it. The readability is awful with the astyled sources. Not indenting after #if is the killer for me.

AnHardt, 2015-09-30:
Wrong button.

Roxy-3DPrintBoard, 2015-09-30:
I'm bringing up my new Delta Printer kit using the current RC.  So far, so good.   I'm fighting a number of issues because I've never had a Delta before.   I'm having some issues with the AllenKey mechanism because the source code doesn't envision that some AllenKey's are deployed manually.   I'm trying to be clean about how I handle that so it can be pushed back for other people.

clefranc, 2015-09-30:
@Roxy-3DPrintBoard Can you start a new issue, so we can discuss about it? :-)

BTW, welcome to the deltas world...

closed 2015-10-04 by Wackerbarth