MarlinFirmware/MarlinDev #220: M502 does not restore all settings, only some.

drbogger, 2015-09-28:
When doing a M502, only certain settings are restored, not all of them like it should.
Same thing happens when doing it from the LCD Menu.

fmalpartida, 2015-09-28:
Confirmed behavior. This should be fixed on RC2 looking on to building 1.1.0.

clefranc, 2015-09-28:
@drbogger Can you tell which settings are not restored?

drbogger, 2015-09-28:
I have a print going at the moment, so I'll have to wait and check later.

drbogger, 2015-09-30:
That's interesting... I check now for which settings do not restore to factory defaults, and I can't find any. 

However, I did find that the settings for "AutoTemp" do not SAVE into EEPROM.

Wurstnase, 2015-09-30:
> However, I did find that the settings for "AutoTemp" do not SAVE into EEPROM.

Which setting do you mean?

drbogger, 2015-09-30:
Any of the settings for "AutoTemp" do not save to EEPROM.

Wurstnase, 2015-09-30:
What did you mean exactly with 'AutoTemp'. I don't know which settings do you mean. The PID-values?

drbogger, 2015-09-30:
No, not PID. The AutoTemp settings in the LCD Menu.

From the Code:
```
/**
 * Automatic Temperature:
 * The hotend target temperature is calculated by all the buffered lines of gcode.
 * The maximum buffered steps/sec of the extruder motor is called "se".
 * Start autotemp mode with M109 S<mintemp> B<maxtemp> F<factor>
 * The target temperature is set to mintemp+factor*se[steps/sec] and is limited by
 * mintemp and maxtemp. Turn this off by excuting M109 without F*
 * Also, if the temperature is set to a value below mintemp, it will not be changed by autotemp.
 * On an Ultimaker, some initial testing worked with M109 S215 B260 F1 in the start.gcode
 */
#define AUTOTEMP
#if ENABLED(AUTOTEMP)
  #define AUTOTEMP_OLDWEIGHT 0.98
#endif
```

Wurstnase, 2015-09-30:
Ah ok, i've never used that feature.

drbogger, 2015-09-30:
Yeah, I was going to test it out, and it won't even keep the settings between resets.

clefranc, 2015-09-30:
@drbogger Are you sure this setting have been saved to EEPROM before? Can't find any code in EEPROM that support it.

> You enter the autotemp mode by a M109 S<mintemp> B<maxtemp> F<factor>
Turn this off by excuting M109 without F*

I've never used that feature too. How do you use it? In your slicer (is it supported natively, which ones)? Host? Manually?

Sorry for my ignorance, just want to understand if this can be included in EEPROM.

drbogger, 2015-09-30:
I am trying to Enable it via the LCD Menu. I've never used it before, so I'm trying to learn and see what it does exactly.

closed 2016-05-04 by jbrazio