MarlinFirmware/MarlinDev #223: AllenKey strategy going forward...

Roxy-3DPrintBoard, 2015-09-30:
I'm bringing up a new Delta G2s printer.  I've never had a Delta Printer before so this is quite a learning experience for me.  It is very possible some of what I'm thinking is not true and I just don't know enough yet.

But with that said, the biggest problems I'm having are related to the AllenKey on the G2s Printer.  It seems the deploy_z_probe() & stow_z_probe() routines pretty much assume you can swing the effector close to a tower and catch the AllenKey to deploy it.  And the stow function assumes you can drive the AllenKey into the bed to latch it and get it out of the way.

With my G2s, that is not the case.   I'm adding code to let me manually deploy it and stow it.   I happen to have an LCD Panel on that machines so I'm going to use lcd_clicked() to confirm the printer can continue after the manual operations are performed.  But I have to admit I'm kind of surprised nobody else has run into this yet.

And to my horror...  I was trying to see if my effector was positioning accurately.  Of course I try to run an M48 on it.   And it doesn't work???     And for that matter, the Grid Based Leveling options aren't really all there.   Guess who is going to be working on Delta issues soon.

Roxy-3DPrintBoard, 2015-10-01:
Update:   I ended up adding a few options to get things limping along.   I was originally thinking that when we have a better file structure these would all be buried in the Delta area.  But the more I think about it, these are all useful for Cartesian machines too.   So maybe they just get pushed down into the Auto Bed Leveling area???    Comments and direction will be appreciated:

#define Z_PROBE_ALLEN_KEY
#define MANUAL_ALLEN_KEY_DEPLOYMENT
#define IGNORE_STOWED_Z_PROBE

Wackerbarth, 2015-10-01:
I agree that there should be a common Leveling "area" that gets used by all machines.

But I would name things a little less specifically.

#define MANUAL_PROBE_DEPLOYMENT

For ANY probe, we can have a probe deployment and probe stow sequence.
EVERY deployment should start be following a path (perhaps of zero length)
Raise the printhead to provide clearance, etc.
Then there is a deployment action (For a servo, activate it. For an Allen Key, no action required. For manual, pause.)
Finally, we are ready to start watching the probe "switch" and follow the probing pattern.

At the end, there is a similar stow sequence.

I'm not sure what you are trying to accomplish by "#define IGNORE_STOWED_Z_PROBE".
Until it is starts deployment, and after it is finished probing, the probe should be ignored.

If the probe is using a Z-min "pin", I see no reason that we cannot have both "attached" to the same pin.

In the code, we just address it one way or the other depending on its current duty.

Roxy-3DPrintBoard, 2015-10-01:
> But I would name things a little less specifically.
> #define MANUAL_PROBE_DEPLOYMENT

Agreed.  I'll change it.

>I'm not sure what you are trying to accomplish by "#define IGNORE_STOWED_Z_PROBE".
>Until it is starts deployment, and after it is finished probing, the probe should be ignored.

Well...  That may be the intent, but that isn't what is happening.  If the probe is Triggered when it is stowed, it affects your ability to move the Z axis downward.   Just for kicks, take a piece of Scotch Tape and make your Z-Probe switch be triggered.   And see what happens when you try to home and move around.

So in Stepper.cpp I put this code:
~~~                step_events_completed = current_block->step_event_count;
            }
          #else // !Z_DUAL_ENDSTOPS
	    if (z_probe_is_stowed==false)
               UPDATE_ENDSTOP(Z, MIN);
          #endif // !Z_DUAL_ENDSTOPS
        #endif // Z_MIN_PIN

        #if ENABLED(Z_MIN_PROBE_ENDSTOP)
	  if (z_probe_is_stowed==false)
              UPDATE_ENDSTOP(Z, MIN_PROBE);

          if (z_probe_is_stowed==false && TEST_ENDSTOP(Z_MIN_PROBE))
          {
            endstops_trigsteps[Z_AXIS] = count_position[Z_AXIS];
            endstop_hit_bits |= BIT(Z_MIN_PROBE);
          }
        #endif
      }
      else { // z +direction~~~

From what you are saying, if this really is true, it is a bug???  Should we add it to the bug list if so?

Wackerbarth, 2015-10-01:
I consider it a "design flaw". But whether, or not, it is a "bug", I cannot say because there is no good specification of the behavior. Perhaps I don't understand just how you SHOULD be specifying that part of your configuration.

Roxy-3DPrintBoard, 2015-10-01:
Well...   It is rare, but some Z-Probes are in the Triggered state when they are stowed.   I think the AllenKey probes are more likely to behave that way because when you lower them they press the switch until something lifts them from the bottom side of the AllenKey.    I played with a million different combinations.  Everything from Dual Endstops, to Ignore Min Endstops, to what ever...  I could not get it to behave correctly.   

That was when I just said "I want it to work." and added that code.

Wackerbarth, 2015-10-01:
I would think that "ignore min end stops" SHOULD be appropriate to suppress the action that you are seeing.  The problem is that the current options are not "clear". Sometimes you have to set multiple ones, clear others which were set by default, stand on one leg, and whistle Dixie to get the correct combination compiled.

Wackerbarth, 2015-10-01:
"That was when I just said "I want it to work." and added that code." -- And thus Marlin gains yet another "wart".  Isn't it "fun" to try to work with something that has no design?

Roxy-3DPrintBoard, 2015-10-01:
> I would think that "ignore min end stops" SHOULD be appropriate to suppress the action that you are seeing. 

I had the wording wrong.   Turning on:

#define DISABLE_MIN_ENDSTOP

Does not produce the correct behavior.  Perhaps I should try eliminating my new #define and making the same changes using that #define.

Wackerbarth, 2015-10-01:
I found these lines
```
#define DISABLE_MIN_ENDSTOPS // Deltas only use min endstops for probing.
#define Z_MIN_PROBE_PIN ***
```
So, I would say that there is probably a bug somewhere.

I have an Allen key that is untriggered when stowed. So I don't see the problem on my machine.

clefranc, 2015-10-01:
@Roxy-3DPrintBoard Why are you using `#define Z_PROBE_ALLEN_KEY`? It looks to me that you have a manual deploy/stow probe like mine. I've also a probe that is triggered when stowed, all commands are doing fine with it. It's not like if you had the choice of endstop, for a delta, the `Z_MIN_PROBE_PIN` is the `Z_MIN_PIN`.

Roxy-3DPrintBoard, 2015-10-01:
OK???  Are you saying if I don't #define Z_PROBE_ALLEN_KEY and I just set #define Z_MIN_PIN I can get out of the ditch and back on the road?

And in case it doesn't make sense to you, the reason I did the #define Z_PROBE_ALLEN_KEY is because I ***DO** have an AllenKey probe!!!  :)

clefranc, 2015-10-01:
I think so. At least it's a good start to ignore the `Z_PROBE_ALLEN_KEY` in your case.

The `Z_PROBE_ALLEN_KEY` are just defining set of coordinates to move the probe around before and after ABL for deploy/stow. I admit that just because it's the shape of an allen key, and someone decided to call it this way in the firmware is confusing.

I agree with @Wackerbarth that a more *generic* use of the deploy/stow is needed (not the first time he mention it).

Just as an example, for me, the perfect deploy/stow will activate a servo on the side of the bed (not near the hotend) that will "show" a magnet to the probe for deployment at a specific coordinate. Do the ABL stuff, then "show" a raised pad to the probe for retraction. In the end, I need the behavior of both a servo probe and allen key. Weird isn't it?

I don't want to blur water further more, but you can take a look at my [config](https://github.com/clefranc/Marlin4Delta/tree/biv25_config). It's a lightly modified MarlinDev...

KiteLab, 2015-10-02:
Anyone tested the influence of `#define ENDSTOPS_ONLY_FOR_HOMING`?
For inductive and 'triggered when undeployed' probes this is mandatory - at least when connected to a z-min-endstop.
Maybe `enable_endstops(...)`  is missed somewhere or in the wrong order.

KiteLab, 2015-10-02:
If not superseded somewhere else this:
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/stepper.cpp#L1058
looks suspicious to me.

Roxy-3DPrintBoard, 2015-10-02:
I tried Clefranc's suggestion...   No Joy!  I have not modified the one KiteLab this is suspicious.  I'm kind of hesitant to change things like that without a better understanding.

clefranc, 2015-10-02:
@Roxy-3DPrintBoard Can you fork the repository you're using and create a branch with your complete configuration? It'll be easier for troubleshooting.

Roxy-3DPrintBoard, 2015-10-02:
Ah...  No...  Because I just grabbed a .ZIP file of the latest RC.   But I could ZIP that up and email it to you.

broncosis, 2015-10-04:
Funny Roxy I have been doing this on my prusa for a year or so now just with some creative gcode in my start file I had build a manual deploy  mag held probe  and its been working great 

http://www.thingiverse.com/thing:226440

but to get this all running on a sanginoloulo 1284  with a lcd and auto leveling 
I had to butcher much of the code scraping a bunch of menu functions and what not to make it fit 
with the gen7 boatloader  (don't ask its a long story)

shoot me a message if I can help test or something my code skills are very crude 
but I can muddle around when I need to

Roxy-3DPrintBoard, 2015-10-04:
Thanks Broncosis!   But I'll be OK.   It's mostly a question of whether the #define options are already in place to support this.  And if they are, we should have a discussion about why they are not named correctly and documented well so people can easily figure this out.

But right now, I'm going down a rabbit hole.   The Delta printers make U shaped curves when the   do_blocking_move_to() and do_blocking_move_relative() functions are called.  The further you move, the bigger the dip the nozzle does.   I think this is easy to fix with just a couple lines of code.   Because without a fix, nozzles are getting ground into the bed.

clefranc, 2015-10-04:
@Roxy-3DPrintBoard A LOT of people are using Marlin in their delta (even if I think it's not a very good firmware to use in deltas), so why do you need to fiddle with do_blocking_move_to() and do_blocking_move_relative()?

If you got U shaped curves when moving the nozzle around, it's likely your DELTA_RADIUS is not set properly.

> Increase value for dome shape and decrease for bowl shape nozzle travel.

Wackerbarth, 2015-10-04:
@clefranc -- There are two modes of movement. If, during printing, you are not getting flat (straight) movement, then, as you mention, the issue is that there is an error in the DELTA_RADIUS and/or DELTA_DIAGONAL_ROD lengths. However, during probing, the head is moved along a curved path to its target rather than attempting to force a straight line movement. It's like the difference in a G0 movement and a G1 movement.  To avoid a problem during those movements, you just need to allow sufficient clearance so that the "dip" doesn't cause the head to crash.

Roxy-3DPrintBoard, 2015-10-04:
> To avoid a problem during those movements, you just need to allow sufficient clearance so that the "dip" doesn't cause the head to crash.

Actually...  That doesn't always work.  You can set your clearance after a G28 as very generous and that will be OK.   But if you do a G29 and follow it with another G29 you are going to scrape the nozzle hard.  The reason is you are starting closer to the bed, but also G29 leaves the towers in a worst case position for the U shaped dip to happen.

So I'm looking at doing Plan-B.    I don't think I've noticed any recursion in Marlin yet.  If it is there, I don't remember seeing it.   But by just making do_blocking_move_to() check how far it is moving, and if it is too far splitting the move up into two smaller do_blocking_move_to()'s, I think the problem goes away.   And of course, each of those smaller do_blocking_moves_to()'s would be checking if they are too big, so we could recurse a few times.   I'll add up the numbers (and probably run the free_memory_watcher code) to make sure everything is OK, but winding up the stack 100 bytes to smooth out the moves would save a lot of effectors.

clefranc, 2015-10-04:
> However, during probing, the head is moved along a curved path to its target rather than attempting to force a straight line movement

@Wackerbarth What's the reasoning for not doing a straight line using delta kinematics?

Wackerbarth, 2015-10-04:
The moves are "straight" in the tower coordinates, which is a curve in the bed coordinates.
It is simpler and because it does not depend on the calibration matrix that is used during printing movements. (If your calibration is incorrect, you will get a curve anyway) If doing consecutive G29s is an issue, just increase the RAISE_BEFORE/AFTER value as well as the "between" clearance.

Roxy-3DPrintBoard, 2015-10-04:
Well...  there are a couple of issues here.  First, it is so simple to make sure people don't damage their printers, it seems like a couple lines of code is worth the effort.    But the other reason is I would have to set the RAISE_BEFORE/AFTER to such large values to be safe for the general case that I would rather not have it moving that far for no reason.  

I remember a couple of threads where people were complaining about the U-Shaped moves.  Now I understand why they didn't like them!

Wackerbarth, 2015-10-04:
Just what kind of movement are you doing? I have 
```
#define  Z_RAISE_BETWEEN_PROBINGS 6  // How much the Z axis will be raised when traveling from between next probing points.
#define  Z_RAISE_AFTER_PROBING 30    // How much the Z axis will be raised after the last probing point.
```
That provides plenty of clearance for the probe to go from one side of the bed to the other.

Roxy-3DPrintBoard, 2015-10-04:
I have:
```
#define Z_RAISE_BEFORE_HOMING 20      
#define Z_RAISE_BETWEEN_PROBINGS 15 
#define Z_RAISE_AFTER_PROBING 15   
'''

Wackerbarth, 2015-10-04:
And how much is that dipping? It seems as if you have some other setting that is incorrect.

Roxy-3DPrintBoard, 2015-10-04:
Well...  Have you gone the long distance from one side of the bed to the other?   It is probably dipping 30 mm.  Maybe more.   I'll measure it later and see.

Roxy-3DPrintBoard, 2015-10-04:
Ha!!!  A few lines of code and the U Shaped dip is gone!  :)    It turns out, even allowing each axis to move 50 mm without breaking the move up is just fine.   I hardly see any dip with that much movement.      

Now I need to fix the Z_RAISE_AFTER_PROBING that isn't happening.

clefranc, 2015-10-04:
@Roxy-3DPrintBoard @Wackerbarth 
Looking to remove the U shape movements in G29, can you try replacing the `prepare_move_raw()` with `prepare_move()`in the following code in Marlin_main.cpp:
```cpp
static void do_blocking_move_to(float x, float y, float z) {
  float oldFeedRate = feedrate;
#if ENABLED(DEBUG_LEVELING_FEATURE)
  if (marlin_debug_flags & DEBUG_LEVELING)
    print_xyz("do_blocking_move_to", x, y, z);
#endif
#if ENABLED(DELTA)
  feedrate = XY_TRAVEL_SPEED;
  destination[X_AXIS] = x;
  destination[Y_AXIS] = y;
  destination[Z_AXIS] = z;
  prepare_move_raw(); // this will also set_current_to_destination
  st_synchronize();
#else
  feedrate = homing_feedrate[Z_AXIS];
  current_position[Z_AXIS] = z;
  line_to_current_position();
  st_synchronize();
  feedrate = xy_travel_speed;
  current_position[X_AXIS] = x;
  current_position[Y_AXIS] = y;
  line_to_current_position();
  st_synchronize();
#endif
  feedrate = oldFeedRate;
}
```
https://github.com/MarlinFirmware/Marlin/issues/1731

Please report here. Thank you!

Roxy-3DPrintBoard, 2015-10-04:
OK!   But it will have to wait until tomorrow!

clefranc, 2015-10-05:
It was (a little) more complicated than I thought. Complete solution here: https://github.com/clefranc/Marlin4Delta/commit/c3bfd3c5589b06474bc187600ad1b3c6e7b07d68

Roxy-3DPrintBoard, 2015-10-05:
> It was (a little) more complicated than I thought. Complete solution here: clefranc@c3bfd3c

I have a question:  For a complete solution it was more complicated and you needed that extra logic in G29?    

Because I think what you are saying is if you are not doing a G29 and just using do_blocking_move_to() it is sufficient to do this change:

> Looking to remove the U shape movements in G29, can you try replacing the prepare_move_raw() with prepare_move()in the following code in Marlin_main.cpp:

clefranc, 2015-10-05:
> For a complete solution it was more complicated and you needed that extra logic in G29? 

Replacing `prepare_move_raw()` with `prepare_move()` in `do_blocking_move_to()` removed the U shaped moves, but I noticed something weird at the end of the probing, then confirmed with a G0 X0 Y0 Z0 that the nozzle was not leveled.

The `prepare_move()` is calling `prepare_move_delta()` that calls `adjust_delta()`. This is in this last call that offset is applied to `delta[]` when using bed leveling, and it should not during a G29. With `prepare_move_raw()`, there is no `adjust_delta()`, but produces U shape.

To bypass this adjustment in `prepare_move()`, for doing straight line during bed leveling, I've added a logic bool in G29.

It's crude, but it works. I'm sure you can provide a more elegant way to achieve this.

Hope this help.

clefranc, 2015-10-05:
Looking through G29, there is a `reset_bed_level()`, perhaps it's not used properly. Will look at this later...

Roxy-3DPrintBoard, 2015-10-05:
Thank You for the description.  With that explanation it is easier to understand what you did.  But the reason I was messing with the do_blocking_move_to() functions is I was making a simple routine that would probe the height of my 4 spring loaded set screws on the bed.  The purpose was to have exact measurements of how much any of them had to move to be at the same height as the other screws supporting the bed.   

I was using do_blocking_move_to() in order to accomplish that.    So, of course the G29 should be fixed.   But with what I understand right now, the do_blocking_move_to() functions should be fixed also.   And from what you are saying, in my case I don't want them applying any G29 corrections.  So I would need to be setting that flag during the operation of that function:
````
bool bed_leveling_in_progress = false;

AnHardt, 2015-10-05:
You are currently deep in the delta-ABL-code. Maybe you can help in https://github.com/MarlinFirmware/Marlin/issues/2665. Still cabling my delta.

clefranc, 2015-10-06:
@Roxy-3DPrintBoard I'm very interested in your manual bed leveling routine. I think it should be the next step after endstops calibration, followed by the delta radius adjustment.

Do you think a fully automatic delta radius calibration can be implemented?

Roxy-3DPrintBoard, 2015-10-06:
> Do you think a fully automatic delta radius calibration can be implemented?

I don't know...   I'm going to see if I can get myself through the process with just the code I'm writing.  That doesn't mean the code is good enough to handle everybody, but certainly, I should be able to get myself through the process.     Just to help with understanding what I'm thinking and doing, here is the comment block so far:

~~~~
// G27 3-Point Supported Bed Adjustment
//
// This function greatly helps in physically leveling a bed supported at its perimeter by 3 points.
// There are several modes of operation.  The function defaults to C-ircular if no other valid option
// is specified.  Suggested usage is to use the C option ignoring the outer circle numbers.  Then use
// the R option to get the DELTA_RADIUS closer and then fine tune the C option's outer circle numbers.
//
// C -	Circular operation.  It will probe the 3 specified points and report the Z height of each point.
// 	It will perform 3 circles.  The first one at full distance to the points.  The second circle
//	closer towards the center.  And the last circle at a distance halfway from the center to the specified
//	points.   The outer circle numbers are useful because they most closely measure the height of the bed
//	at the mounting points.  However, those numbers should not be as trusted as the inner numbers because
//	Delta printers have positioning inaccuracies that compound as they move further from center.   The
//	inner circles are more useful for accurate leveling.
//
// R -	Radial lines from the 3 mounting points through the origin and continueing to the far side of the bed.
// 	The is useful to see the slope of the bed straight away from the mounting point.  But it is especially
// 	valuable when trying to get the DELTA_RADIUS number set accurately.  
~~~~

Wackerbarth, 2015-10-06:
Given a “big enough” computer, YES, it can (with 11 total degrees of freedom).

However, I’m not sure that we have enough ram to make it work on an avr firmware.

The approach that I am trying is to use the printer to take a lot of measurements, reporting each back to the host. Then, I will do my computations on the host and use them to set the parameters for the kinematics.

I have found that using the kinds of sampling patterns that you describe is very useful in getting to an “initial guess”.

I also make the following observations:

If you truly have a slanted, as opposed to warped, bed, then there is NO value in correcting the probing position for the lateral offset of the probe from the symmetric center of the printhead. Any slant translate into a CONSTANT error in the print height (The offset * the slope in that direction) which is extremely minimal and can equally be a result of an incorrect estimate of the trigger height.

Therefore, I do my probing with the center of the printhead along the radial lines. It really simplifies the math.

The math is also more amenable to calculation if the samples are taken close to the expected x-y position and measure the corresponding z.
Therefore, I suggest that it would be appropriate to apply your current “best estimate” of the calibration parameters when taking the readings. If everything was perfect, all of the Z readings would be zero (well, the probe height). Then, any error is used to calculate the change in the coefficients rather than their direct values.

Roxy-3DPrintBoard, 2015-10-06:
> The approach that I am trying is to use the printer to take a lot of measurements, reporting each back to the host. Then, I will do my computations on the host and use them to set the parameters for the kinematics.

> I have found that using the kinds of sampling patterns that you describe is very useful in getting to an “initial guess”.

Another idea comes to mind...   Certain patterns are better for this kind of stuff because they cancel some errors and magnify other errors.    The firmware has access to the machines current configuration.  It would seem it should be able to adjust those parameters on the fly and do a brute force search looking for 'better numbers'.   It would be cool if you could just get some reasonable numbers into the machine and from there it starts probing for an hour to adjust the numbers.   (Almost for sure...  that is too much to hope for...  but it would be cool!)

Wackerbarth, 2015-10-06:
That's exactly what "simulated annealing" is all about. I'm just not sure if, with 11 degrees of freedom, we can hold all of the intermediate calculations  Because various parameters tend to "cancel each other", there may be difficulties in finding the direction in which the readings get better unless you can include enough points, spread around the bed, in the calculation.

thinkyhead, 2016-04-18:
@Roxy-3DPrintBoard Anything in this issue worth preserving, or should we close and revisit if it comes up again — after we migrate the 1.1.0 code over here?

closed 2016-05-27 by jbrazio