MarlinFirmware/MarlinDev #225: Help needed: Arduino compiler stuck with error blocking any further work

justmyopinion, 2015-10-01:
When I was issuing some performance changes with Marlin RC I got caught with this error and cannot figure out how I got there, some multiple declarations maybe. 
Any clue to solve this would be helpgull as I am stuck to this error:

:\Users\EO\AppData\Local\Temp\build2350060330530899328.tmp/core.a -LC:\Users\EO\AppData\Local\Temp\build2350060330530899328.tmp -lm 
C:\Users\EO\AppData\Local\Temp\build2350060330530899328.tmp/core.a(HardwareSerial0.cpp.o): In function `__vector_25':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial0.cpp:48: multiple definition of `__vector_25'
C:\Users\EO\AppData\Local\Temp\build2350060330530899328.tmp\MarlinSerial.cpp.o:C:\Users\EO\AppData\Local\Temp\build2350060330530899328.tmp/MarlinSerial.cpp:52: first defined here
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
collect2.exe: error: ld returned 1 exit status

Wurstnase, 2015-10-01:
http://forums.reprap.org/read.php?146,564899

Wackerbarth, 2015-10-01:
Or, alternately, use the Marlin version of the Arduino "core".

We really need to move to the current IDE. But, at the same time, Marlin has made some changes to the handling of Serial interrupts.  To work around that problem, we have a copy of the older, Marlin compatible, core in the Marlin platform support files.

Wurstnase, 2015-10-01:
arduino.org is not the 'current' IDE. The IDE is only 'current' on arduino.cc.

Wackerbarth, 2015-10-01:
True. Since I personally track, and contribute to, it directly on the github account, I tend to forget that the splintering of Arduino has led to that ambiguity.

Personally, I see no reason to use any version of the IDE older than 1 OCT 2015.
It has a lot of good features that are not present in the older versions and, given that we have a Marlin "platform", we can control the core, and, if necessary, even the compiler that will be used to do our builds.

I'm working on getting the support folded into the IDE's Boards Manager.

It's just a work-in-progress. So I have it over on my github account until it seems ready to be used.
I invite users to try it and provide me with some feedback.

Start by downloading the "hourly" version of the Arduino IDE 1.6.6
(Note: I make use of a feature that was added only this week).

Then visit     https://github.com/Wackerbarth/MarlinDev/tree/MarlinFirmware
for additional instructions.

justmyopinion, 2015-10-01:
Well it does not look to be my problem as I have tried both the version 17.7/1.65 and the problem persists due to some code changes that I cannot verify or roll back. I can compile earlier backups of marlin with no problem.
Actually I was working on a replacement for WIRE library when this happened. I was hoping similar problems has been met by other nerds.  :-)

Wackerbarth, 2015-10-01:
The problem is that Marlin code IS NOT compatible with the current Wiring, etc. cores.
You MUST use an older core.
```cores\arduino/HardwareSerial0.cpp``` and ```MarlinSerial.cpp```  are in conflict.

justmyopinion, 2015-10-01:
I have np problems with current release...
How can I get around with another core and is this part of new environment?

Wackerbarth, 2015-10-01:
No, it is not a part of the new environment changes. It all has to do with the above mentioned conflict in the sources.

What do you mean "I have no problems with current release..."?
To which toolset, AVR core, and Marlin sources do you refer? (Please be specific so that I might be able to reproduce your "result")

justmyopinion, 2015-10-01:
I downloaded Marlin RC1 a while ago and this has been the base for some optimization I have investigated, basically in LiquidCrystalTWI2 and Wire lib .
 My core base is arduino IDE ranging from 1.5-1.66- 1.77, currently "nightly version" from yesterday I assume is "1.66 dev". .
My codebase is Marlin RC 1, I assume changes are minor in RC2.
I have a less modified optimized version (based RC1) with LiqCry.. and Wire lib that compiles with any Arduino core and downloadable to printer with no known problems printing.
I also have a further modified version with a furher modified LiqCry... driver and wire Lib replaced with i2c lib without interrupts which was compileable until 24 hours ago and now i do not know the changes making this to happen, but it does not seem to be related to a specific core unless....

Wackerbarth, 2015-10-01:
As for the Marlin codebase, you are correct to note that RC1 and RC2 are little different, and those differences should not affect any of the areas that you mention.

However, there is an issue with "toolset" and "core".  I don't know the source of your IDE. However, the version numbers that you mention do not match those of the official IDE. (From arduino.cc).

And the IDE provides both the compiler toolset and the source for the "core" library (which includes HardwareSerial0.cpp as well as libc, etc.)  So, I suspect that something in that part was recently changed.

justmyopinion, 2015-10-01:
To avoid further confusion I will stick to arduino.cc ver 1.66dev but i am not convinced it has anything to do with coreversions sofar or even RC1.1 as i can use any coreversion on RC1.1 without these problems.
problem came after having changed my dev basecode without compiling for a while and now I cannot roll back without too much work.
problem is that I have a fairly modified RC1dev version that cannot get through compiler/linker at the moment and I need to rollback to a stable version or find the problem. 
24 hours ago I had a fairly stable modified RC1.1 version running without wire lib :-(.. and timing looked very promising.

justmyopinion, 2015-10-01:
What compiler tells me is that hardwareSerial0.cpp  core has this definition:
HardwareSerial0.cpp:48: multiple definition of__vector_25' 
and that MarlinSerial.cpp had the first definition of same ISR vector here:
//#elif defined(SIG_USART_RECV)
#if defined(M_USARTx_RX_vect)
// fixed by Mark Sproul this is on the 644/644p
//SIGNAL(SIG_USART_RECV)
SIGNAL(M_USARTx_RX_vect) {
  unsigned char c  =  M_UDRx;
  store_char(c);
}
#endif

I did not change anything in these files so how did this become a problem all of a sudden.
If I comment out the definition in marlinSerial. Marlin compiles fine but I have to live without a serial receiver then which of course is no solution, but points to the problem some how.

Wackerbarth, 2015-10-01:
What compiler tells me is that hardwareSerial0.cpp core has this definition:
...
and that MarlinSerial.cpp had the first definition of same ISR vector here:
...

And that is what I have been trying to tell you. The HardwareSerial0.cpp that you are trying to use is newer than the one which we use in the "core" when we compile with either an ancient IDE (1.0.x) or if you compile from a core taken from the marlin platform as defined in the support area.

I would like to clean up the issue, but, first, we have to wean the users off of the ancient IDE.

justmyopinion, 2015-10-01:
But how on earth can you explain to me i do not have this problem if I compile Marlin RC1.1 in same environment without any  or minor modifications to code.?
Thankyou for trying

Wackerbarth, 2015-10-01:
OK, since you are "downloading", can you supply the EXACT commits (The commit SHA for each) for both the version that was working and the one that is failing. That way, I can try it myself.

justmyopinion, 2015-10-01:
I am impressed by your generousity, but it has no meaning for you to try compiling "my code".
I have downloaded RC 1.1 and I assume there are not more than one version, but this is it!
This one is running fine despite from display update frequency which you have already responded in RC 1.2. In attempt to get better performance I tried to modify RC1.1 by replace wire lib etc..
So now i have two "RC1.1's versions"  One that is clean and one RC1.1_dev that has been modified on my PC only.
When i compile the clean RC1.1 no compileproblems, but when i compile the modified version I get the above compiler response which can only be surrounded by commenting out ISR vector code in Marlinserial and compile without that code piece, and that one is running but without serial RX..
So short: I have changed my RC1.1 code to something that cannot be compiled anymore and I cannot figure out how this was done, but tried to ask some experienced nerds  in this forum whether they have seen something similar, but if you have no experience of this behaveour you should not spend more time on this issue, I will follow up otherwise.

Roxy-3DPrintBoard, 2015-10-01:
> We really need to move to the current IDE. But, at the same time, Marlin has made some changes to the handling of Serial interrupts. To work around that problem, we have a copy of the older, Marlin compatible, core in the Marlin platform support files.

I'm very curious about this.  Do you know what has changed in the Serial Interrupts off of the top of your head?  Can you explain to us why it is causing these problems?

Wackerbarth, 2015-10-01:
The problem is that Marlin has replaced the serial driver with its own so that it can stay in the ISR for an excessive length of time. Some time ago, the Serial driver in Arduino was changed, but many of the Marlin folks stuck with the old IDE (1.0.5 or perhaps earlier) and the core distributed with it.

The change to the Serial driver is not compatible with the call from the Marlin ISR because they handle the I/O interrupts (particularly on AT90USB chips) differently.

If you go slow enough, you can work around the timing by dropping the special check in the ISR and run with the current I/O drivers. But that limits your top end stepping speed and you get, as some have encountered, either "stuttering" in the motion or communications errors.

Roxy-3DPrintBoard, 2015-10-01:
Why does the ISR want to stay active for so long?  What is it doing with the time it spends there?

But what I'm really thinking is, can we just replace (or override) that one function and then not care what core we are using?

Wackerbarth, 2015-10-01:
Well, there are two sides to the situation. (1) Marlin can quit pushing beyond the interrupt capability, or (2) you run with a core that isn't the one distributed as a part of the modern Arduino IDE.

"that one function" is a part of the core. Certainly, you can create a custom hybrid core that incorporates some of the other changes that have been made to the core, but you still has a core that is not "stock".

justmyopinion, 2015-10-01:
It is easy to understand that computers go crazy in  intercommunication when you can observe what is happening here.
Please understand I have NO REAL TIME PROBLEMS observed with current installation of serial drivers no matter which IDE  version used.
My problem is  a compiler problem that came to me during developing of some code improvements from one day to another. Please let not these problems interfere with your considerations of Arduino environments installed with Marlin. There is no connection to your worries and my problems at all (as far as I know). 
I may think that I may have introduced some sort of #include statement in any file of Marlin that cause compiler to observe that some basic ISR routine for serial code has been declared twice.which the basic settings of avr compiler does not allow.
Problem is that I have no experience to tell which editing caused this if any. As i told you above the not edited code is running fine and besides the serial ISR the edited code is also running with a wire lib that is not based on interrupt. So please gentlemen, do not worry if not necessary, it is probably my coding skills that lacks here to understand the problem and not some Marlin excercises in the past. 
Sorry tio disturb your circles, but we are at different planets at the moment. i worry of the quality of marlin coding but you seem to be elsewhere right now. i would be happy if this circus could have some ending soon..

Wackerbarth, 2015-10-01:
We (at least some of us) also worry about the quality of the Marlin code. Over time it has grown so many warts and grown in a manner that is not sustainable. As a result, it is very difficult to determine just what the code is supposed to do. This, in turn, leads to quality issues.  Hopefully we will soon have the ability to utilize more modern coding practices and thereby improve the quality of the code.

justmyopinion, 2015-10-03:
I have to capitulate on this one.
I am now convinced that it IS Marlin CORE environment causing this as it has proved that the only change I can boil it down to is that I replace wire libs whath a similar I2C lib without interrupt blocking and that it is this maneouvre causing that the linker finds two instances of serial ISR.
How this is explained,? well don't ask me I thought I removed a problem but got a new one.
I am afraid that it somehow has to do with the #include spaghetti found in Marlin, but I gave up on understanding the structure but it is a seriuos problem. despite from this ISR problem Wire to I2C replacement executes exellent and reliable and reduces the load from processor.

Wackerbarth, 2015-10-03:
@justmyopinion -- A few comments. First, I agree that "the #include spaghetti found in Marlin" is a major issue. But, I am finding it virtually impossible to do anything about those kinds of problems.  There is one faction that is happy with their IDE 1.0.5 builds and doesn't want ANYTHING changed. There is another faction that wants to use the IDE 1.6.5 "right out of the box".  So, we are stuck in the middle (and spinning our wheels). The old, 1.0.5-compatible, code doesn't work with the Arduino core from 1.6.5 and there is constant objection to making the changes to even attempt to rectify the situation.

When you speak of replacing the Wire library with another, it is likely that your replacement expects a more recent core and, without some additional work, will not "play well" with the MarlinSerial and ISR from our current build.

However, I cannot help you further if I cannot see the EXACT files that you are trying to compile. At this point, I really don't KNOW what you are doing. I can only GUESS at possible issues.

justmyopinion, 2015-10-03:
Well I think I have described problem detailed, could it be that your focus is otherwise at the moment.
Problem is very simple to establish and is caught by the linker not compiler (I think).
1: download Marlin RC 1.1 and make a configuration for my printer which includes installation of LiquidTWI2 lib for my display and this is working out of the box using TWI and Wire lib.
2. TEST1: install Wire alternative I2C lib which is not ISR dependatn like Wire.
3 change two (2) #include statements, 1 in Marlin.ino and one in ULTRALCD_IMPLEMENTATION_HITACHI_HD44780.H press compile and compilation/link fails with the above described message of 2 instants of serial ISR.
THATS IT. if ISR in marlinSerial line 42 is commented out it compiles but then you have no serial receiver :-(.

Wackerbarth, 2015-10-03:
Well, since you are unwilling to precisely describe the #includes that you changed, and more importantly, the changed libraries, and thus, the #includes that the inclusion of their headers might affect, there is nothing I can do.

The problem is not in the linker. It is just the component that is catching the incompatibility caused by the alterations to the set of source code that you are attempting to use.

justmyopinion, 2015-10-03:
I give up...may ne like the rest of the Marlin voluntaries.. Good luck with YOUR project

KiteLab, 2015-10-03:
Oui. Non. Oui. Non. Oui. Non. Arrrgggh.
Greetings from  Louis de Funès

Roxy-3DPrintBoard, 2015-10-03:
> Well, since you are unwilling to precisely describe the #includes that you changed, and more importantly, the changed libraries, and thus, the #includes that the inclusion of their headers might affect, there is nothing I can do.

I think it is more likely he doesn't fully understand the information you need in order to help work through the issue.   

> I give up...may ne like the rest of the Marlin voluntaries.. Good luck with YOUR project

Part of the problem is you don't have stuff setup as a fork of the MarlinDev code base.  I most certainly am not pointing any fingers because I operate the same way.   I grab a .ZIP file from GitHub and start modifying it.   The problem comes in when somebody here needs to look things over to help or exchange ideas.   GitHub is setup so that changes made by one person can easily be reviewed by somebody else and they can find the relevant changes quickly and easily.   Even if you made a .ZIP file of your stuff and emailed it to him, he would have a hard time figuring out what you changed because all of the source code control information has been lost.

@justmyopinion can you be talked into doing a 'fork' and making the same modification that are causing you trouble?  Because if you duplicate that code inside of the GitHub system you can just Wackerbarth a link and he will be able to very quickly tell you where the problem is and what is going sour.

Wackerbarth, 2015-10-03:
Roxy,
Although you are describing the expedient method of communicating the information, if he would provide the files in any machine-readable form, I might be able to help. But, there is no way for me to figure out whats going on when he only complains about the result and provide vague assertions about what he did to get there.

Please remember that, although you take some action with good intention, the result may be other than expected simply because, in the actual implementation, there is some, almost imperceivable, deviation from the intended.  Computers are very exacting. There are cases where a difference of only one character can completely change the result.

Roxy-3DPrintBoard, 2015-10-03:
> if he would provide the files in any machine-readable form...

@justmyopinion : Did you see that?  You don't even have to create a fork to get help.

justmyopinion, 2015-10-05:
An unnoticed call to Serial for error messaging in the added library was causing this and when removed, linking with no problem, This issue will be closed.

closed 2015-10-05 by justmyopinion