MarlinFirmware/MarlinDev #226: Configure from scratch

orangeudav, 2015-10-01:
Hi all,
need help with configure MarlinDev from scratch for RUMBA for any Delta. Any examples? Wiki? Where is Configuration.h? I have configured Marlin for RUMBA for Prusa i3...

Wackerbarth, 2015-10-01:
I suggest that you go to the MarlinFirmware/Marlin repository and utilize the RC branch. Hopefully it will soon become our latest release.

If you have your Prusa configuration, compare it to the delta/generic example configuration.

It should be pretty obvious which parameters apply to a general printer (use the values from your current experience) and which apply only to delta printers (you'll need to enter those settings)

thinkyhead, 2015-10-12:
@orangeudav The `configuration_examples` folder is a good start to getting a handle on Delta settings (and configuration in-general). The Kossel, Kossel Pro, and a "Generic" Delta are all included there. Nothing RUMBA-specific to worry about that I can think of…

closed 2016-05-04 by jbrazio