MarlinFirmware/MarlinDev #229: Try the new build

Wackerbarth, 2015-10-04:
After installing the styling changes provided by @thinkyhead, I think that we have a code base ready for trial. Please visit the [trial_dev] branch and try the new build procedure.

Until it stabilizes, the build process is hosted on my github account.

1) Download and install the Arduino IDE 1.6.6
2) Download the Marlin source as a library from the trial_dev branch here (== dev in my repository)
3) Get the URL to be entered in the Arduino Boards Manager from the MarlinFirmware branch of my repository.
4) Use the boards manager to download the support platform.
5) (Important) Select a board from the Marlin section of the available boards.
6) Select a configuration from the Boards->Printers menu
7) Open the .ino file  File->Examples->marlin->MarlinFirmware
8) Compile

This is currently a work-in-progress. I know that parts are incomplete. Your feedback, and assistance, would be appreciated.

9) Added feature -- you can add a "boards.local.txt" to the platform and place an entry for your configuration in it. Then you can select your printer as the configuration to build.
(Note: If you edit any of the platform files, you will need to exit and restart the IDE for the change to be recognized.)

Roxy-3DPrintBoard, 2015-10-04:
Just to be clear:   This is the path we need to go down to be able to break pieces of the code out into libraries, right?

Wackerbarth, 2015-10-05:
That is correct, at the present time. If, in the future, we find it advantageous, we can separate it into multiple libraries.

By breaking away from the "helpful" things that the Arduino IDE does for us, we are able to return to the traditional C/C++ structure of modular programming.

At the same time, we wil, in the future, also be able to leverage the new distribution mechanisms provided by the IDE and, if we organize our configuration protocol appropriately, provide almost "automatic" updates of user systems.

In that regard, it is my hope that  the various vendors will provide appropriate configuration files so that their customers can create custom builds by providing only a few specification lines.

Roxy-3DPrintBoard, 2015-10-05:
Can you put the M48 code into a separate library that gets linked in?   Because I know how to fix it so we can make a version of the command that works for Delta printers.

We need one small example of a piece of code being pulled out and turned into optional libraries.   If you get the Cartesian M48 turned into a library, I can copy that and make a Delta version of the library.

Wackerbarth, 2015-10-05:
@Roxy-3DPrintBoard -- I think that you may  be using the wrong terminology. A library would be something like the generic driver for certain LCD displays (U8glib) Often, a library is intended to be reused in multiple projects. An example of a part of the Marlin code that would be a candidate to be made into a library is the planner/stepper section. I could see it useful in, for example, a robot that is not driven by g-code, but some other language.

What I think that you are actually seeking is to have the M48 code split off into a separate module.

We already have some modules that are split into individual files, for example temperature.cpp. And a number of the coders feel that it would be an improvement to have even more or them. By doing so, it decreases the likelihood that one change will conflict with another because, if properly designed, actions are limited in their scope and only affect the rest of the system through their defined interface.

The major reason for changing the layout of the Marlin files is to allow us to utilize a structured file hierarchy rather than the flat layout that is imposed by the Arduino Sketch mechanism.

As an example, it would be logical to move all the "language" files into a folder and similarly, the "motherboard (pins)" files into another. Those files have no direct interaction with each other and by utilizing this feature of the file system, the files are better organized.

As to your specific mention of M48, certainly that action could be made into a separate module. However, I would suggest that you should, first, split out the probing action itself. Define a module that would handle the deployment, and stowing of a probe, and would take a probe measurement at some specified location (exactly how that location is specified is, at this time, undetermined). The significant aspect of the design is that this routine cares neither where, or why, the reading is being taken. We leave that up to the calling routine (such as M48, G29, or, even in some cases G28, etc. to control and utilize the readings.

However, the probing module would be totally responsible for the multiple methods which are used to deploy the probe. Also note that my description does not specify how the code within this module is organized to process those various variants. That would be left to the implementer. (And it could, at some later time, be changed without affecting the rest of the code base.

Roxy-3DPrintBoard, 2015-10-05:
Perhaps I'm using the wrong terminology.  But maybe I'm not fully understanding the mechanism to do this clean up.   I was thinking that in order to escape the assumptions that the Arduino IDE imposed on the code, we needed to compile the code as a library (or set of libraries) and then just have a very simple 1 line program that called into the library and kicked things off.

I thought part of the reason to do this was different areas could be pulled out of the main code and put into their own, stand alone library to help simplify things.   And these stand alone libraries could be built using code focused on the machine type.   For example, for auto bed leveling, we might end up with a Delta version, a Cartesian version, and a Mesh Leveling version, a Core-XY version.

Perhaps I'm not understanding the things well, but that is what I thought we were trying to accomplish.   And with that mindset, that is why I asked if you could bust out a simple command to be the first 'simple library' to be selectively linked with the monster main library.   If I had a sample 'simple library' linking in and doing things, I could make alternative versions of that 'simple library' to fix outstanding issues in the Delta area.

Wackerbarth, 2015-10-05:
> On Oct 5, 2015, at 9:51 AM, Roxy-3DPrintBoard <notifications@github.com> wrote:
> 
> Perhaps I'm using the wrong terminology. But maybe I'm not fully understanding the mechanism to do this clean up. I was thinking that in order to escape the assumptions that the Arduino IDE imposed on the code, we needed to compile the code as a library (or set of libraries) and then just have a very simple 1 line program that called into the library and kicked things off.
> 
That is correct.
> I thought part of the reason to do this was different areas could be pulled out of the main code and put into their own,
> 
also correct
> stand alone library to help simplify things. And these stand alone libraries could be built using code focused on the machine type. For example, for auto bed leveling, we might end up with a Delta version, a Cartesian version, and a Mesh Leveling version, a Core-XY version.
> 
What you seem to be missing is that the “simple library” doesn’t need to be a “library” itself, but just a module within a library.
The conversion is to move all of the code from a “sketch” format into a “library” format so that we can further break it down into modules in a structured manner.
> Perhaps I'm not understanding the things well, but that is what I thought we were trying to accomplish. And with that mindset, that is why I asked if you could bust out a simple command to be the [...]
Correct, you COULD do so. However, I feel that your example is a poor one. In the particular case of M48, if probing is appropriately broken out, then there will be a much smaller difference between the various versions of the command.

If you intend to break down a multi-variant collection, then you could choose to first break it according to any of the variables. However, it often makes more sense to prefer one characteristic over another because there is a greater commonality in that clustering.

Take as an absurd example that we have a collection of {textbooks, automobiles} {maroon, orange}.
Depending on your purpose, it might make better sense to first separate all of the textbooks from the automobiles because textbooks are more like other textbooks than they are like automobiles.  But, if we are looking for trucks to transport the items, it might make better sense to put the maroon textbooks in the same shipment with the maroon automobiles because they are all going to College Station rather than Austin.

The same idea applies to the lines of Marlin code. Here you are looking to split up the action routines for various G-codes which vary according to the particular kinematics.  I’m suggesting that you split off a particular kinematic procedure that could be used in multiple G-code actions rather than first splitting off a particular G-code.

If you split things in the wrong order, you will just end up duplicating functions because of the split. And, if split in another order, you may well find that you no longer really gain any clarity by further splitting of a particular item.

Roxy-3DPrintBoard, 2015-10-05:
OK...  I understand what you are saying.   But I need to see something...  Anything...  Split out so I can see how to turn on and off a given 'module'.

Wackerbarth, 2015-10-05:
It isn't necessarily a case of "turning on" a module. However, a "module" like the "pins" files might be simplified in the process of doing so.

The primary purpose in developing modules is to provide a structure to the interaction of various parts of the code.  At the present time, there is significant interaction with global variables and, in general, the code files have grown too big. This creates the "one stove" situation where coders are stumbling over each other and, also, introducing unintended consequences because of the way that the code is(n't) organized to provide well defined interfaces between sections of the code.

Roxy-3DPrintBoard, 2015-10-05:
Agreed...  But there is another issue.   Just to add the code to handle the manual deployment and retraction of my 'AllenKey' probe took several hours because I was nested 37 layers in on #ifdefs.   We need to pull out all the non-important #ifdef stuff from each module.  

As a first step...   What makes sense to me is close the Auto Bed Leveling Code 4 times.  There would be a Delta module, a Cartesian module, a Mesh Leveling module and a Core-XY module.  All of the #ifdef code that are not active for the module doesn't get turned off.   It gets deleted.

If we had a Delta, Cartesian, etc, etc, module...   The code within any module would be much simpler to look at and maintain.    That is why I want to turn on and off a given module.   Obviously, we only want one module turned on when we are doing Auto Bed Leveling for a machine.  Presumably, it would be the one that best matches the machine type.    And then we don't care what crazy stuff is happening in the other related modules.

Anyway...  That is why I want to see an example of turning on and off modules.

Wackerbarth, 2015-10-05:
That is easy enough to do.

But, first, you need to create a module that duplicates the current code in two pieces.

One is the code that you are removing and the other is the rest of Marlin_main.cpp.

Turn the part that you are splitting out into a subroutine and, in Marlin_main, replace it with a call to your subroutine.

Now, you have a module that splits something out for you to work on without affecting the rest of the world.  Test it, etc. the make sure you haven't introduced a problem while doing the split.

Now, one easy way to make variants of this new module is to clone it for each variant and then simply put an #ifdef enabling construct around the whole thing, using a different variant for each copy.

The compiler will generate code for the enabled variant and nothing for the other copies. So you still get the same result.

Then, you are free to simplify each copy. For example, if you KNOW that a particular block of code is NEVER active, given the enclosing variant selector, you can eliminate those lines of code from that copy.
Similarly, if you KNOW that some block is ALWAYS enabled, you can remove the block enabling test that surrounds it.

Now, to keep things neatly organized, you can move all of these variant files into a folder so that the various variants are kept together.

But how you do that organization is a separate issue. There are a number of reasonable ways from which we can choose to organize the files once we are freed from the "flat" structure that is a restriction of the "sketch" layout.

Roxy-3DPrintBoard, 2015-10-05:
OK.   We are back on the same page.

Wackerbarth, 2015-10-06:
A module should be both a .ccp file and the corresponding .h file that describes the interface.

clefranc, 2015-10-08:
> 1) Download and install the Arduino IDE 1.6.6

**DONE**
> 2) Download the Marlin source as a library from the trial_dev branch here (== dev in my repository)

There is no trial_dev branch here, why go to your repository? What do I do with this download? Just **Add .Zip Library...** in *Sketch/Include Library* of the IDE?

Wackerbarth, 2015-10-08:
@clefranc -- "trial_dev" is obsolete. Since that message was posted, it has been promoted to "dev".

The reason for using another repository was only one of timing and initial testing.
As we get things "shaken out", we will be moving to the kinds of entries that you are suggesting.
However, you should not be doing initial testing on a live, operational, system.

Your question concerning "What to do with ?" is a documentation issue.
In the short term, I assume that the developers have some familiarity with the Arduino IDE installation.
```
~/Documents/Arduino/libraries/marlin
```
is where user libraries get installed on my Mac.

Once we really get to a good Release, we will be able to post it as an entry in the IDE distribution scheme and it will become really easy for a new user to just "click" to do the download and installation.

clefranc, 2015-10-08:
> 1) Download and install the Arduino IDE 1.6.6

**DONE**
> 2) Download the Marlin source as a library from the trial_dev branch here (== dev in my repository)

**DONE**
> 3) Get the URL to be entered in the Arduino Boards Manager from the MarlinFirmware branch of my repository.

**DONE**
> 4) Use the boards manager to download the support platform.

**DONE**
> 5) (Important) Select a board from the Marlin section of the available boards.

**DONE**
> 6) Select a configuration from the Boards->Printers menu

You mean *Tools/Printers*.
**DONE**
> 7) Open the .ino file File->Examples->marlin->MarlinFirmware

Where is the `.ino` located? In `~/Documents/Arduino/libraries\MarlinDev-master\examples\MarlinFirmware`?

Wackerbarth, 2015-10-08:
"You mean Tools->Printers" . Yes, thanks.  I'm still waiting for some resolution of how we are going to handle the aspects of documentation that change from one version to the next.

The .ino is in a folder (probably at the path that you describe). But the menu calls it up directly for you.

clefranc, 2015-10-08:
> 7) Open the .ino file File->Examples->marlin->MarlinFirmware

**DONE**
> 8) Compile

```
C:\Users\Christian\Documents\Arduino\libraries\MarlinDev-master\examples\MarlinFirmware\MarlinFirmware.ino:5:31: fatal error: LiquidCrystal_I2C.h: No such file or directory

 #include <LiquidCrystal_I2C.h>

                               ^

compilation terminated.

exit status 1
Error compiling.
```

Wackerbarth, 2015-10-08:
You also need to install the supporting libraries. How were you getting them previously?

clefranc, 2015-10-08:
I was hoping that your "new system" will take care of it. Alas, it's not in your "incomplete" "work-in-progress".

I never had to install `LiquidCrystal_I2C` library. It's just since I try to compile your "new system" that I get this error.

clefranc, 2015-10-08:
```
C:\Users\Christian\Documents\Arduino\libraries\MarlinDev-master\examples\MarlinFirmware\MarlinFirmware.ino:6:24: fatal error: LiquidTWI2.h: No such file or directory

 #include <LiquidTWI2.h>

                        ^

compilation terminated.

exit status 1
Error compiling.
```

clefranc, 2015-10-08:
OK, I've installed the missing library that SHOULD be included with Marlin. The MarlinFirmware is open from the `File` menu. The IDE reopen with 2 tab: `MarlinFirmware`and `FirmwareCustomization.h`. Going to `FirmwareCustomization.h`:
```
/*
 *
 *  Customization for a particular printer
 *

    Although you could define all of your customization here,
    the preferred procedure is to assign a unique id to each printer
    and place the customization for that printer in the customization area.
    You can then reference that file either from the command line or by defining the path here.

    If you have multiple printers, it is suggested that you create a file for each of them.
    You can then select one of them from a menu in the Arduino IDE.
    Please see ... for more information

    Even if you have only one, you can utilize that same mechanism to specify the id that you have assigned to that printer
    or, you could define it here *
 */
```
> place the customization for that printer in the customization area

Where is located this *customization area*?
> You can then reference that file either from the command line

I've never used a command line to compile Marlin before, how do you do this?
> You can then select one of them from a menu in the Arduino IDE

Which menu? The `Files/Examples`?
> Please see ... for more information

???
> Even if you have only one, you can utilize that same mechanism to specify the id that you have assigned to that printer or, you could define it here *

I suppose *Please see ... for more information* contain **more information** on this...

### THIS NEED MORE WORK

clefranc, 2015-10-08:
```
  // You can obtain a unique id by visiting https://www.uuidgenerator.net/version4

  #define  SPECIFIC_PRINTER_ID e9cbe275
```
The generator gave me this number `cddbbcbb-ffa6-4682-bbae-c5711ca0968c`.

Why yours is only 8 characters?

Do I need to use the whole version 4 UUID or only part of it?

Wackerbarth, 2015-10-08:
@clefranc -- I agree that the scheme needs more work, just like a number of other parts of the Marlin system.
Most of your complaints are directly related to insufficient documentation and packaging. I'm sorry that you feel it necessary to have complete end-user level documentation in order for you to adapt to changes in the Development branch.  Most Developers are able to adapt to change by examining examples and using them as a guide to create their own customization.

As to your question about UUID, the "difficulty" is in the fact that the preprocessor treats the "-" as a call for subtraction. Since the SPECIFIC_PRINTER_ID is just a label, it can be created in many ways. It need be unique only on a local basis unless it is convenient to share configurations (for example while trying to provide assistance to some user.)

clefranc, 2015-10-09:
> Most Developers are able to adapt to change by examining examples and using them as a guide to create their own customization.

Fair enough, I'll return to play in my fork then.
> Your feedback, and assistance, would be appreciated.

You have a curious way to show your appreciation...

iplayfast, 2015-10-15:
Just decided to try to compile with dev system one month ago. Then updated, and tried again. (wow, that's a big change).

I don't really understand the advantage of bypassing the arduino help stuff. I do understand putting as much as possible in libraries. But I feel this is going overboard. People who aren't programmers who nevertheless try to compile the firmware for their printer are going to be totally lost.

I'm an experienced programmer with both normal c++ and arduino, and I'm not able to get things to work now.  Just saying. (include files can't be found).

Wackerbarth, 2015-10-15:
@iplayfast -- What do you mean by "the arduino help stuff"? Other than the fact that we need to get the needed lcd display support libraries into the environment, (at this point, I don't know if we want to provide specific versions ourselves or rely on the Arduino library distribution to provide them), the IDE handles everything for you. After editing one configuration file (assuming that you need to customize something), you just select the MarlinFirmware from the menu and compile it. And, for the configuration file, if you have some "stock" design, rather than copying the entire file(s), all you will need to do is specify which one you want to include. I'm hoping that we will get standardized files for many of the commercial printers and kits.

Are the include files that you cannot find related to the "missing" libraries?

Wackerbarth, 2015-10-15:
By the way, I still waiting for a viable way to document, on a permanent basis, the significant differences which distinguish the various versions. 1.0.x has quite a different set of configuration parameters and functional differences from 1.1.x. And 1.2.x already has significant differences in the build procedure and we will expect it to also gain additional features. Then 1.3.x will differ still in some way.  Just because a version is old and obsolete, we cannot just forget to preserve the documentation related to it.

Thus far, we don't have any strategy in place to allow the user to find the correct documentation for his particular version.  And, I sorry to say, relying on some wiki to which we don't have access for a number of committers is not a workable solution. Right now, we are at the mercy of the few who do have that access. And they are not willing to spend the time to document parts of the system that are not of their design. Thus, we have no reasonable way to document the system.

shaggythesheep, 2015-10-15:
I realise people have spent a lot of time and effort developing the wiki, but perhaps a more suitable location would be something like [ReadTheDocs](https://readthedocs.org/) - it supports different versions and languages all within the same structure. I'v also noticed quite a few other open source projects use it, such as OctoPrint

Roxy-3DPrintBoard, 2015-10-15:
What is wrong with having the main documentation right next to the source code?   Maybe as a Word file or as a .PDF file?   And if it needs to be multiple places so people can find it easier, then copy it a few other places too?   The important thing is that it can be found easily.

iplayfast, 2015-10-15:
@Wackerbarth  The arduino help stuff, was actually referring to all "By breaking away from the "helpful" things that the Arduino IDE does for us, we are able to return to the traditional C/C++ structure of modular programming." you mentioned early on.

I pulled last night, and the only ino file I could find was in examples/MarlinFirmware/MarlinFirmware.ino
which when I tried to compile, gave me the error MarlinFirmware.h not found. 
Played with a bunch of other stuff, but this was the crux of the problem.

So is the idea that this whole thing is going to be a library that you compile an example to create your project?

Before, there were a number of default configuration files to choose from which could be modified by commenting out the defines to turn on and off various items. Now there is FirmwareCustomization.h which I assume is to do the same thing, but it's got #if 0 for only a few items I think I'm missing something or the example is incomplete. (I understand this is development, just showing my thoughts).

Arduino doesn't restrict the location of include files, so to me the most straight forward and obvious way of doing what you want is have directories full of include files underneath the main source of the program. Main source just includes configuation.h first and then all the other include files. Ifdef's in the configuration turn off or on features. Include files can include #ifdef'd source code as well as defines, so I'm not sure what you are going to gain by this current arrangement. which looks to me to have all the source in the main directory.  Again, I'm probably missing something basic since this looks obvious to me, and you've not gone this route, so I'm not critiquing, just explaining what I expected.

Lastly, I like the document PDF file idea in the codebase. Actually why not have a readme.txt, along with gettingstarted.txt. Unless we have a lot of pictures, I don't even see the advantage of PDF over text (or html).  Looking over the wiki, it doesn't actually look like there is gobs of information there. Just one long page. Saving it to the Documentation folder only adds 51k. Trivial. Changing the links, to be reference instead of webpage/reference would put the wiki info into the documentation.

Wackerbarth, 2015-10-15:
> On Oct 15, 2015, at 11:11 AM, Chris Bruner <notifications@github.com> wrote:
> 
> @Wackerbarth <https://github.com/Wackerbarth> The arduino help stuff, was actually referring to all "By breaking away from the "helpful" things that the Arduino IDE does for us, we are able to return to the traditional C/C++ structure of modular programming." you mentioned early on.
> 
> I pulled last night, and the only ino file I could find was in examples/MarlinFirmware/MarlinFirmware.ino
> 
That IS THE .ino file.
> which when I tried to compile, gave me the error MarlinFirmware.h not found. 
> 
I suspect that you are trying to compile this using one of the Arduino “platform” definitions rather than the Marlin “platform”. My instructions specifically state that you need to use the platform definition that I have provided. Not only does it supply the correct include directories (which are needed to “invert” the idea that the main program is in the library, but it also provides a set of core libraries that are compatible with the Marlin version of the Serial IO.
> Played with a bunch of other stuff, but this was the crux of the problem.
> 
> So is the idea that this whole thing is going to be a library that you compile an example to create your project?
> 
If I understand your remark, I think the answer is “yes”.

The reason for the unusual placement of the included “FirmwareCustomization.h” is to bring the potential customization file up to the level where it would be visible to a rather unsophiscated user who sees it as a part of the “sketch” rather than buried down in the “library”.
> Before, there were a number of default configuration files to choose from which could be modified by commenting out the defines to turn on and off various items. Now there is FirmwareCustomization.h [...]
Yes, this is “development”. There are many ways that you can specify the configuration parameters. The various "#if 0” sections are there as a hint to current developers.

Once we get something decided about documentation, we will be in a position to document the various schemes in detail so that they will be more readily understood.

For the short term, I would expect a user to either
(1) just copy the desired “example” to this customization file, or
(2) place a modified version of an “example” over in the configurations/custom/ area and make a reference to it either from the IDE menu or in this customization file.

In the mid term, the idea is to turn the current configuration files into “default” configurations so that you can select one of them and then override a few settings, here and there.

Ultimately, I would like to see a complete definition for each manufacturer/model that is itself, just some overrides for a more generic description.

The entire “design” is to implement specific configurations in much the same style that you would use in defining a class structure in a language such as C++. But here, the “class” is represented by an #include file.

> Arduino doesn't restrict the location of include files, so to me the most straight forward and obvious way of doing what you want is have directories full of include files underneath the main source of the program.
> 
As you say, the location of the includes is not restricted (as long as you are part of a library rather than a sketch). But, rather than burying the configuration files down in the code hierarchy, I feel that it is better to keep them (together) in a folder high in the structure.
> Main source just includes configuation.h first and then all the other include files. Ifdef's in the configuration turn off or on features. Include files can include #ifdef'd source code as well as [...]
All of the code is in the main directory only because that is the way that it was previously. Now that we are in a library structure, I anticipate that there will be a migration to a more manageable structure. But, the point in making the change was to enable such a migration rather than attempting to redo everything all at once.
> Again, I'm probably missing something basic since this looks obvious to me, and you've not gone this route, so I'm not critiquing, just explaining what I expected.
>
> Lastly, I like the document PDF [...]
Agreed. I don’t advocate using PDF. Instead, we need something which we can edit much as you would edit lines of the source code. There are a number of documentation “languages”, for example, markup, which could be used. Again, I am not advocating any particular one.
> Looking over the wiki, it doesn't actually look like there is gobs of information there. Just one long page. Saving it to the Documentation folder only adds 51k. Trivial.
> 
The only argument that I have seen for the use of a wiki has to do with the editing. But I don’t care how it gets edited, just that we have a scheme in place so that we can develop more, and better documentation.
> Changing the links, to be reference instead of webpage/reference would put the wiki info into the documentation.

iplayfast, 2015-10-16:
OK, your first comment with instructions were for trial_dev, and I had pulled master. Therein lies the confusion. I should have read more... (sorry bout that).

**done**
    1) Download and install the Arduino IDE 1.6.6
    2) Download the Marlin source as a library from the master branch installed as sketchbook/libraries/MarlinDev

**I can't find this. can you provide a link?**
    3) Get the URL to be entered in the Arduino Boards Manager from the MarlinFirmware branch of my repository.
    4) Use the boards manager to download the support platform.
    5) (Important) Select a board from the Marlin section of the available boards.
    6) Select a configuration from the Boards->Printers menu
    7) Open the .ino file File->Examples->marlin->MarlinFirmware
    8) Compile

**is this local to the ino file? Sorry not familiar with this.**
    9) Added feature -- you can add a "boards.local.txt" to the platform and place an entry for your configuration in it. Then you can select your printer as the configuration to build.

sebastian-dev, 2015-10-20:
here the missing files.. 
https://github.com/Wackerbarth/MarlinDev/tree/MarlinFirmware

iplayfast, 2015-10-30:
Thank you for that. I'm now upto step 8 trying to compile. 
```
Arduino: 1.6.6 Hourly Build 2015/10/14 10:26 (Linux), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560), Mini Kossel"

/home/chris/bin/arduino-nightly/arduino-builder -dump-prefs -logger=machine -hardware "/home/chris/bin/arduino-nightly/hardware" -hardware "/home/chris/.arduino15/packages" -hardware "/home/chris/Dropbox/sketchbook/hardware" -tools "/home/chris/bin/arduino-nightly/tools-builder" -tools "/home/chris/bin/arduino-nightly/hardware/tools/avr" -tools "/home/chris/.arduino15/packages" -libraries "/home/chris/Dropbox/sketchbook/libraries" -libraries "/home/chris/bin/arduino-nightly/libraries" -fqbn=Marlin:avr:mega:cpu=atmega2560,marlin=travis25 -ide-version=10606 -build-path "/tmp/build4922ff188a31c6c1cbbd86d8f96439fd.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/chris/Dropbox/sketchbook/MarlinDev/MarlinFirmware/MarlinFirmware.ino"
/home/chris/bin/arduino-nightly/arduino-builder -compile -logger=machine -hardware "/home/chris/bin/arduino-nightly/hardware" -hardware "/home/chris/.arduino15/packages" -hardware "/home/chris/Dropbox/sketchbook/hardware" -tools "/home/chris/bin/arduino-nightly/tools-builder" -tools "/home/chris/bin/arduino-nightly/hardware/tools/avr" -tools "/home/chris/.arduino15/packages" -libraries "/home/chris/Dropbox/sketchbook/libraries" -libraries "/home/chris/bin/arduino-nightly/libraries" -fqbn=Marlin:avr:mega:cpu=atmega2560,marlin=travis25 -ide-version=10606 -build-path "/tmp/build4922ff188a31c6c1cbbd86d8f96439fd.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/chris/Dropbox/sketchbook/MarlinDev/MarlinFirmware/MarlinFirmware.ino"
WARNING: Spurious .git folder in 'PID' library
WARNING: Spurious .git folder in 'MFRC522' library
mkdir -p "/tmp/build4922ff188a31c6c1cbbd86d8f96439fd.tmp/sketch/"
generate_version_header_for_marlin "/home/chris/Dropbox/sketchbook/MarlinDev/MarlinFirmware" "/tmp/build4922ff188a31c6c1cbbd86d8f96439fd.tmp/sketch/_Version.h"
"/home/chris/bin/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -DCONFIGURATION_FILE=configurations/deprecated_configurations/delta/kossel_mini/Configuration.h -DMARLIN -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DUSE_AUTOMATIC_VERSIONING  -I/tmp/build4922ff188a31c6c1cbbd86d8f96439fd.tmp/sketch  "/tmp/build4922ff188a31c6c1cbbd86d8f96439fd.tmp/sketch/MarlinFirmware.ino.cpp"
In file included from /home/chris/Dropbox/sketchbook/MarlinDev/MarlinFirmware/MarlinFirmware.ino:2:0:
MarlinFirmware.h:41: error: #error Failed to properly include configuration parameters
   #error Failed to properly include configuration parameters
    ^
exit status 1
#error Failed to properly include configuration parameters
```
So at this point I'm supposed to modify FirmwareCustomization.h to have the specific defines for my printer? (I previously selected the "Mini Kossel" printer so I thought this was done).

Still a bit lost here.

Wackerbarth, 2015-10-30:
@iplayfast -- Chris,
I'm not near my main build machine this weekend. So, I may have a bit more difficulty in determining exactly what is happening. However, from the output that you provided, I'll try to figure out what is going wrong and, when I do, provide a fix for it.

Wackerbarth, 2015-10-31:
@iplayfast -- Chris,
Nothing obvious. It appears that you have selected the "Mini Kossel" variant from the build menu. (Which compiles just fine on my laptop) Did you modify the `FirmwareCustomization.h` at all?
What are the contents of `"/tmp/build4922ff188a31c6c1cbbd86d8f96439fd.tmp/sketch/_Version.h"`
(note: the particular /tmp/build... will change when you rerun it.)

Wackerbarth, 2015-10-31:
@iplayfast -- Chris,
I think I found something..
Comparing the compile lines, I found that you have a different path to the .ino file.
Did you open that file by selecting the IDE File->Examples->marlin->MarlinFirmware menu item?

iplayfast, 2015-10-31:
No I think I opened it at the sketchbook/MarlinDev/MarlinFirmware/MarlinFirmware.ino
Building from the example I get 
```
Arduino: 1.6.6 Hourly Build 2015/10/14 10:26 (Linux), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560), Mini Kossel"

/home/chris/bin/arduino-nightly/arduino-builder -dump-prefs -logger=machine -hardware "/home/chris/bin/arduino-nightly/hardware" -hardware "/home/chris/.arduino15/packages" -hardware "/home/chris/Dropbox/sketchbook/hardware" -tools "/home/chris/bin/arduino-nightly/tools-builder" -tools "/home/chris/bin/arduino-nightly/hardware/tools/avr" -tools "/home/chris/.arduino15/packages" -libraries "/home/chris/Dropbox/sketchbook/libraries" -libraries "/home/chris/bin/arduino-nightly/libraries" -fqbn=Marlin:avr:mega:cpu=atmega2560,marlin=travis25 -ide-version=10606 -build-path "/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/chris/Dropbox/sketchbook/libraries/MarlinDev/examples/MarlinFirmware/MarlinFirmware.ino"
/home/chris/bin/arduino-nightly/arduino-builder -compile -logger=machine -hardware "/home/chris/bin/arduino-nightly/hardware" -hardware "/home/chris/.arduino15/packages" -hardware "/home/chris/Dropbox/sketchbook/hardware" -tools "/home/chris/bin/arduino-nightly/tools-builder" -tools "/home/chris/bin/arduino-nightly/hardware/tools/avr" -tools "/home/chris/.arduino15/packages" -libraries "/home/chris/Dropbox/sketchbook/libraries" -libraries "/home/chris/bin/arduino-nightly/libraries" -fqbn=Marlin:avr:mega:cpu=atmega2560,marlin=travis25 -ide-version=10606 -build-path "/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/chris/Dropbox/sketchbook/libraries/MarlinDev/examples/MarlinFirmware/MarlinFirmware.ino"
WARNING: Spurious .git folder in 'PID' library
WARNING: Spurious .git folder in 'MFRC522' library
mkdir -p "/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch/"
generate_version_header_for_marlin "/home/chris/Dropbox/sketchbook/libraries/MarlinDev/examples/MarlinFirmware" "/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch/_Version.h"
"/home/chris/bin/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -DCONFIGURATION_FILE=configurations/deprecated_configurations/delta/kossel_mini/Configuration.h -DMARLIN -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DUSE_AUTOMATIC_VERSIONING  -I/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch  "/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch/MarlinFirmware.ino.cpp"
"/home/chris/bin/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -DCONFIGURATION_FILE=configurations/deprecated_configurations/delta/kossel_mini/Configuration.h -DMARLIN -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DUSE_AUTOMATIC_VERSIONING  -I/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/cores/arduino" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/variants/mega" "-I/home/chris/Dropbox/sketchbook/libraries/MarlinDev" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/libraries/LiquidCrystal/src" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/libraries/U8glib" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/libraries/Wire" "/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch/MarlinFirmware.ino.cpp"
"/home/chris/bin/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -DCONFIGURATION_FILE=configurations/deprecated_configurations/delta/kossel_mini/Configuration.h -DMARLIN -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DUSE_AUTOMATIC_VERSIONING  -I/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/cores/arduino" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/variants/mega" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/libraries/Wire" "-I/home/chris/Dropbox/sketchbook/libraries/MarlinDev" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/libraries/LiquidCrystal/src" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/libraries/U8glib" "/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch/MarlinFirmware.ino.cpp"
"/home/chris/bin/arduino-nightly/hardware/tools/avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -DCONFIGURATION_FILE=configurations/deprecated_configurations/delta/kossel_mini/Configuration.h -DMARLIN -w -x c++ -M -MG -MP -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -DUSE_AUTOMATIC_VERSIONING  -I/tmp/buildae24d6c6472f9cca1f56d236ee5d98b4.tmp/sketch "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/cores/arduino" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/variants/mega" "-I/home/chris/Dropbox/sketchbook/libraries/MarlinDev" "-I/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/libraries/Wire" "-I/home/chris/.arduino15/p [... cut ...]
/* This file is automatically generated by an Arduino hook
 * Do not manually edit it
 * It does not get committed to the repository
 */

#define BUILD_UNIX_DATETIME 1446302579
#define STRING_DISTRIBUTION_DATE "2015-10-31 10:42"
#define SHORT_BUILD_VERSION "1.1 master"
#define DETAILED_BUILD_VERSION "1.1 master-241-g0be6bd0"
#define SOURCE_CODE_URL  "https://github.com/MarlinFirmware/MarlinDev/"
// Deprecated URL definition
#define FIRMWARE_URL  "https://github.com/MarlinFirmware/MarlinDev/"

```

iplayfast, 2015-10-31:
Looking deeper I think the bad file you found was from a previous attempt. I installed the libraries missing in the above error and .... SUCCESS!
Now to upload to my Kossel mini and see what happens!  Thanks for your patience.

Wackerbarth, 2015-10-31:
OK, that's fine. Now to customize your configuration 

You will want a file with the defines the options that you mention above like this
```
#define CUSTOMIZATION_H

#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define ULTIPANEL
#define NEWPANEL
#define SDSUPPORT
#define LCD_WIDTH 20
#define LCD_HEIGHT 4

#include "configurations/transitional_default_configurations/delta/kossel_mini/Configuration.h"
```
(Note that the recommended include will change as we continue with the modularization of the configuration parameters. But this will continue to work.)

The easiest way is to just insert that in the `FirmwareCustomization.h` file. That is what I would recommend for non-developers who have only one printer configuration.

A more complete way to handle it is to make a separate configuration file.
After choosing <SOME_ID> to be a unique identifier (I use `2ca060e1` which is the first part of the UUID I assigned to the printer. You might use a similar scheme. Or you could use something which incorporates a distinctive phrase such as `iplayfast`, etc.  But, in case we want to share the files to help in debugging, make the name distinctive. Many users might have a 'my_printer')

Save the file as `configurations/custom/<SOME_ID>/Customization.h`
Then you can create a menu entry for your custom configuration by creating the menu additions in
`/home/chris/.arduino15/packages/Marlin/hardware/avr/1.0.0/boards.local.txt`

That file would have entries such as 
```
mega.menu.marlin.local02=My Delta Printer
mega.menu.marlin.local02.build.unique_id_flags=-DSPECIFIC_PRINTER_ID=<SOME_ID>
```
You can add multiple configurations by creating additional local?? pairs.

Roxy-3DPrintBoard, 2015-10-31:
Is the plan to have everything in the right place so the Developer just unpacks the archive and everything is there and works?

Wackerbarth, 2015-10-31:
That all depends on how the LCD libraries get handled in the evolving IDE. I would prefer to exclude them IF we are not requiring special versions of them. But, if we are using the standard version of the libraries, the library manager should end up taking care of the details for those libraries.
And, rather than unpacking an archive, the user will either do a git fetch or just download the Marlin library using the IDE library manager.

 Of course, if the user isn't using a completely standard configuration, he will have to do a little configuration editing to specify his machine.

Roxy-3DPrintBoard, 2015-10-31:
Mostly, I'm thinking we need a totally painless and easy way for the user to get the environment setup on their machine.   IPlayFast is obviously motivated and competent and he is still having problems.   I just want to confirm that the goal is to make this very simple and painless for the end user that doesn't know anything about coding.

Wackerbarth, 2015-10-31:
Yes, that is the goal. However, there is still work to do to get it to that "easy-to-use" point.
But, although we COULD do so, I'm not particularly keen on including standard libraries just so the user doesn't have to follow an instruction to "Load the following libraries by selecting them in ..." (the library manager)

Roxy-3DPrintBoard, 2015-11-01:
Would it make sense to do our own Arduino distribution.  Basically just take the official Arduino and warm over the libraries and such so it works with our stuff.   And then have our stuff in the right place so everything just works?

Wackerbarth, 2015-11-01:
Basically, I would say no. There are too many parts that have nothing to do with our code which we would have to maintain. And, additionally, we would get stuck with the customer service problems associated with setting up Arduino. As it is, we don't have a bunch of contributors who are interested in doing documentation. So, it will be much easier for us to simply tell the users to install the Arduino IDE (using documentation from the Arduino community) and then load the Marlin platform.  Once we get to a release based on the current dev branch (rather than the RC version), we should be able to get that download into the standard Arduino library manager and then the download will be nothing more than electing a short list of libraries, one of which will be Marlin itself.

Then, assuming that the vendors wish to cooperate, the build may be nothing more than selecting the appropriate configuration from a menu.

(Note: the process will remain a little more complicated while we incrementally make the changes to support all of this. But, as a replacement for the 1.1 builds, it really isn't more complicated. It is just that a few files have moved.)

iplayfast, 2015-11-02:
Well, sad to say, it doesn't work.

With your system, I don't have the crystal display, the push button, or even serial communication, so I think I've got something major wrong.
This is the modifications I've added to the FirmwareCustomization.h file
```
#define CUSTOMIZATION_H

#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define ULTIPANEL
#define NEWPANEL
#define SDSUPPORT
#define LCD_WIDTH 20
#define LCD_HEIGHT 4
#define SERIAL_PORT 0
#define BAUDRATE 115200
#define MOTHERBOARD 33
#define POWER_SUPPLY 1
#include "configurations/transitional_default_configurations/delta/kossel_mini/Configuration.h"
```
This is a copy of a configuration of an older marlin build that has the display and menus working
With this configuration I can talk to it with the serial monitor as well (the baud rate needs to be 1/2 the compiled baud rate, but at least it communicates). 

```
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
// this one seems to work on the Delta display
// This configurtion file contains the basic settings.
// Advanced settings can be found in Configuration_adv.h 
// BASIC SETTINGS: select your board type, temperature sensor type, axis scaling, and endstop configuration

//User specified version info of this build to display in [Pronterface, etc] terminal window during startup.
//Implementation of an idea by Prof Braino to inform user that any changes made
//to this build by the user have been successfully uploaded into firmware.
#define STRING_VERSION_CONFIG_H __DATE__ "  8-15-13 " __TIME__ // build date and time
#define STRING_CONFIG_H_AUTHOR "(FOLGER TECH, 8 i3 Prusa Config)" //Who made the changes.

// SERIAL_PORT selects which serial port should be used for communication with the host.
// This allows the connection of wireless adapters (for instance) to non-default port pins.
// Serial port 0 is still used by the Arduino bootloader regardless of this setting.
#define SERIAL_PORT 0

// This determines the communication speed of the printer
//#define BAUDRATE 250000
#define BAUDRATE 115200

//// The following define selects which electronics board you have. Please choose the one that matches your setup
// 10 = Gen7 custom (Alfons3 Version) "https://github.com/Alfons3/Generation_7_Electronics"
// 11 = Gen7 v1.1, v1.2 = 11
// 12 = Gen7 v1.3
// 13 = Gen7 v1.4
// 3  = MEGA/RAMPS up to 1.2 = 3
// 33 = RAMPS 1.3 (Power outputs: Extruder, Bed, Fan)
// 34 = RAMPS 1.3 (Power outputs: Extruder0, Extruder1, Bed)
// 4  = Duemilanove w/ ATMega328P pin assignment
// 5  = Gen6
// 51 = Gen6 deluxe
// 6  = Sanguinololu < 1.2
// 62 = Sanguinololu 1.2 and above
// 63 = Melzi
// 7  = Ultimaker
// 71 = Ultimaker (Older electronics. Pre 1.5.4. This is rare)
// 8  = Teensylu
// 81 = Printrboard (AT90USB1286)
// 82 = Brainwave (AT90USB646)
// 9  = Gen3+
// 70 = Megatronics
// 90 = Alpha OMCA board
// 91 = Final OMCA board
// 301 = Rambo

[... 299 lines cut ...]
  #define LCD_WIDTH 20
  #define LCD_HEIGHT 4
  
#else //no panel but just lcd 
  #ifdef ULTRA_LCD
    #define LCD_WIDTH 16
    #define LCD_HEIGHT 2    
  #endif
#endif

// Increase the FAN pwm frequency. Removes the PWM noise but increases heating in the FET/Arduino
//#define FAST_PWM_FAN

// M240  Triggers a camera by emulating a Canon RC-1 Remote
// Data from: http://www.doc-diy.net/photo/rc-1_hacked/
// #define PHOTOGRAPH_PIN     23

// SF send wrong arc g-codes when using Arc Point as fillet procedure
//#define SF_ARC_FIX

#include "Configuration_adv.h"
#include "thermistortables.h"

#endif //__CONFIGURATION_H
```

iplayfast, 2015-11-24:
I notice that arduino has released the ide version 1.6.6 So you can remove the build-nightly stuff.

Wackerbarth, 2015-11-24:
PR #287 started to address removing the "nightly" references. Please let me know, as a comment on that PR, if you find other spots needing cleanup.

"now I can't install boards because the CRC doesn't match." -- Please open an issue for this and describe EXACTLY (in a manner that will allow me to duplicate it) what you have as a setup.

closed 2016-05-04 by jbrazio