MarlinFirmware/MarlinDev #231: Where is tne platformio support

thawkins, 2015-10-05:
We seem to have regressed backwards and lost the support for platformio...

Wackerbarth, 2015-10-05:
Not "lost", as such. It just needs to be ported over.  What I haven't determined is whether there should be as a separate branch for each build method or we should have multiple methods as sub-folders on a common support branch.

thawkins, 2015-10-05:
Ok cool, i just went to pull down the latest dev version to build onto a
new bit and it was not there.

Wackerbarth, 2015-10-05:
Don't try to use the current "dev" from this repo for the next day -- Get the "dev" branch from my repository instead. The massive restyling is about to be rolled back and "dev" will get a wholesale change. (It's also available on a branch called "trial-dev" in this repo right now.)

thawkins, 2015-10-05:
Ok i will hold off untill next weekend

Wackerbarth, 2015-10-05:
Except for the branch name, it should be good.

Roxy-3DPrintBoard, 2015-10-06:
> The massive restyling is about to be rolled back and "dev" will get a wholesale change. (It's also available on a branch called "trial-dev" in this repo right now.)

Question:   In this post:  https://github.com/MarlinFirmware/Marlin/issues/2665#issuecomment-145698494  I pasted in some code for the user to look at and consider using.  The 'restyling' is going to mess up the section in the 2nd code block, right?

````
 //
 // Should we break up the move because it is too big ???
 //
      if (abs(x-current_position[X_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE   ||
          abs(y-current_position[Y_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE   ||
          abs(z-current_position[Z_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE ) {
        do_blocking_move_to( current_position[X_AXIS] + (x-current_position[X_AXIS])/2.0,
                     current_position[Y_AXIS] + (y-current_position[Y_AXIS])/2.0,
                     current_position[Z_AXIS] + (z-current_position[Z_AXIS])/2.0);
//
// At this point, we are halfway to where we want to be.  Do the other half of the move.
//
````

Wackerbarth, 2015-10-06:
As long as you are quoting complete statements (to do otherwise would be VERY unusual), the code will work however it is styled. We now have no enforcement of style requirements. We currently have a number of slightly different styles in various areas. All I can say is "Try to make it follow the style of similar code". As far as I know, there is no automated way to reasonably match this style.

This means that I cannot use an automated styler to make my code consistent without then having to edit it manually.  And if that code includes parts of surrounding code, I will end up messing that part up also.

thawkins, 2015-10-06:
Jetbrains cLion ide will style code automaticaly using tbe clang compiler's
parsing stages to give it tbe insights it needs not to break code.

If you want i can clone the repo,  restyle it and commit it back so you can
see the results.

Roxy-3DPrintBoard, 2015-10-06:
@thawkins More help and suggestion will be beneficial.   I mostly was just pointing this out as an example of where automatic styler stuff probably makes some areas harder to read.   Making each of those statements a single line is going to be difficult to look at and understand.   I was hoping the Astyle program was going to be smart enough to make an exception for them.

Wackerbarth, 2015-10-06:
Tim,

Scot, et. al. are all “up in arms” about the way they indent conditionals nested within code blocks and each other. For indention purposes, they want the #if/#else/#endif to be indented as if they were the code equivalent if block.

astyle has nice indenting in many styles. But, as far as I can tell, it does not have options which will produce that effect.

If you have another automated styler which will provide that kind of indention, I would be VERY interested in seeing what it can do.

As much as some insist on their preferred style, I want to insist on ANY reasonable style that can be easily automated and tested.
In my normal workflow, I am accustomed to having the styler run just before, if not as a part of the commit to a repository.
That keeps the repository clean and consistent. It also tends to preserve commit blocks because they don’t get fragmented by cosmetic tweaks.

Richard

thawkins, 2015-10-06:
I think this is a reasonable objective,

Im doing a load of go code now, and the language tools include a formatter
that imposes a single style which everybody uses,  it eliminates all
arguments about styling.

C/c++ has had these rows about block indentation, function definition
indentation and tabs vs spaces for as long as i can remember. I wish there
had been something like "go fmt" back in the day, as most of this would not
have evolved.

I will dig around and see what i can find, the cLion one is good, is fully
configureable, but its paid for code.

thawkins, 2015-10-06:
Check out clang format

For the one cLion uses

http://clang.llvm.org/docs/ClangFormat.html

On Wed, Oct 7, 2015, 00:12 Tim Hawkins <tim.thawkins@gmail.com> wrote:

I think this is a reasonable objective,

Im doing a load of go code now, and the language tools include a formatter
that imposes a single style which everybody uses,  it eliminates all
arguments about styling.

C/c++ has had these rows about block indentation, function definition
indentation and tabs vs spaces for as long as i can remember. I wish there
had been something like "go fmt" back in the day, as most of this would not
have evolved.

I will dig around and see what i can find, the cLion one is good, is fully
configureable, but its paid for code.

>
> Jetbrains cLion ide will style code automaticaly using tbe clang
compiler's
> parsing stages to give it tbe insights it needs not to break code.
>
> If you want i can clone the repo, restyle it and commit it back so you can
> see the results.
>
>
> > As long as you are quoting complete statements (to do otherwise would be
> > VERY unusual), the code will work however it is styled. We now have no
> > enforcement of style requirements. We currently have a number of
slightly
> > different styles in various areas. All I can say is "Try to make it
follow
> > the style of similar code". As far as I know, there is no automated way
to
> > reasonably match this style.
> >
> > This means that I cannot use an automated styler to make my code
> > consistent without then having to edit it manually. And if that code
> > includes parts of surrounding code, I will end up messing that part up
also.
> >
> > <
https://github.com/MarlinFirmware/MarlinDev/issues/231#issuecomment-145906705
https://github.com/MarlinFirmware/MarlinDev/issues/231#issuecomment-145907468

—
<https://github.com/MarlinFirmware/MarlinDev/issues/231#issuecomment-145910663>.<img
src="
https://ci3.googleusercontent.com/proxy/Uw5OWby4i77SEzU54OaLo3h04bMvFMe_jbdwFWuc5AsxRhHWrp-TCheYoyCtJWsuYbOlOXtVtaw_g_6Rnr5WY6kxSvZPgv62T4p5RzRutsUFQBntcOhkaUTMSQAHllt0RyAEzMo5EAchJjtYu1opuDMuT2eA5w=s0-d-e1-ft#https://github.com/notifications/beacon/AAA4ySAOd90ba-0qBPp0IrB1dA79Nt8Bks5o4-gIgaJpZM4GIuUN.gif
">

Wackerbarth, 2015-10-06:
I see that this supports a number of the base styles, just a astyle does so.

However, those styles do not address the nesting of preprocessor directives within the conditional nesting of the blocks in the code.

That is the hangup about using astyle.

thawkins, 2015-10-06:
What about

--indent-preproc-cond

thawkins, 2015-10-06:
I think thats because it treats prepoc and lang stuff as seperate, ie tbe
clang/llvm works on t e language after tbe preproc has expanded everything.

thawkins, 2015-10-06:
Oops wrong way around

--indent-preproc-block

thawkins, 2015-10-11:
I will give the RC branch from /Marlin a whirl...  that will cover my needs for now.

closed 2016-05-04 by jbrazio