MarlinFirmware/MarlinDev #232: Delta Calibration Discussion

Roxy-3DPrintBoard, 2015-10-06:
I am trying to get my Delta Printer kit's nozzle movement calibrated.   I was under the impression some work had been done but it wasn't fully baked.   I can't find any Delta Calibration code.  Can somebody point me to it?

With that said, I am writing some code to assist in manually leveling beds that are supported by 3 positioning screws at the bed's perimeter.   It makes several 'Circles' in line with the positioning screws.  This allows the user to easily see which screw needs to be adjusted up or down.

I wanted to look at the Delta calibration code because I think I know how to assist the user in calibrating the DELTA_RADIUS_TRIM_TOWER and DELTA_DIAGONAL_ROD_TRIM_TOWER numbers.   I was going to add this as an option if the current Delta Calibration code doesn't help with those numbers.

Suggestions are welcome!

clefranc, 2015-10-06:
Can you take a look at my pet project: http://clefranc.github.io/Marlin4Delta/
See delta example in [Configuration.h](https://github.com/clefranc/Marlin4Delta/blob/master/Marlin/example_configurations/delta/generic/Configuration.h#L315).

Roxy-3DPrintBoard, 2015-10-06:
Thank You...   I'll be looking at that later today!

Wackerbarth, 2015-10-06:
@clefranc -- How many variables do you have that specify the tower positions? You see to have 3 raduis measurements and additionally 2 angular adjustments.  This is mathematically unstable because you have 5 variables trying to specify only 3 degrees of freedom.

clefranc, 2015-10-06:
Tower C is the reference, always 0°, so you can count it as the sixth variable.

Wackerbarth, 2015-10-06:
Yes, that has already been factored into both aspects. That is why you have only 5 variables. But one can ALWAYS describe the relative positions of the towers with only 3 variables. There is no way for you to have more. You have 3 points in a plane (6 coordinates), but 3 of those are "arbitrary" For example, the  0° restriction that you mention above. You can shift the coordinate system to place the origin at any arbitrary location and rotate it to meet your angular restriction.

clefranc, 2015-10-06:
The location of the towers are assume to be on a perfect circular arrangement.

Wackerbarth, 2015-10-06:
No, they are not.

float delta_tower_a_x = -sin((delta_alpha_ca - 60) * M_PI / 180.0f) * delta_radius_a;
float delta_tower_a_y = -cos((delta_alpha_ca - 60) * M_PI / 180.0f) * delta_radius_a;
float delta_tower_b_x =  sin((delta_alpha_cb - 60) * M_PI / 180.0f) * delta_radius_b;
float delta_tower_b_y = -cos((delta_alpha_cb - 60) * M_PI / 180.0f) * delta_radius_b;

You have BOTH an individual RADIUS AND an individual ANGLE  for a total of 5 degrees of freedom.

One way to describe 3 degrees of freedom is to use a COMMON RADIUS and 2 ANGLES.
However, I prefer using the 3 individual RADIUS measurements and FIXED angles.

Mathematically, they are equivalent and have only 3 degrees of freedom.

clefranc, 2015-10-06:
As long as my caliper gives me NEAR perfect print dimension, my solution suit me well. If you have a better solution, then submit a PR please.

Wackerbarth, 2015-10-06:
Understand that when you have too many variables, you can pick some set that works. In fact, you can pick many such sets. Except for a translation and rotation about the Z axis, their results are identical. (And I am sure that your calipers are not measuring those characteristics)

The problem occurs when you attempt to perform a calculation to determine the optimum values based on experimental data. There is no solution to which the result will converge.

As for an improvement to the method, that’s easy. Just drop two of the 5 parameters mentioned.

As I have said, my preference is to fix all 3 angles and fit the tower locations to their radial offset from a common center. (Again, this makes the math simpler). However, you could equally choose other ways to describe the 3 points.

If you want to use angles, you fix all of the tower radius setting to the same, but a variable, value. You find that radius and the two angles.

Another simple orientation is to confine the front towers to an X axis and confine the back tower to lie on the Y axis. There you might define the object origin to be 1/3 of the way up the y-axis.

Any of those schemes will work.  But the important thing is that they each have only 3 variables and therefore a unique solution.

Roxy-3DPrintBoard, 2015-10-06:
OK, but in this description, you are assuming all 3 towers are equally spaced around the center in both distance and angle.  I don't think there are many machines that are not this way, but right now the extra parameters lets you setup a printer that isn't symmetrical.

Wackerbarth, 2015-10-06:
That is the general design of a delta printer.  However, note that the towers usually ARE NOT in the exact design position. Finding, and correcting for, those small differences is important when it comes to getting the printhead to operate within the tolerances we expect.

Roxy-3DPrintBoard, 2015-10-06:
OK.  That is helpful.  And it should not be that hard to detect what happens if one arm is slightly longer or shorter than it should be.  Or for that matter, if its pivot point is slightly out of position.   If we had a table showing what happened if just one variable was out whack a little bit, it might be straight forward to have the calibration routine detect and make recommendations.

BuzzBumbleBee, 2015-10-12:
@Roxy-3DPrintBoard i wrote the commit to add the delta rod and radius trim functions, if you need any extra info just highlight me here :)

Roxy-3DPrintBoard, 2015-10-12:
> @BuzzBumbleBee 
> i wrote the commit to add the delta rod and radius trim functions, if you need any extra info just highlight me here :)

Oh!  Cool!   I had to tear my printer apart to replace the plastic pieces that cracked during assembly.  It is almost back together.   But as you might have noticed, I got side tracked writing some code to help physically level Delta beds.   I would love to add functions that automatically (auto-magically) detect and measure and provide updated numbers for the user.   

But I'm probably going down a rabbit hole.   I haven't even printed anything with my Delta yet!

BuzzBumbleBee, 2015-10-12:
@Roxy-3DPrintBoard the method I used was very manual (move --> test --> repeat) kind of thing, for reference :- 

1) DELTA_RADIUS_TRIM affects any concave / convex artifacts in the movement in the printer
2) DELTA_DIAGONAL_ROD_TRIM effects the physical size of the prints / moves made by the printer

So for 

1) I tweaked each trim to bring the concave / convex movements to a minimum for each axis
2) I printed a test object that could me measured over each "tower line", these measurements where then calculated into an error and added / subtracted on the trim value for the tower in question 

The thing and instructions here (not mine) http://www.thingiverse.com/thing:745523 do a better job of explaining this than I can ;)

I am unsure how to auto process the DELTA_RADIUS_TRIM but the DELTA_DIAGONAL_ROD_TRIM should be easy to feed the expected size and the result size (while knowing the current rod lengths per tower) into a formula to correct these errors

BTW

>I had to tear my printer apart to replace the plastic pieces that cracked during assembly. It is almost back together

My 3DR is undergoing a "double the print volume" upgrade so i know the feeling :D have you added any bed level sensors to yours ? (i have a very good FSR implementation with a arduino nano)

Roxy-3DPrintBoard, 2015-10-12:
No bed level sensors...   I'm trying to get that all accomplished with firmware and the Z-Probe.

But you can help me with a couple of questions I have:

1) What tower number is typically the 'Back' Tower?  What is typically the 'Left' tower?

2) As you adjust for concave or convex movement, does the shape have any 'noise' on it?  It is for all practical purposes a 2nd order curve (like a parabola) ?  Or is it much more complex with little parabola's added on to the main, bigger parabola ?

BuzzBumbleBee, 2015-10-13:
@Roxy-3DPrintBoard 

1) Front Left is X or 1 , Front Right is Y or 2 and Rear tower is Z or 3

3) Im 99% its always going to move in one big curve along the line in question with no noise along it (only reason i could think that there would ever be "noise" as you described would be if a motor is missing steps :/ )

Roxy-3DPrintBoard, 2015-11-13:
@BuzzBumbleBee

My apologies for moving forward so slow on getting this printer setup.  But I keep going down rabbit holes!   M48 now works on Deltas!   And I've found (or perhaps confirmed) a bug when switching Extruders on Dual Extruder Deltas.   The Printer gets horribly confused about where the Effector is.

But anyway...   I have a couple of questions so I know what to expect:

1. There are video's showing how to do the endstop_adj[] using M666 for each tower.  But I don't understand why this is important.   Changing a particular tower's endstop should just change where the nozzle ends up centered on the bed, right?   I have adjustable bolts that trigger the end stop switches.  I can pretty much dial in where I want each carriage to be and then it is just a question of setting the Z_PROBE_OFFSET so everything works out correctly (for the center of the bed).  Can you comment if this is true?

2. I have the Delta_Radius set pretty well.   My movement is pretty flat from side to side.   But on the Z (gamma) Tower it lifts up a bit on the tower side.    Can you tell me what to expect as I change the Delta_Radius_Trim settings?   I'm guessing if it is only the Z(gamma) tower that is off I'm only going to be changing that trim adjustment.   Do the same rules apply that I would reduce that trim setting to try to flatten the movement to and from the Z tower?   Are there side effects?  If I mess with the Z Delta Radius Trim does it cause ripples else where in the Effector's movements?

AnHardt, 2015-11-13:
@Roxy-3DPrintBoard 
Moving an delta endstop not only moves the zero point but also inclines the plane.

Roxy-3DPrintBoard, 2015-11-13:
@AnHardt 

> Moving an delta endstop not only moves the zero point but also inclines the plane.

Oh???  Maybe that is why I didn't understand things!   I put my nozzle's close to a tower and turned the motors off so I could play with the carriages.   I see what you mean.   The carriages move up and down and cause the Effector to move in an arc.   I was thinking it just pushed the Effector in and out and obviously that isn't true.  That is only true if the other carriages are moving the right amount to compensate for what the other carriage is doing.

THANKS for the help!   That is what I needed to get moving again!

Roxy-3DPrintBoard, 2015-11-14:
@AnHardt
> Moving an delta endstop not only moves the zero point but also inclines the plane.

Thanks again!!! I don't know why I could not see that... I needed you to say it before it made sense to me.   I have to make both of my nozzles level with each other before I can proceed further.  But I have already played around enough after you typed that to know that I can get across the finish line.   I'm actually thinking I can get it good enough I won't need the Auto Bed Leveling to print.  I hope that is the case!    (But don't kid yourself.  I like Auto Bed Leveling!  I'll be getting that to work on my Delta!)

closed 2016-05-04 by jbrazio