MarlinFirmware/MarlinDev #233: Delta G29 - data to crunch

clefranc, 2015-10-06:
Here are a relatively good samples of my G29 results. I'm looking to determine how feedrate has an impact on my bed leveling routine. I'm also looking to determine if removing the U shaped moves during G29 has negative effects.

There is 3 round of data, with 5 test in each, using `prepare_move_raw()` and `prepare_move()` at different feedrate. I'm using http://www.maui-3d.com/cgi-bin/plotG29 to show `RMS Z Err` `Max Z Err` `Avg Z` as a starting point.

* 1st round: `prepare_move_raw()` at 4/8/16 feedrate divisor
* 2nd round: `prepare_move()` at 4/8/16 feedrate divisor
* 3rd round: `prepare_move_raw()` & `prepare_move()` at 4/8/16 feedrate divisor

If someone with statistical background can give me a hint on which method is more efficient (or less precise), it'll be appreciated. You can also tell me that it's just a waste of time.

The first round of G29 tests are done using `prepare_move_raw()` (gives U shaped moves) and a homing feedrate divisor of 4, 8 and 16. I've done each test 5 times, grouped by feedrate:

```
prepare_move_raw only
feedrate = homing_feedrate[Z_AXIS] / 4

RMS Z Err	Max Z Err	Avg Z
0.104		0.296		-0.056
-0.02	-0.02	-0.02	-0.02	0.01	-0.16	-0.16	-0.02	0.23
0.01	-0.04	0.01	-0.03	-0.11	-0.10	-0.03	0.15	0.24
0.10	-0.00	-0.07	-0.04	-0.04	-0.04	0.07	0.09	-0.08
0.09	0.04	-0.01	0.01	-0.07	-0.01	-0.06	-0.02	0.02
-0.01	-0.06	-0.02	-0.08	-0.07	-0.07	0.04	0.06	0.14
0.02	0.01	0.00	-0.10	-0.09	-0.07	-0.08	-0.04	0.00
0.04	-0.10	-0.13	-0.08	-0.07	-0.16	-0.07	-0.05	-0.12
-0.16	-0.16	-0.16	-0.19	-0.13	-0.18	-0.18	-0.07	-0.02
-0.19	-0.24	-0.31	-0.30	-0.19	-0.20	-0.29	-0.20	-0.07

RMS Z Err	Max Z Err	Avg Z
0.105		0.335		-0.045
-0.09	-0.09	-0.09	-0.11	-0.01	-0.16	-0.07	0.02	0.26
-0.03	-0.07	-0.06	-0.07	-0.08	-0.06	0.03	0.16	0.29
0.07	-0.02	-0.07	-0.03	-0.05	0.04	0.06	0.17	0.14
0.07	0.03	-0.01	0.01	-0.10	-0.04	0.01	0.06	0.11
-0.01	-0.06	-0.01	-0.07	-0.03	-0.03	-0.02	0.04	0.11
0.14	0.03	-0.08	-0.12	-0.11	-0.10	-0.10	-0.07	-0.04
0.07	-0.08	-0.07	-0.11	-0.10	-0.10	-0.07	-0.05	-0.12
-0.07	-0.10	-0.13	-0.16	-0.13	-0.18	-0.18	-0.04	-0.00
-0.13	-0.15	-0.22	-0.21	-0.16	-0.26	-0.29	-0.26	-0.01

RMS Z Err	Max Z Err	Avg Z
0.099		0.275		-0.055
-0.12	-0.05	-0.05	-0.05	-0.05	0.03	0.02	0.01	0.01
-0.06	-0.10	-0.01	-0.05	-0.12	-0.01	-0.02	0.01	0.12
0.14	-0.03	-0.08	-0.05	-0.04	-0.05	0.01	0.10	0.13
0.08	0.04	-0.00	0.02	-0.03	0.01	-0.04	0.05	0.14
-0.00	-0.07	-0.06	-0.03	-0.04	-0.03	0.02	0.08	0.15
-0.01	-0.02	-0.03	-0.06	-0.06	-0.14	-0.06	-0.03	0.00
-0.02	-0.05	-0.07	-0.11	-0.10	-0.10	-0.07	-0.07	-0.08
-0.07	-0.08	-0.09	-0.20	-0.16	-0.20	-0.20	-0.11	-0.08
-0.09	-0.11	-0.30	-0.29	-0.22	-0.30	-0.33	-0.30	-0.15

RMS Z Err	Max Z Err	Avg Z
0.084		0.216		-0.05
-0.06	0.01	-0.01	-0.03	-0.05	-0.14	-0.06	-0.03	0.06
-0.05	-0.09	-0.02	-0.06	-0.05	-0.03	-0.02	0.04	0.14
0.04	-0.05	-0.12	-0.09	-0.09	0.08	0.02	0.11	0.14
0.03	-0.01	-0.05	-0.03	-0.07	0.00	-0.04	0.05	0.14
-0.05	-0.07	-0.06	-0.08	-0.09	-0.08	0.01	0.07	0.14
-0.01	-0.02	-0.03	-0.07	-0.03	-0.11	-0.08	-0.04	0.00
0.02	-0.05	-0.08	-0.12	-0.11	-0.14	-0.11	-0.04	-0.09
-0.04	-0.08	-0.12	-0.19	-0.15	-0.17	-0.16	-0.11	-0.06
-0.08	-0.12	-0.27	-0.26	-0.15	-0.20	-0.23	-0.18	-0.11

RMS Z Err	Max Z Err	Avg Z
0.093		0.276		-0.044
0.07	0.08	0.06	0.00	-0.06	-0.19	-0.10	-0.01	0.17
0.05	-0.00	0.01	-0.03	-0.10	-0.08	-0.01	0.11	0.23
0.11	0.01	-0.07	-0.06	-0.06	0.03	0.05	0.13	0.07
0.07	0.02	-0.03	-0.01	-0.06	-0.01	-0.05	0.02	0.09
[... 495 lines cut ...]

RMS Z Err	Max Z Err	Avg Z
0.106		0.336		0.304
0.17	0.37	0.33	0.29	0.34	0.26	0.30	0.38	0.64
0.26	0.23	0.34	0.30	0.25	0.30	0.36	0.53	0.64
0.43	0.30	0.29	0.31	0.27	0.34	0.42	0.49	0.41
0.40	0.37	0.34	0.35	0.27	0.31	0.34	0.39	0.44
0.34	0.28	0.31	0.30	0.33	0.32	0.37	0.43	0.47
0.38	0.35	0.32	0.26	0.26	0.26	0.29	0.32	0.35
0.39	0.27	0.25	0.23	0.25	0.27	0.28	0.31	0.27
0.27	0.24	0.21	0.18	0.17	0.17	0.16	0.30	0.33
0.23	0.19	0.11	0.13	0.21	0.07	0.04	0.05	0.32

RMS Z Err	Max Z Err	Avg Z
0.108		0.344		0.306
0.24	0.33	0.30	0.27	0.36	0.27	0.31	0.39	0.65
0.27	0.27	0.35	0.29	0.27	0.31	0.37	0.54	0.61
0.43	0.31	0.30	0.31	0.28	0.35	0.43	0.48	0.41
0.41	0.38	0.35	0.33	0.28	0.32	0.35	0.39	0.43
0.35	0.29	0.33	0.33	0.34	0.31	0.37	0.42	0.48
0.41	0.36	0.31	0.29	0.27	0.27	0.30	0.33	0.36
0.39	0.27	0.25	0.25	0.25	0.27	0.28	0.33	0.29
0.23	0.21	0.21	0.18	0.18	0.16	0.16	0.29	0.36
0.17	0.15	0.11	0.11	0.21	0.05	0.04	0.05	0.30
```

AnHardt, 2015-10-06:
Compressed that a bit
```
RMS Z Err	Max Z Err	Avg Z	Avg RMS Avg Z Err	Max Z Avg Err	Avg Z
						
0,104	0,296	-0,056		0,097	0,280	-0,050
0,105	0,335	-0,045				
0,099	0,275	-0,055				
0,084	0,216	-0,050				
0,093	0,276	-0,044				
						
0,104	0,324	0,186		0,099	0,326	0,182
0,088	0,329	0,181				
0,104	0,400	0,180				
0,094	0,300	0,180				
0,104	0,278	0,182				
						
0,095	0,307	0,303		0,101	0,324	0,302
0,102	0,321	0,299				
0,101	0,334	0,306				
0,103	0,321	0,299				
0,103	0,338	0,302				
						
						
0,117	0,387	-0,057		0,110	0,377	-0,059
0,105	0,333	-0,06				
0,102	0,432	-0,052				
0,121	0,396	-0,066				
0,105	0,339	-0,06				
						
0,102	0,406	0,174		0,104	0,369	0,175
0,103	0,365	0,175				
0,106	0,4	        0,18				
0,103	0,342	0,17				
0,107	0,332	0,178				
						
0,098	0,312	0,298		0,102	0,335	0,299
0,101	0,337	0,303				
0,1	       0,341	0,299				
0,102	0,34  	0,3				
0,107	0,345	0,295				
						
						
0,095	0,256	-0,054		0,105	0,342	-0,049
0,099	0,341	-0,041				
0,119	0,357	-0,047				
0,11	        0,37	        -0,07				
0,102	0,384	-0,034				
						
0,108	0,387	0,183		0,107	0,381	0,179
0,105	0,369	0,181				
0,101	0,338	0,172				
0,11  	0,411	0,179				
0,109	0,399	0,181				
						
0,107	0,379	0,311		0,103	0,334	0,306
0,098	0,308	0,302				
0,098	0,302	0,308				
0,106	0,336	0,304				
0,108	0,344	0,306				
```

All your variations seem to have no serous influence on the RMS Z Err.
All your variations seem to have no serous influence on the Max Z Err.
It looks as if probing with higher speeds will end up higher.

AnHardt, 2015-10-06:
Sorry. higher divisor = lower speed -> will end up higher.

clefranc, 2015-10-06:
So, basically, with lower speed (high divisor), the probe is triggered sooner, but it doesn't gives much precision over high speed. And whatever method I use (raw or not), I end up with the same result.

There is only one G29 pass before any print, so I was looking for the best compromise. It's not worth much attention.

Thanks

closed 2016-04-18 by thinkyhead