MarlinFirmware/MarlinDev #234: [Proposed] Workflow Modification

Wackerbarth, 2015-10-08:
I would like for us to try a change to the workflow in the hope that it will allow us to improve the interaction of coders and help create improved documentation and testing.

Background: At the present time, someone typically goes off and creates some code and then presents it with a request that it be included in the development branch. Because of the way that github functions, done in this manner, the community has little opportunity to interact with the proposed change until it is essentially complete.

And then, the biggest thing missing is often a good explanation of just what the code is expected to accomplish, and how it is supposed to function.  As a result, that leaves the users guessing, and results in code which has no testing provided.

In an effort to address those issues, we might try a modified workflow. If it doesn't work well, we can always try something else, or return to some past workflow.

<h1> Proposed Workflow </h1>
<h2> Issue </h2>
Before the code is presented, you start by opening an issue.
In the issue you should discuss the problem being addressed and describe the external characteristics of the intended change.
<h2> Establish Feature Branch</h2> 
When there is sufficient agreement that the proposed design is likely to provide an improvement to Marlin, a feature branch is opened in MarlinDev. The branch starts at the current head of the appropriate family (eg: 1.1.x or dev)
<h2> Initial pull request</h2> The primary coder will then push the initial version of the code to this branch..
<h2> Refinement</h2> Others can offer suggested changes by providing pull requests for the feature branch. In the process of developing the feature, it should be considered that adequate documentation is a requirement, Provisions for testing should also be included.
<h2> Testing </h2> At this point in the development, the code needs to be tested by at least one independent tester in addition to the "author". If the feature works as described, and is a useful addition to the Marlin code, Then the branch needs to be rebased to the most recent head of the appropriate family and any styling adjustments, etc. should be made before a final Pull Request is made to merge the new feature into the head of the family.
<h2> Integration </h2> Only the committer responsible for integration should actually merge the feature into the head of the family and thus include that feature as a part of the next version of that family.

Let's give this a try and see how well it works in practice.
Discussion of, and suggestions to improve, the process are encouraged.

AnHardt, 2015-10-08:
@"Deutscher Bürokrat" the texanian.
This workflow will prevent to get any "suggestions" (PRs) from occasional contributors from outside the "inner circle".
Issue
A well reasoned and described PR is as good as a issue.
Establish Feature Branch
If there is already a PR there is already a branch one can make PRs too. Not at MarlinFirmware but in the repository of the contributor.
Initial pull request
Is already done.
Refinement
Yep. PRs to the contributors branch are likely more effective then long and misunderstandable textual  descriptions in a language not you mother tongue.  At least we all "speak" C (and the grammar checkers - compilers are close to perfect)
Testing
Finding testers is a serious problem.The most "new" features depend on special or new hardware. The 4 or 5 people who are at least occasionally try to  review some PRs might not have the required hardware to test. And sometimes it's really hard to get our interest for a theme. If non of the collaborators is daring enough to merge a syntactical correct and not obvious crappy PR that will bum around and will be outdated some day. Our PR history is full of that.
What i want to say is: Don't require a real test. In many cases a review by the merger (who should not be the author) must be enough. The errors then will be detected by the future uses and hopefully fixed by the features author. 
No risk - no progress.

Integration
I'm totally against a self announced integrator!

In the special case of https://github.com/MarlinFirmware/Marlin/issues/2679 where 3 of us are interested to fix that bug, your scheme might work. But note - this bug is around since at least May 2014 and proves how hard it is to get interest for a problem.

We should not choke engagement with bureaucracy!

Roxy-3DPrintBoard, 2015-10-08:
I'm not sure this is accurate.  But It kind of seems like this altered work flow is aimed at making sure nothing gets merged that isn't well thought out and well documented.   If that is the case, can't we just operate that way?

Wackerbarth, 2015-10-08:
@Roxy-3DPrintBoard -- Yes, part of the objective is to improve the quality of the material that gets merged.

We cannot "just operate that way" because the current workflow is missing the intermediate level of having an editable version of the proposed addition. At present, that level of interaction is on various remote repositories. That isn't working because, within the Marlin developers, we have no way to make needed changes before something gets merged.

This workflow adds that intermediate step. If you look at the github description of workflows, you will find that the proposed workflow is much closer to those that they describe.

AnHardt, 2015-10-08:
Says the guy who is committing withisout PRs and resetting branches.

If the aim is "making sure nothing gets merged", making this scheme mandatory, seems to be a reliable strategy.

clefranc, 2015-10-08:
> "making sure nothing gets merged"

Saying "making sure nothing gets merged without Richards approval" is more accurate IMHO.

Wackerbarth, 2015-10-08:
> On Oct 8, 2015, at 4:49 AM, AnHardt <notifications@github.com> wrote:
> A well reasoned and described PR is as good as a issue.
> 
I agree. However, we rarely, if ever, get such contributions. Instead, from an initial proposal, the good qualities you describe have to be developed.
> Establish Feature Branch
> If there is already a PR there is already a branch one can make PRs too.
> 
That would be true IF the initial PR is appropriately assigned, and then maintained, against the appropriate primary branch.
Unfortunately, most would-be contributors fail to understand the need, or usage, of staging branches. They start with the assumption that their, incomplete, untested, and, perhaps ill-advised, addition should be applied directly, and immediately, to the main, published, actively distributed, “production” version of the product.

If github supports a way to “move” a PR to another branch without recreating it, I haven’t found how to do so.
> Not at MarlinFirmware but in the repository of the contributor.
> 
That doesn’t work because the “inner circle” has no way to repair the problems when it is in that form. And all too often, an author wants to present some code with the expectation that someone else will “finish” the needed integration details for them.

Take the recent correction of the “EXTRUDER_SWITCHED_OFF” message.
In reality, that is a change of ONLY one line of code. The commit that makes the correction should not affect anything else.

However, the change was initially presented with a number of additional artifacts and it remains un-corrected on the contributor's repository. That, in large part, is why it had not been merged. Now, finally, you resurrected it as a new PR. However, even then, you failed to clean out all of the unrelated changes.
> Yep. PRs to the contributors branch are likely more effective then long and misunderstandable textual descriptions in a language not you mother tongue. At least we all "speak" C (and the grammar checkers - compilers are close to perfect)
> 
“C” is a very poor language to use to describe intent. Using it for that purpose presumes that the code is a perfect implementation of an otherwise unspecified design. Thus, “by definition”, because there is no description to the contrary, there are no bugs in the code.
> Finding testers is a serious problem.
> 
Agreed. 

> In many cases a review by the merger (who should not be the author) must be enough. The errors then will be detected by the future uses and hopefully fixed by the features author.

Unfortunately, I have to agree with you on that point. But I think that much of this is caused by the rush to overload the firmware with “warts” that target the special interest of the author, and no one else.

Roxy-3DPrintBoard, 2015-10-08:
I don't know if this is the case, but I kind of feel like these extra check points are being suggested because I'm writing some code to help people do Physical Bed Leveling of new Delta Printers.   I did not work through any of the issues or factor in any of the current development plans.  I saw a need and decided I wanted some code to help me.   I am doing the code nice so other people can benefit from it but I'm not going through all the process you want me to do. 

And the truth is, I don't want to work through the process of trying to get everybody in agreement that what I'm doing is perfect for them to.   If I have to do that, I'm just going to get out the feeler gauges and level the bed that way.

The big problem that I see is our Collaborators are very senior engineer types.   They know their stuff.  They are volunteering their time to help Marlin because it is an interesting hobby for them.   We can't put barriers in their way unless they add a lot of value. 

I think we have to trust our Collaborators to do the 'Right' thing.   If one of them repeatably merges in bad code, lets have a discussion about it.  But my feeling is this group of individuals does not need (or want) to be ruled with an iron fist.   

And really...   GitHub is a Source Code Control system.   If something crazy happens, we can roll back and undo stuff.   (This should hardly ever happen if we have good people doing the merges!!!)

Wackerbarth, 2015-10-09:
@Roxy-3DPrintBoard --"I kind of feel like these extra check points are being suggested because I'm writing some code..." It is not at all anything about you. If you have been following any of the discussion about bed leveling, you should notice that there is quite a bit of confusion caused by the lack of a specification for the behavior. The same can be said about thermal runaway and display update. If rather than just some code, it would be a lot easier if we had some written description of the behavior that was intended and the reasoning that went into those choices.

Roxy-3DPrintBoard, 2015-10-09:
> If you have been following any of the discussion about bed leveling, you should notice that there is quite a bit of confusion caused by the lack of a specification for the behavior. The same can be [...]

Perhaps this is worth a short discussion.   With regard to the Auto Bed Leveling, I don't know who wrote that code originally.  But they never would have been able to get complete agreement on the direction and use cases if they had tried to do that.  What they did produce and donate to the Marlin community has proven to be very very valuable.  

I do admit there is value to having fully specified behavior for code that is being developed (and maintained).  However, there is a problem with that.  Until the code and the machines behavior is fully understood, it is very difficult to produce that kind of document.   Let me give a very simple example with the Manual Bed Leveling Probe code I'm doing.   Originally, I wanted to probe the Z-Height of each of my 3 support screws.  That was useful.  With that information I would be able to immediately see which screw needed to move in which direction.  Then later I realized, I needed a line of probed points coming from each tower.  And I needed that line to be optionally the nozzle or the probe itself because the angle of the diagonal rods is going to change depending upon which method was used.  The code is fairly clean right now.   But I think I can further simplify and clean up the code by not having any of the points nailed down with #defines.   I now think I can make it totally specified (with reasonable defaults) at command time.

There is no way I could have been smart enough to produce that document in advance.  I had to work through the issues in order to learn enough that I fully understand the problem.

And I'll suggest one more idea gets some thought.   Is it possible the problems with Auto Bed Leveling is not a result of not having a design specification but instead having 'Simple' changes made without sufficient testing of the changes?  And then this process was repeated and the errors compounded?   I suspect the 2nd explanation is closer to the truth.  This is a big part of the reason I want the G28/G29 code broken out by machine type.  Besides being easier to understand, people will only be making changes to code that they have a machine available to test the updates.

Wackerbarth, 2015-10-09:
@Roxy-3DPrintBoard -- I am sure that you are correct in observing that many of the problems with bed leveling are a direct result of changes piled on top of changes without adequate testing.

I also agree that, unless you are working on government contracts, even in real business environments, getting adequate design documentation always seems to be an issue. I don't expect us to solve that with a bunch of volunteer programmers.

However, I do think that we should expect SOME documentation. One of the reasons that we get patches piled on patches is that one programmer doesn't understand what the previous programmer was trying to do and, because of this ignorance, takes off in another direction, thinking that he is making things better.

I do recognize that you don't know exactly HOW you are going to end up solving a particular problem. You (as with all of us), are likely to try something only to find that, in practice, the idea wasn't as good as you had imagined. This will result in a modification of the approach, or perhaps abandoning it completely.
And all of that is fine, and just a part of the development process.

I'm just trying to address these problems. If you have a better suggestion, I'm all for it.

1) The submitted code is offered as a pull request against the wrong branch. We currently have three branches and really need all of them. But each on has some differences from its predecessor. The patches need to be made relative to the branch where they will be installed initially.

2) Submitted patches contain useful code, but they also include extraneous material. When they are in some other repository, only the owner of that repository can merge corrections. And when they fail to do so, then there are complaints that something didn't get integrated.

3) By the time a feature is complete, we really NEED a sufficient description so that we know what to expect if we attempt to test it. We also need documentation of any configuration options so that a user understands the "what and why" of the activation of the feature.

4) In order to be able to make significant changes to some of the functionality, we need to be able to implement the change in steps. While this process is going on, we need to have others avoid making other incompatible changes. This can be accomplished only by communication among those involved.

I welcome your suggestions as to how we can better address these issues.

Roxy-3DPrintBoard, 2015-10-09:
> 1) The submitted code is offered as a pull request against the wrong branch. We currently have three branches and really need all of them. But each on has some differences from its predecessor. The patches need to be made relative to the branch where they will be installed initially.

This probably gets better as things move forward.  It is fairly recent that there even was a 'Development Branch'.  I can see how this happens.   I'm sympathetic on both sides of this.   I haven't thought this through, but I think my reaction is:  This can't be tested code if you are trying to put it into the wrong branch.

> 2) Submitted patches contain useful code, but they also include extraneous material. When they are in some other repository, only the owner of that repository can merge corrections. And when they fail to do so, then there are complaints that something didn't get integrated.

I can see how this happens.  But don't we only have two repositories ?   I was under the impression our collaborators can merge either branch.   Assuming this is true, my thoughts are everybody is volunteering their time.  If it takes too much time to see what is valuable because it is mixed in with random other stuff, it is very possible nobody wants to spend the time to sort through the mess.

> 3) By the time a feature is complete, we really NEED a sufficient description so that we know what to expect if we attempt to test it. We also need documentation of any configuration options so that a user understands the "what and why" of the activation of the feature.

This should happen naturally.   Any code block of size should have a header comment that tells us about it.   If it is a new command, it should fully describe the parameters and theory of operation.   This does not mean the code itself is without comments just because the header comment is there.

If a situation happens where the code block is missing the 'suggested' comments I think we end up in a position where a choice needs to be made.   Is the code valuable enough that somebody offers up their time to get the code commented?  if so, maybe we grumble and just do it.  If not maybe it doesn't get merged.

> 4) In order to be able to make significant changes to some of the functionality, we need to be able to implement the change in steps. While this process is going on, we need to have others avoid making other incompatible changes. This can be accomplished only by communication among those involved.

I'm a believer.   Let's get G28 & G29 broken out for starters.  And once they are broken out, cloned into a version for each machine type.

Wackerbarth, 2015-10-09:
Roxanne,

The repository issue isn’t between OUR two repositories. The whole purpose of my suggestion is to get the code INTO our repository rather than off on some private repository.

Just considering the Development (ignoring the Release/Release Candidate) side of things, at the present time, we have what amounts to only one “gate" through which everything tries to get on the “dev” train. All code is developed “outside” and then wants to get in the only line. What I am proposing is to create a separate “gate” for each feature and then, through pull requests, get it refined “inside” our repository. That way, it is available for multiple coders to contribute corrections before we finally merge it (with a final merge request to “dev”).
This solves (1) because the “gate” is created in the proper spot; (2) because the code has been moved from the private repository to our public one where we have access to make changes; It also addresses (3) because it is easier to say “we could merge it if …”; and (4) because it makes the intended development more transparent.

Roxy-3DPrintBoard, 2015-10-09:
OK.  That explanation is helpful.   I need to go back to the very beginning and re-read everything.

Roxy-3DPrintBoard, 2015-10-10:
Here is a thought...   Instead of having just one head to the Development Branch...   When we receive a Pull Request, can we clone the head of the Development Branch and merge the changes there?  

The intent would be to get the changes into 'our' system but still keep them separate from the main branch.   This would give us the opportunity to test things a little bit and make sure things are stable.  If development is hot and heavy, we may have multiple clones of the head of the Development Branch, each with its own Pull Request merged into it.

As confidence grows in any of these Pull Requests, we merge them with the head of the Development Branch and the Development Branch moves forward.

Does this accomplish the goal of having multiple gates to let Pull Requests travel through?

Wackerbarth, 2015-10-10:
In most respects, that is exactly what I am doing. However, as far as I can tell, you cannot “move” the initial pull request to another branch. Thus, it is stuck to the branch to which it was initially posted. If two coders post changes to the same branch, whether it is called “master” or “XYZ”, there is no way to separate them.  That is why I am suggesting that the coder request that a branch be made for their feature. That branch will be nothing more than a new branch name pointing to the then current head of the development branch. But, each feature then has its own branch on which it can progress until it is tested and ready to be merged into the true development branch.

Roxy-3DPrintBoard, 2015-10-11:
That seems like a serious limitation of GitHub that things can't be merged into other branches.  That just doesn't seem right.   The problem is, the person doing the changes won't know they need another branch until very far into development.  And from what you are saying, we can't just move the changes to another branch.

There must be a way to do a 3-way merge such that the new code could be dropped into a cloned branch.   It certainly is possible to do that with the Visual Diff tools I use.  GitHub must have a way to handle this.  Would it be worth an email to GitHub to ask if we are missing something obvious?

Wackerbarth, 2015-10-11:
The changes can be moved. But to do so, you have to create a new PR (or, to the objection of some, bypass the public acceptance of the PR, etc. and do the merging in a hidden transaction).

It all has to do with the workflow design of Github. That design expects development to be done on branches and then offered to be merged into the specific destination branch.

Roxy-3DPrintBoard, 2015-10-11:
When a Pull Request comes in and it looks reasonable...  Would it be that hard for us to fork the head of the Development Branch and apply the changes manually to the branch?  I understand it isn't automatic.  But all of the information is there showing the differences between what was submitted and what the Development Branch looks like.    

I wonder if we aren't just making extra work by considering this approach.  If we can start to get Marlin (and especially Marlin_main.cpp) broken apart and the various parts put into folders, a lot of Pull Requests won't even overlap in the same modules.

Would our time be better spent making it a priority to get Marlin cleaned up?

closed 2016-05-04 by jbrazio