MarlinFirmware/MarlinDev #238: EXTRUDER_RUNOUT_PREVENT breaks bed autoleveling in 1.1.0-RC2

m0n5t3r, 2015-10-10:
... and I guess other long-running commands as well;

steps to reproduce:
* set temperature above `EXTRUDER_RUNOUT_MINTEMP`
* run G28, G29 (the latter with 9 points)

on my Electron Prusa I3, when it gets to the middle test point it just starts raising the Z axis very slowly until it gets a reset

AnHardt, 2015-10-15:
Its probably not the middle test point but  `EXTRUDER_RUNOUT_SECONDS` is over.

The work around is to increase `EXTRUDER_RUNOUT_SECONDS` to be longer then G29 needs to finish.
The fix would be to update `previous_cmd_ms` during G29, let's say for every probe point. We don't want a G29 to be interrupted.
I can't see why the z-axis should rise, but an extrusion is bad enough.

AnHardt, 2015-10-15:
Fix is already included in https://github.com/MarlinFirmware/Marlin/pull/2686 to prevent stepper releases.

m0n5t3r, 2015-10-16:
Thanks.

Yup, I assumed it's related to the timeout, not the actual command (see my point about other long running commands)

I guess the reason for the Z rise was this sequence of events:

* G29 runs happily, touches down, starts raising Z to confirm elevation
* extruder runout protection kicks in
* G29 is aborted somehow, but the routine that sends pulses to the stepper keeps running
* the rest of the firmware assumes G29 is still ongoing, so it won't respond to input

Luckily in my case the thing happened while Z was raising and X / Y were stopped, otherwise the result would be the print head crashing into the bed / axis ends; the latter would be less tragic because belts are elastic and will skip (don't ask me how I know that :p), but the Z axis can deliver quite a lot of down force.

AnHardt, 2015-10-16:
@m0n5t3r 
Sorry - was not in. But now is. Makes perfect sense in that context.

Roxy-3DPrintBoard, 2015-10-16:
Yeah...  When I first read the bug report I was scratching my head saying "The Extruder Runout stuff isn't even checked in G29???"    And then after the 'Verified Bug' label got put on it, it all made sense.

AnHardt, 2015-10-16:
Both EXTRUDER_RUNOUT_PREVENT and DEFAULT_STEPPER_DEACTIVE_TIME are checked in idle(), both rely on previous_cmd_ms.
previous_cmd_ms is reset once in G29 - that's in setup_for_endstop_move() just before the loops over the probing points begin. Probing 9 or more point can easily last longer then the default EXTRUDER_RUNOUT_SECONDS 30 seconds. So - yes a refresh is needed.

AnHardt, 2015-10-16:
I still have no real clue why the z axis is moving.  
https://github.com/MarlinFirmware/Marlin/blob/RCBugFix/Marlin/Marlin_main.cpp#L7008
does not look so bad.
When we reach this line we had. if all went right, no fresh movement command for 30 seconds and the motors do not move anymore. The plan buffer is empty.  current_position[] and destination[] are the same. 
What happens if the things went wrong and we reach the code when the buffer is not empty,  current_position[] and destination[] are not the same?
Does it make sense to add a additional condition like if(movesplanned() == 0) to add some additional safety here?

Roxy-3DPrintBoard, 2015-10-16:
@AnHardt   Have you verified the Z Axis moving on your machine?   If it is really doing that, that is kind of scary.   Looking at the code it must be somewhere in  manage_inactivity() when it happens.  But it sure doesn't seem like that should be happening.

Any way...  The reason I'm asking is if you can make it happen on your machine, it might be possible to add some debug statements to give us a clue where (and why) this is happening.

AnHardt, 2015-10-16:
Log 

c lines: current_position[]
d lines: destination[]
columns: x, y, z,  (e+ EXTRUDER_RUNOUT_EXTRUDE * EXTRUDER_RUNOUT_ESTEPS / axis_steps_per_unit[E_AXIS]), (EXTRUDER_RUNOUT_SPEED / 60. * EXTRUDER_RUNOUT_ESTEPS / axis_steps_per_unit[E_AXIS]), ((int)active_extruder)

```
22:16:41.203 : N51 G0 X0 Y0 Z10 *71                  // move to [0,0,10]
22:16:41.205 : echo:N51 G0 X0 Y0 Z10 *71
22:16:41.209 : ok
22:18:28.175 :c 0.00 0.00 10.00 2.44 0.61 0 // first extrusion - move was very slow
22:18:28.175 :d 0.00 0.00 10.00 2.44 0.61 0
22:18:28.175 : 
22:18:43.181 :c 0.00 0.00 10.00 2.44 0.61 0 // second extrusion - DEFAULT_STEPPER_DEACTIVE_TIME was 15
22:18:43.186 :d 0.00 0.00 10.00 2.44 0.61 0
22:18:43.186 : 
22:18:58.176 :c 0.00 0.00 10.00 2.44 0.61 0
22:18:58.180 :d 0.00 0.00 10.00 2.44 0.61 0
...                                                                  // and so on
22:21:58.185 :c 0.00 0.00 10.00 2.44 0.61 0
22:21:58.185 :d 0.00 0.00 10.00 2.44 0.61 0
22:21:58.185 : 
22:22:13.191 :c 0.00 0.00 10.00 2.44 0.61 0
22:22:13.191 :d 0.00 0.00 10.00 2.44 0.61 0
22:22:13.191 : 
22:22:28.186 :c 0.00 0.00 10.00 2.44 0.61 0
22:22:28.186 :d 0.00 0.00 10.00 2.44 0.61 0
22:22:28.186 : 
22:22:37.020 : N88 G29 V4 *64
22:22:37.028 : echo:N88 G29 V4 *64
22:22:37.028 : G29 Auto Bed Leveling
22:22:41.316 : Bed X: 10.000 Y: 51.000 Z: 0.378
22:22:42.991 : Bed X: 100.000 Y: 51.000 Z: 0.595
22:22:44.552 : Bed X: 190.000 Y: 51.000 Z: 0.882
22:22:47.043 : Bed X: 10.000 Y: 120.000 Z: 0.433
22:22:48.705 : Bed X: 100.000 Y: 120.000 Z: 0.628
22:22:50.372 : Bed X: 190.000 Y: 120.000 Z: 0.841
22:22:52.026 :c 10.00 139.00 0.84 2.44 0.61 0  // G28 is interrupted
22:22:52.026 :d 0.00 0.00 10.00 2.44 0.61 0 // nozzle moves extremely slow towards [0,0,10]
22:22:52.026 : 
22:23:07.030 :c 10.00 139.00 0.84 2.44 0.61 0 // more and more extremely slow moves pile up
22:23:07.034 :d 0.00 0.00 10.00 2.44 0.61 0
22:23:07.034 : 
22:23:22.024 :c 10.00 139.00 0.84 2.44 0.61 0
22:23:22.028 :d 0.00 0.00 10.00 2.44 0.61 0
...
22:26:37.121 :c 10.00 139.00 0.84 2.44 0.61 0
22:26:37.121 :d 0.00 0.00 10.00 2.44 0.61 0
22:26:37.121 : 
22:26:37.125 :c 10.00 139.00 0.84 2.44 0.61 0
22:26:37.125 :d 0.00 0.00 10.00 2.44 0.61 0
22:26:37.126 : 
22:26:37.130 :c 10.00 139.00 0.84 2.44 0.61 0
22:26:37.133 :d 0.00 0.00 10.00 2.44 0.61 0
22:26:37.133 :                          // until something is flowing over
22:26:37.133 : echo: too long extrusion prevented
22:26:37.298 : Error:MINTEMP triggered, system stopped! Heater_ID: bed
22:26:37.298 : Error:Printer halted. kill() called!
22:26:37.502 : echo: too long extrusion prevented
```

To remember the definition is:
```
void plan_buffer_line(float x, float y, float z, const float &e, float feed_rate, const uint8_t extruder)
```

G29 (grid) does not use destination[]. destination[] is untouched and shows the coordinates of the last regular move. At least the last regular move is normally a save place. The move to there with the default extrusions speed of the EXTRUDER_RUNOUT feature takes for ever

Experiments with different times tell us, that current_position[] is updated and therefor is much better then destination[].
As said before after normal moves  current_position[] ==  destination[].

So ether the questionable code should look like:
```
      plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS],
                      current_position[E_AXIS] + EXTRUDER_RUNOUT_EXTRUDE * EXTRUDER_RUNOUT_ESTEPS / axis_steps_per_unit[E_AXIS],
                      EXTRUDER_RUNOUT_SPEED / 60. * EXTRUDER_RUNOUT_ESTEPS / axis_steps_per_unit[E_AXIS], active_extruder);
```
or destination[] should be updated during G29.
To not make G29 more complicated than it already is i suggest the first.

Haw the change influence the surrounding code (saving some variables) i'll have to see in the over next week.

Until then.

Roxy-3DPrintBoard, 2015-10-16:
> G29 (grid) does not use destination[]. destination[] is untouched and shows the coordinates of the last regular move. At least the last regular move is normally a save place.

Careful here!    G29 uses probe_pt() which calls do_blocking_move_to_z() and do_blocking_move_to_xy().   Both of these get their work done by calling do_blocking_move_to() which for Deltas does:

~~~~
      	destination[X_AXIS] = x;
      	destination[Y_AXIS] = y;
      	destination[Z_AXIS] = z;
      	prepare_move_raw(); // this will also set_current_to_destination
      	st_synchronize()
~~~~

I'll look at it more carefully in the morning.  But I have to run out the door right now.

AnHardt, 2015-10-16:
How ever. Preventing the extrusion by resetting the timer has priority. 
Merge https://github.com/MarlinFirmware/Marlin/pull/2686

clefranc, 2015-10-16:
Is inserting idle() in probe_pt() can help?

Roxy-3DPrintBoard, 2015-10-17:
> Is inserting idle() in probe_pt() can help?

I am not aware of any race conditions in probe_pt().  But if the first probe_pt() takes extra long I could believe an idle() would be a good thing to put in there.

> How ever. Preventing the extrusion by resetting the timer has priority. 
Merge MarlinFirmware/Marlin#2686

@AnHardt   Are you saying this is the LCD Panel doing this funny Z-Axis move ???

KiteLab, 2015-10-17:
The primary error is, not refreshing `previous_cmd_ms`for a too long time, during G29. For the same reason  `DEFAULT_STEPPER_DEACTIVE_TIME` can run out. This is one of the problems addressed in  MarlinFirmware/Marlin#2686. Please reread https://github.com/MarlinFirmware/Marlin/issues/2661#issuecomment-145260993

Roxy-3DPrintBoard, 2015-10-17:
> The primary error is, not refreshing previous_cmd_msfor a too long time, during G29. For the same reason DEFAULT_STEPPER_DEACTIVE_TIME can run out. This is one of the problems addressed in MarlinFirmware/Marlin#2686. Please reread MarlinFirmware/Marlin#2661 (comment)

I have re-read both of those threads.  I did not realize they were related to this.   I suspect G29 is not the right place to fix this.   If we are going to make calls to something we should do it at a lower layer like run_z_probe().   The reason being other GCodes make extensive use of run_z_probe() also.   For example: M48.     I suspect right now M48 can be made to fail in a similar way.

AnHardt, 2015-11-03:
@Roxy-3DPrintBoard 
Shifted the refresh of `previous_cmd_ms` to `run_z_probe()`.
Now all the long lasting g-codes, probing multiple points, should be fixed.

Roxy-3DPrintBoard, 2015-11-03:
I just noticed something.  This thread is in MarlinDev but it is talking about a bug in RC2.   The truth is the bug is in both places.  Is the plan to patch both branches?

AnHardt, 2015-11-03:
@Roxy-3DPrintBoard 
Of cause! Bugs should be fixed in Marlin/RCBugFix and MarlinFirmware/master. But due to the convinience of the environment i usually start at  Marlin/RCBugFix. Until now both branches are similar enough to just cherry-pick from the one to the other.

thinkyhead, 2016-04-18:
@m0n5t3r If you have a chance to test the latest code in `RCBugFix` this issue should be fixed now. If you still have any problem, feel free to re-open this issue.

closed 2016-04-18 by thinkyhead