MarlinFirmware/MarlinDev #244: Travis checks broken

AnHardt, 2015-10-15:
#241 should have failed.

AnHardt, 2015-11-02:
Can't find where CUSTOMIZATION is evaluated or included.
I can find the define
```
- DISPLAY=:1.0 ~/bin/arduino --verify --board marlin:avr:mega  --pref build.extra_flags=-DCUSTOMIZATION="configurations/travis/Customization_01.h" examples/MarlinFirmware/MarlinFirmware.ino
```
but not the use.

Wackerbarth, 2015-11-02:
You seem tohave an obsolete version of the boards.txt file.
On my machine, I have entries such as
`mega.menu.marlin.travis21.build.unique_id_flags=-DCONFIGURATION_FILE=configurations/travis/Customization_21.h`
which define `CONFIGURATION_FILE` rather than `CUSTOMIZATION`

I think that `https://github.com/Wackerbarth/MarlinDev/tree/MarlinFirmware`
has the most recent version.

There are two things keeping me from moving this into the MarlinFirmware/MarlinDev repository.
1) We don't seem to have a viable strategy for the way to document this without affecting the 1.1 Release.  -- Things differ between 1.0, 1.1, and 1.2. Although there is a lot in common, each needs its own documentation. I think that this documentation needs to be organized in a way that newly arrived users can get to the appropriate documentation without having to "filter out" what they need from the documentation that applies to the other versions.

2) I had hoped that @justmyopinion would help me determine which LCD libraries need to be special to Marlin and which ones can use the versions provided by the Arduino Library Manager.  Once that is determined, I can build the ones that are special into the download and provide the instructions to install the others with the library manager.

AnHardt, 2015-11-02:
It's not me or my installation, it's MarlinFirmware/MarlinDev
The Travis tests happen on the server.
https://github.com/MarlinFirmware/MarlinDev/blob/master/.travis.yml#L59

CUSTOMIZATION seems not to be defined inside the sketch at all.

closed 2015-11-05 by AnHardt