MarlinFirmware/MarlinDev #245: Using higher extrude temperatures

atunguyd, 2015-10-16:
Hi sorry for logging this as a request but my google skills must be failing as I cannot find anything referring to this.

I am trying out some filaments that will require higher extrude temperatures, how do I modify Marlin to allow me to do this? Currently the maximum temperature I can set the hotend to via my LCD screen is 260 - I have not tried this with Gcode yet but I wanted to check what marlin's maximum is with regard to what it can be set to and also what it can sense (ie thermocouple tables).

Thanks

Anthony

Roxy-3DPrintBoard, 2015-10-16:
You need to rebuild the Marlin firmware.  In the Configuration.h file you will find these numbers.  You can change them to what ever makes sense.

~~~~
#define HEATER_0_MAXTEMP 240
#define HEATER_1_MAXTEMP 240
#define HEATER_2_MAXTEMP 240
#define HEATER_3_MAXTEMP 240
#define BED_MAXTEMP 15
~~~~

atunguyd, 2015-10-20:
Sory I thought I commented and closed this already.

My comment was perhaps the comments in configuration.h should be a bit more desciptive, currently as they stand they seem to indicate that HEATER_X_MAXTEMP is a safety setting and not the maximum that you can use.

closed 2015-10-20 by atunguyd