MarlinFirmware/MarlinDev #246: Consolidate Delta Issues

BuzzBumbleBee, 2015-10-19:
Hello All,

Iv been busy rebuilding / tooling my delta printer (3DR) and now its all built and calibrated I would like to start hunting down some of those painful delta bugs / issues / key missing features, I am looking at putting a couple of evenings a week into getting deltas up to scratch within marlin.

So can we make a list of all known issues affecting deltas here ?

FWIW my new delta specs are as followed (so you know what i can directly test) :- 

* Based on a 3DR (now much bigger and more solid frame)
* Build diameter 270mm (circle glass build plate)
* Build height 350mm aprox
* Heated bed (covers 200mm of the build area)
* FSR + arduino nano based auto level
* RAMPS 1.4 electronics
* Home made dual extruder will be fitted in the next few weeks

Roxy-3DPrintBoard, 2015-10-19:
Right now I'm fighting an Auto Bed Leveling issue.   There is a problem with how the code is setup for Delta's.  Because the Delta's typically use a Spider Ring it means the X & Y Offset From Extruder is usually fairly large.    And depending on which side you are probing, you can push the nozzles off the bed.

When that happens, the probe stops and all of the carriages go up.   Arrrrggghhhh!

Natealus, 2015-10-19:
@Roxy-3DPrintBoard I'm not on current Dev right now because I haven't been testing ABL for a while. I'm currently printing with the Release Candidate. But when I was I had to adjust the  #define DELTA_PROBABLE_RADIUS (DELTA_PRINTABLE_RADIUS - 60) because my capacitive probe sticks out 60mm in one direction. I also have an E3D Kraken so that takes up a chunk of space from center too. 

One thing I'm not 100 percent on is are we offsetting from effector center or from where my nozzle is which is offset 10mm in 2 directions depending on which nozzle you're talking about.

Another issue I was having was that if I set extruder offset in the firmware, it caused a failing issue in dual extrusion where Z would be about 3mm off when switching between the two filaments which obviously...didn't stick lol. So I comment out the extruder offsets in the firmware and let the slicer take care of offset and that fixed the Z-hopping during nozzle switch issue.

Couple of oddities that I haven't had time to notate and work on cause I've been catching up on production. Had to go down to 50 microns to print these really small train pieces one of my relatives wants to replace Dominoes trains lost. I made it hard on myself by using a higher detailed model. Even 100 microns was not sufficiently small enough to print the details for the build size at 11x21x18mm

BuzzBumbleBee, 2015-10-19:
@Roxy-3DPrintBoard reducing DELTA_PROBABLE_RADIUS should fix your issue as @Natealus suggested

Roxy-3DPrintBoard, 2015-10-19:
> @Roxy-3DPrintBoard reducing DELTA_PROBABLE_RADIUS should fix your issue as @Natealus suggested.

Yes, Thank You!    But there are two issues with this.  First, I want to figure out a good way to clamp that movement.  The problem is the extruder is going into an 'undefined' region and the sqrt() of a negative number.  At that point everything is going to be wrong and never recover.

The other issue is I want the correction matrix to reflect as large of area as possible.   So if the Z-Probe is offset, I understand that limits how far the bed can be probed in the other direction.  But we should still be able to get to the edge of the bed in the probe's direction.

(I'm not just trying to get the printer up and running...  I'm trying to improve some of the stuff that is going on too...)

thinkyhead, 2016-04-18:
The code from `RCBugFix` will be migrating here and getting reorganized in the same manner as the MarlinKimbra fork. Once that code is moved over we will need to revisit and see what issues are left for Delta. We fixed a lot.

closed 2016-04-18 by thinkyhead