MarlinFirmware/MarlinDev #247: Boot Loader problems???

Roxy-3DPrintBoard, 2015-10-19:
I was debugging an Auto Bed Leveling issue on the Delta Printer I'm trying to bring up.   All of a sudden, I can't upload any new sketches to it.    The compile and link goes OK, and I get the status message about how much free space I still have and it says it is uploading.  But the upload never finishes.

I tried going to Tools/Burn_Bootloader  and that doesn't work either.   It says:

avrdude:  stk500_recv(): programmer is not responding.    

I have my Board set to Arduinio/Genuino Mega or Mega 2560
and I have my Programmer set to AVR ISP

That is what I've always used on this board.   Can somebody throw me a life line?

CONSULitAS, 2015-10-19:
Hi Roxy,

this sounds in fact like a crashed boot loader. I had this problem once with an arduino uno.

It helped to take an other arduino and program is as an ISP and flash the boot loader again. Everything worked fine after this.
https://www.arduino.cc/en/Tutorial/ArduinoISP
http://forum.arduino.cc/index.php?topic=126160.0

But for your board you will have to find the correct pins.

Hope this fixes your problem! Hold this line tight! :-)

nitewing76, 2015-11-18:
I realize this is almost a month old, but I have been drowning in Solidworks.  lol

All bootloader problems I have had so far, have always ended up being a problem with my thermistor (loose connection or broken wire).

Roxy-3DPrintBoard, 2015-11-18:
> All bootloader problems I have had so far, have always ended up being a problem with my thermistor (loose connection or broken wire).

Are you saying that bad connections to your external hardware caused problems when trying to flash new firmware to the AVR ?

KiteLab, 2015-11-19:
He's likely saying, that he is using an older version of Marlin, where the machine was killed before the display was initialized, when a min- or max- temp_error occurs because the thermistor was not connected. That maked the impression the flash went wrong. That's fixed

closed 2016-04-22 by thinkyhead