MarlinFirmware/MarlinDev #254: manual G1 command movement of E inconsistent

Abrinkmark, 2015-10-27:
When changing filament I send "G1 E400 F6000 " to feed new filament through the bowden tube.
This works as expected and 400mm of filament is moved in. But this only works once! 
Second time I send "G1 E400 F6000 "  the movement is reversed! and filament is rectracted, why is this?
I can still feed filament using the "extrude" button in Octoprint,  this sends "G1 E# F300" and this moves correct.
If i reset the printer "G1 E400 F6000 " works again.

Cartesian printer using mega2560/ramps1,4 running (Marlin 1.1.0 Release Candidate 2 - 29 September 2015) and Octoprint as host.

AnHardt, 2015-11-02:
For the second  "G1 E400 F6000 " i'd expect no movement at all.

When you push one of the move-buttons on a host a sequence of commands is send. Like
```
13:01:35.973 : echo:N92 G91 *10
13:01:35.973 : echo:N93 G1 Z10 F100 *30
13:01:35.977 : echo:N94 G90 *13
```

The G91,G90 commands switch between relative (G91) and absolute (G90) coordinate mode.
Maybe some M114 can help you to find out what is going on.

thinkyhead, 2016-04-18:
@Abrinkmark If you have a chance to test the latest `RCBugFix` code, it's possible you will not see this behavior any more. We've fixed a lot of issues there. If you continue to have a problem, let us know and we'll have another crack at it!

closed 2016-04-18 by thinkyhead