MarlinFirmware/MarlinDev #255: Dual X/Y/Z

caenan0, 2015-10-27:
Not so much an issue but rather a comment in case anybody has been wanting to build a similar setup - my design has dual Y/Z and dual X carriage all enabled within Marlin and despite the sanity checks usually preventing this from happening, it compiles and runs prints without issue (so far) if you just comment out those errors.

Wackerbarth, 2015-10-28:
How do you have things set up? By that, I mean, which pins control the various stepper motors?

caenan0, 2015-10-28:
Using an Azteeg X3 Pro with 8 stepper outputs, so basically 2 for each axis XYZE.

AnHardt, 2015-11-02:
Interesting concept with that 4? 6? 8? carriages.
Or do you mean "dual Y/Z and dual X __stepper__" ?

caenan0, 2015-11-02:
@AnHardt, the latter.

Roxy-3DPrintBoard, 2015-11-02:
Is this just to increase the force moving the carriage around?   Is this to get higher travel speed?

caenan0, 2015-11-02:
For the Y/Z, I needed the extra power of an extra motor to drive large components (my design has just under 5 cubic feet of build volume). If you're not familiar with the dual X arrangement, you can check out http://www.thingiverse.com/thing:115885.

Roxy-3DPrintBoard, 2015-11-02:
That is cool!  With some Slicer changes you could optimize a lot of things.

caenan0, 2015-11-27:
So if anybody wants to actually see this in action, I've got a kickstarter video up here: https://www.kickstarter.com/projects/aon3d/aon-real-industrial-3d-printing

closed 2016-05-04 by jbrazio