MarlinFirmware/MarlinDev #256: Questions about Arduino

Roxy-3DPrintBoard, 2015-10-28:
I have a few questions about Arduino.   I'm doing a whole bunch of compiles and firmware uploads trying to get my new Delta printer debugged.

1. Every time I fire up Arduino I have to expand the size of it's window.  It always fires up too small.  And it is particularly annoying because I have to expand the error box at the bottom.  It is always too small and in order to grab the box edge (which is hidden) you have to be on exactly the right pixel.   I see a .ini file in in Arduino's directory.  Can I change the setting so it powers up nicely for me?

2. Can I change Arduino to leave the pre-processed source files laying around so I can see exactly what got compiled into the code and what did not?   With all of the nested #ifdef's it would be nice to just go look at what actually got compiled into my code.

Wackerbarth, 2015-10-28:
Unsure about your issue (1). If you can find a setting for the default window in the .ini file, quit out of the IDE and edit the file. (If you try to change the file while the IDE is running, things get overridden. But I have enough screen real-estate that I don't have an issue resizing the window when I want to actually read the compile log (which is the only thing I try to use in the  IDE window)

As for your desire to preserve the pre-compiler output, that is just a function of how you call the gcc compiler.  The Arduino IDE has a templateing scheme that generates the build commands that actually get executed. The fundamental templates are in the platform.txt section of the Marlin platform definition.
However, you should not change them directly. Instead, override specific template lines in a platform.local.txt file.

Roxy-3DPrintBoard, 2015-10-28:
Hmmmm...  Still no luck!    I went into platform.local.txt and changed:
~~~~
compiler.cpp.extra_flags=-DUSE_AUTOMATIC_VERSIONING -save-temps
~~~~

I can't find any .i files laying around.  I wonder if Arduino is deleting everything when it is done.   I also did what you told me not to do and tried adding -save-temps to a bunch of different lines in the platform.txt file just to see if I could get some .i files to stick around.   No luck there either.

Wackerbarth, 2015-10-28:
Where did you get that option? I am not familiar with it.
The compiler invoked is the gcc-avr compiler which is a derivative of the gcc compilers.
Any intermediate files that the compiler leaves behind will be somewhere within the /tmp/... build directory. I know that you can find the .o files which are the result of the compile and, thus, separate from the link. But I suspect that the standard behavior of the compiler is to delete the output of the pre-processor unless you supply the appropriate flag to cause it to save something.

One approach might be to specify an additional compile operation to regenerate your desired file after the .o file is generated.

Roxy-3DPrintBoard, 2015-10-28:
> Where did you get that option? I am not familiar with it.

That is a standard gcc option to save all temporary files.  There are extra options so you can control the format of the temporary files and even feed them back into the compiler as an input file.

You can go here and search for -save-temps  :  https://gcc.gnu.org/onlinedocs/gcc/Option-Summary.html#Option-Summary

Its documented a whole bunch of places.

Wackerbarth, 2015-10-29:
Perhaps you should try the -save-temps=cwd version and place those temps somewhere that Arduino doesn't expect them to be.

Roxy-3DPrintBoard, 2015-10-29:
Yeah!   That is a good plan.  I'll try that in the morning.

Wackerbarth, 2015-10-29:
@Roxy-3DPrintBoard -- I tried the -save-temps argument on my laptop. Unfortunately, without success.
However, I think that I may have found another issue that is affecting your use of it.
I suspect that you are running on a Windows-based system rather than one based on Linux or Mac OSX.

If that is the case, the `compiler.cpp.extra_flags` line that you quote is not active. For Windows-based systems, the `compiler.cpp.extra_flags.windows` line just below it overrides the general case.
You should adjust that line instead.

Unfortunately, I haven't figured out where the compiler is attempting to write the temp files that it generates when you add the `-save-temps` option.
I get `cc1plus: fatal error: opening output file MarlinFirmware.ino.ii: Permission denied`

Roxy-3DPrintBoard, 2015-10-29:
> For Windows-based systems, the compiler.cpp.extra_flags.windows line just below it overrides the general case. You should adjust that line instead.

> Unfortunately, I haven't figured out where the compiler is attempting to write the temp files that it generates when you add the -save-temps option. I get cc1plus: fatal error: opening output file MarlinFirmware.ino.ii: Permission denied

Yes, I'm using Windoze.  I did try messing with all the different lines in platform.local.txt without any luck.   I had that option on any line that looked reasonable.   And that was when I did what you told me to do and started adding it to different lines in the platform.txt file.

But at least you got it to start complaining about the .ii file it can't make.   I guess it makes .i files out of .c files and it makes .ii files out of .cpp files.   I don't get get any complaints yet.

ntoff, 2015-11-14:
My arduino stores stuff in C:\Users\(username)\AppData\Local\Arduino15 maybe try looking in there for preferences.txt 

There's options for window sizes and preproc.save_build_files=true leaves the files in your temp dir under a build89a7df987erg9087er9f87sg9f87sd8g97 (bunch of random numbers) dir.

closed 2016-04-26 by thinkyhead