MarlinFirmware/MarlinDev #259: LCD Panel to Assist with Firmware Debug

Roxy-3DPrintBoard, 2015-10-30:
I just wanted to throw an idea out there and get people to chime in with their thoughts.   I was debugging a bad pin number used by the Z-Axis stepper that conflicted with the pin number for my 2nd heater.   What would happen would be I could set the temperature, but as soon as I started printing my 2nd extruder would start cooling down.    This problem was my own doing because I fat fingered the key board and changed a 35 into a 3.  And as a result, two things thought they were on Pin 3.

What I want people to think about is this:   It would be nice to have the ability to set and clear little sections of the LCD Display to indicate debug status of different things as the firmware is running.   I found the problem before I got very far down this line.  But what I was thinking about doing was setting and clearing different locations of the LCD Panel to indicate the choices and code paths the firmware was running through.   The problem is the LCD Panel needs to keep running but I wanted to be able to update (and clear) some parts of it separately from what the LCD Panel code is doing.  And do that 'quickly' in code not even associated with the LCD Panel.

I was thinking of hi-jacking the last half of a line and not allowing the LCD Panel code to mess with it.  And I was going to use that space to set and clear flags I could watch as the firmware did things.

I know this isn't elegant, but it was my best thinking on how to get more real time information to help me debug the problem I was fighting.   Just for the sake of discussion, does anybody have better ideas or suggestions?  It might be nice to have a 'Debug' option we could turn on use the LCD Panel to give us real time flags.

AnHardt, 2015-10-30:
The idea for a debug-screen is not that new.
Last time we taught about it we came up with the idea to toggle between status- and debug-screen ba a loooooong press of the encoder button. But this was when all new features was set on hold.

Today i'd say the easiest way to implement, would be a 'Toggle Debug-screen' Menu item in the main menu. Replacing then the status-screen with the debug.screen routine is done fast.
The more complicated thing is finding out what, where and how to display on the screen. Ideally it would be configurable. A DISPLAY_MACRO(x,y,"optional text", x_to_str_function( expession_to_display)) could do the job.

By the way.
We collected a huge bunch of itostr and ftostr functions. Because of the huge amount of divisions and modulo divisions they have a lot of potential for speed improvements. Converting them to atoi() and dtostrf() would save a lot of progmem but i never benchmarked them. Somewhere i have an old benchmark-branch One day i'll use it again.

Roxy-3DPrintBoard, 2015-10-30:
I like those ideas!

thinkyhead, 2016-04-18:
The Debug Screen is still one of those things we would still like – one of these days… So let's keep this open.

closed 2016-05-27 by jbrazio