MarlinFirmware/MarlinDev #268: Prints smaller when size > X with 1.1.0-RC2

Nandox7, 2015-11-04:
Ok, the tittle is odd but it was hard to explain. :)

So what I noticed is that when I tried to print a part that was 185mm in length it ended up 2mm short (~183mm).
I thought it could be the part or the slicer but ended up not.

So far I've covered the following possibilities:
- Create a test rectangular part with 185mm x 80mm in size
- Printed in both orientations (X and Y oriented) - Same result it always prints ~2mm less in the longer side
- Tried two different slicers (Cura & Simplify3D) - Same result

I haven't tested rolling back to a older FW version but so far I excluded problems with the part and slicer.
So all that is left is the printer. It could be a steps/mm issue but as seen when I change the part oriented in 90 degrees the same issue happens, so it's not one axis that has incorrect steps/mm 
(also a calibration cube of 20x20x20 gives me ~0.10mm more in both X and Y so in theory my part would even have to be a bit bigger than 185mm)

So only think I can think off now is the fw, for some reason it reduces the size. I haven't figure out yet when it starts to do that (thus the X in the subject line :-) ) but at least 80mm is fine. 
I may need to increate gradually until I find the point where it fails.

KiteLab, 2015-11-04:
```
/*
Here are some standard links for getting your machine calibrated:
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
*/

```

Nandox7, 2015-11-04:
Thanks, my printer is calibrated. :)
Like I mentioned calibration parts get out with a less than 0.10mm difference.

KiteLab, 2015-11-04:
A single calibration part tells a to short story.
At a 20x20x20 cube you could have to few steps/mm and a bit of over-extrusion to have a perfect part.
The failure in the steps/mm will increase on larger distances, the failure of the over-extrusion will not.

Wackerbarth, 2015-11-24:
Create a part that is the profile of about 7 steps where each step is, for example, 10 mm high and 25 mm wide (all that you should need is the outline. A mm, or so, high should be enough.  Then measure each of the steps as well as the long straight bottom and see if everything is uniform.

Nandox7, 2015-12-01:
Spot on.
All the details from the printer maker were to use 20x20x20mm calibration cubes and that is all I ever used.
I managed to get it down to less that 0.1mm difference but now that I started printing bigger parts where precise matters this issue showed up.

After a few test it was off by 2mm for both X and Y. Now after tweaking the steps/mm it is printing correctly.

Maybe worth adding to the FW documentation that calibration using bigger parts is recommended.
(in case is not there yet)

Thanks!

closed 2015-12-01 by Nandox7