MarlinFirmware/MarlinDev #269: M48 Working on Delta's  ---  Need somebody to test it with a Cartesian Printer

Roxy-3DPrintBoard, 2015-11-04:
I think I have M48 (Z-Probe Repeatability Test) working on Delta Printers.   The problem is I don't have a Cartesian Printer that I can test it against right now.    I was pretty careful to make sure everything I did would work on both types of machines.   But a lot had to be changed to make the code work with Delta's.  

I need a volunteer to test the code on a Cartesian Printer with a reasonably current software base.   It should be as easy as just replacing the existing M48 code in Marlin_main.cpp with the updated version.

[M48.txt](https://github.com/MarlinFirmware/MarlinDev/files/27235/M48.txt)

KiteLab, 2015-11-04:
Be carefull!
Slamed the nozzle into the bed!

```
21:21:32.795 : N41 G28 *6
21:21:38.451 : ok
21:21:47.736 : N43 M48 P5 X70 Y70 V4 *14
21:21:47.755 : M48 Z-Probe Repeatability test
21:21:47.755 : Positioning the probe...                        // BANG!!! 
21:22:24.636 : Bed X: 70.000 Y: 70.000 Z: -111.000
21:23:33.494 : Bed X: 70.000 Y: 70.000 Z: -111.000
21:23:33.497 : 1 of 5   z: -111.000000 mean: -111.000000   sigma: 0.000000

```

Roxy-3DPrintBoard, 2015-11-04:
Damn!

It did it at the 'Positioning the probe...' line?

KiteLab, 2015-11-04:
Yes. Or just after that, on the first way down.
The machine was a prusa i3 with inductive probe.

Ouptput for the former M48 was 
```
Old:

21:09:28.681 : N21 G28 *0
21:09:34.318 : ok
21:09:47.045 : N23 M48 P5 X70 Y70 V4 *8
21:09:47.048 : M48 Z-Probe Repeatability test
21:09:48.211 : Positioning the probe...
21:09:55.096 : 1 of 5   z: 0.511250 mean: 0.511250   sigma: 0.000000
21:09:58.835 : 2 of 5   z: 0.512000 mean: 0.511625   sigma: 0.000375
21:10:02.574 : 3 of 5   z: 0.513250 mean: 0.512167   sigma: 0.000825
21:10:06.315 : 4 of 5   z: 0.514250 mean: 0.512687   sigma: 0.001151
21:10:10.054 : 5 of 5   z: 0.514500 mean: 0.513050   sigma: 0.001259
21:10:12.212 : Mean: 0.513050
21:10:12.216 : Standard Deviation: 0.001259
21:10:12.216 : ok
```

Roxy-3DPrintBoard, 2015-11-04:
If you are willing to give it one more try....    I think replacing the  #ifdef DISABLED(DELTA) with this code will make things better.   

~~~~
   #if DISABLED(DELTA)
    plan_bed_level_matrix.set_to_identity(); 	// we don't do bed level correction in M48 because we want the raw data when we probe
   #else
    reset_bed_level();	// we don't do bed level correction in M48 because we want the raw data when we probe
   #endif
    if ( Z_start_location < Z_RAISE_BEFORE_PROBING*2.0) {
      do_blocking_move_to_z( Z_start_location);
    }
    do_blocking_move_to_xy( X_probe_location-X_PROBE_OFFSET_FROM_EXTRUDER, Y_probe_location-Y_PROBE_OFFSET_FROM_EXTRUDER);
~~~~

KiteLab, 2015-11-04:
No difference. Crash at the same first way down.

Roxy-3DPrintBoard, 2015-11-04:
OK!  Thank You for the help!   I''m going to have to get a Cartesian printer I can load this on.   I'm sorry about the Nozzle Crash!!!!

KiteLab, 2015-11-04:
I can't find a `setup_for_endstop_move();` in the new code.

KiteLab, 2015-11-04:
```
    //
    // OK, do the initial probe to get us close to the bed.
    // Then retrace the right amount and use that in subsequent probes
    //

    enable_endstops(true); //!!!!!!!!!!!!!!!!!!!!!!!
    
    if (deploy_probe_for_each_reading)
       probe_pt( X_probe_location, Y_probe_location, Z_RAISE_BEFORE_PROBING, ProbeDeployAndStow, verbose_level );
    else
       probe_pt( X_probe_location, Y_probe_location, Z_RAISE_BEFORE_PROBING, ProbeDeploy, verbose_level );
```

Makes all the difference. Log will follow.

KiteLab, 2015-11-04:
```
Old:

21:09:28.681 : N21 G28 *0
21:09:34.318 : ok
21:09:47.045 : N23 M48 P5 X70 Y70 V4 *8
21:09:47.048 : M48 Z-Probe Repeatability test
21:09:48.211 : Positioning the probe...
21:09:55.096 : 1 of 5   z: 0.511250 mean: 0.511250   sigma: 0.000000
21:09:58.835 : 2 of 5   z: 0.512000 mean: 0.511625   sigma: 0.000375
21:10:02.574 : 3 of 5   z: 0.513250 mean: 0.512167   sigma: 0.000825
21:10:06.315 : 4 of 5   z: 0.514250 mean: 0.512687   sigma: 0.001151
21:10:10.054 : 5 of 5   z: 0.514500 mean: 0.513050   sigma: 0.001259
21:10:12.212 : Mean: 0.513050
21:10:12.216 : Standard Deviation: 0.001259
21:10:12.216 : ok
21:10:27.800 : N29 G28 *8
21:10:34.177 : ok
21:10:38.885 : N31 M48 P5 X130 Y70 V4 E L5 *2
21:10:38.913 : M48 Z-Probe Repeatability test
21:10:40.076 : Positioning the probe...
21:10:44.343 : x: 123.90y: 31.48
21:10:44.442 : x: 117.57y: 31.59
21:10:44.515 : x: 114.99y: 27.98
21:10:44.573 : x: 0.00y: 200.00
21:10:50.040 : 1 of 5   z: 0.662000 mean: 0.662000   sigma: 0.000000
21:10:52.203 : x: 146.07y: 39.15
21:10:52.355 : x: 200.00y: 200.00
21:10:53.108 : x: 200.00y: 200.00
21:10:53.112 : x: 200.00y: 200.00
21:10:57.540 : 2 of 5   z: 0.662250 mean: 0.662125   sigma: 0.000125
21:10:59.707 : x: 159.89y: 17.39
21:10:59.887 : x: 162.95y: 21.73
21:10:59.948 : x: 166.19y: 27.69
21:11:00.018 : x: 200.00y: 200.00
21:11:05.227 : 3 of 5   z: 0.655000 mean: 0.659750   sigma: 0.003360
21:11:07.394 : x: 116.80y: 33.67
21:11:07.517 : x: 0.00y: 200.00
21:11:08.414 : x: 0.00y: 200.00
21:11:08.414 : x: 0.00y: 200.00
21:11:12.976 : 4 of 5   z: 0.657250 mean: 0.659125   sigma: 0.003105
21:11:15.144 : x: 200.00y: 0.00
21:11:15.500 : x: 200.00y: 0.00
21:11:15.504 : x: 200.00y: 0.00
21:11:15.504 : x: 200.00y: 0.00
21:11:19.444 : 5 of 5   z: 0.667250 mean: 0.660750   sigma: 0.004275
21:11:21.628 : Mean: 0.660750
21:11:21.628 : Standard Deviation: 0.004275
21:11:21.632 : ok

New:

22:55:39.299 : N58 G28 *14
22:55:44.535 : ok
22:55:54.488 : N61 M48 P5 X70 Y70 V4 *14
22:55:54.493 : M48 Z-Probe Repeatability test
22:55:54.496 : Positioning the probe...
22:55:58.498 : Bed X: 70.000 Y: 70.000 Z: 0.501
22:56:02.754 : Bed X: 70.000 Y: 70.000 Z: 0.503
22:56:02.758 : 1 of 5   z: 0.502500 mean: 0.502500   sigma: 0.000000
22:56:06.715 : Bed X: 70.000 Y: 70.000 Z: 0.503
22:56:06.715 : 2 of 5   z: 0.503500 mean: 0.503000   sigma: 0.000500
22:56:10.670 : Bed X: 70.000 Y: 70.000 Z: 0.504
22:56:10.674 : 3 of 5   z: 0.504250 mean: 0.503417   sigma: 0.000717
22:56:14.626 : Bed X: 70.000 Y: 70.000 Z: 0.505
22:56:14.631 : 4 of 5   z: 0.504750 mean: 0.503750   sigma: 0.000848
22:56:18.603 : Bed X: 70.000 Y: 70.000 Z: 0.505
22:56:18.607 : 5 of 5   z: 0.505250 mean: 0.504050   sigma: 0.000967
22:56:18.607 : Mean: 0.504050
22:56:18.607 : Standard Deviation: 0.000967
22:56:18.607 : ok

22:56:25.348 : N65 G28 *0
22:56:30.575 : ok
22:56:39.072 : N67 M48 P5 X130 Y70 V4 E L5 *1
22:56:39.079 : M48 Z-Probe Repeatability test
22:56:39.079 : Positioning the probe...
22:56:43.047 : Bed X: 130.000 Y: 70.000 Z: 0.679
22:56:44.710 : 
22:56:44.714 : Starting radius: 19.00   theta: 3.65 Direction: Counter Clockwise
22:56:44.714 : Going to:  x: 0.00  y: 0.00  z: 4.68
22:56:45.321 : Going to:  x: 0.00  y: 0.00  z: 4.68
22:56:45.324 : Going to:  x: 0.00  y: 0.00  z: 4.68
22:56:45.324 : Going to:  x: 0.00  y: 0.00  z: 4.68
22:56:48.523 : Bed X: 130.000 Y: 70.000 Z: 0.682
22:56:48.527 : 1 of 5   z: 0.681500 mean: 0.681500   sigma: 0.000000
22:56:49.027 : 
22:56:49.031 : Starting radius: 36.00   theta: 3.60 Direction: Clockwise
22:56:49.031 : Going to:  x: 85.06  y: 72.36  z: 0.68
22:56:49.363 : Going to:  x: 0.00  y: 200.00  z: 0.68
22:56:50.051 : Going to:  x: 0.00  y: 200.00  z: 0.68
22:56:50.055 : Going to:  x: 0.00  y: 200.00  z: 0.68
22:56:54.380 : Bed X: 130.000 Y: 70.000 Z: 0.680
22:56:54.384 : 2 of 5   z: 0.679500 mean: 0.680500   sigma: 0.001000
22:56:54.883 : 
22:56:54.889 : Starting radius: 47.00   theta: 5.36 Direction: Counter Clockwise
22:56:54.892 : Going to:  x: 168.03  y: 34.54  z: 0.68
22:56:55.113 : Going to:  x: 180.41  y: 50.65  z: 0.68
22:56:55.241 : Going to:  x: 200.00  y: 0.00  z: 0.68
22:56:55.515 : Going to:  x: 200.00  y: 0.00  z: 0.68
22:56:59.222 : Bed X: 130.000 Y: 70.000 Z: 0.685
22:56:59.225 : 3 of 5   z: 0.685250 mean: 0.682083   sigma: 0.002383
22:56:59.725 : 
22:56:59.729 : Starting radius: 38.00   theta: 1.88 Direction: Clockwise
22:56:59.729 : Going to:  x: 132.04  y: 108.95  z: 0.69
22:57:00.147 : Going to:  x: 200.00  y: 200.00  z: 0.69
22:57:00.667 : Going to:  x: 200.00  y: 200.00  z: 0.69
22:57:00.672 : Going to:  x: 200.00  y: 200.00  z: 0.69
22:57:04.882 : Bed X: 130.000 Y: 70.000 Z: 0.674
22:57:04.886 : 4 of 5   z: 0.674000 mean: 0.680063   sigma: 0.004063
22:57:05.386 : 
22:57:05.390 : Starting radius: 1.00   theta: 0.19 Direction: Counter Clockwise
22:57:05.393 : Going to:  x: 139.06  y: 74.23  z: 0.67
22:57:05.668 : Going to:  x: 200.00  y: 200.00  z: 0.67
22:57:06.295 : Going to:  x: 200.00  y: 200.00  z: 0.67
22:57:06.299 : Going to:  x: 200.00  y: 200.00  z: 0.67
22:57:10.501 : Bed X: 130.000 Y: 70.000 Z: 0.678
22:57:10.505 : 5 of 5   z: 0.678500 mean: 0.679750   sigma: 0.003688
22:57:10.505 : Mean: 0.679750
22:57:10.509 : Standard Deviation: 0.003688
22:57:10.509 : ok
```
A
```
#if ENABLED(ENDSTOPS_ONLY_FOR_HOMING)
  enable_endstops(true);
#endif
```
at the end, to clean up, is needed.

KiteLab, 2015-11-04:
Z_RAISE_BETWEEN_PROBINGS works. Was set to 0 in the logs above.

Roxy-3DPrintBoard, 2015-11-04:
Yikes!!! Can you post your working M48.cpp file?  (But it has to be a .txt file)   The thing that is confusing me is probe_pt() uses run_z_probe() which should handle that stuff.  But looking at G29 they do just what you say and enable_endstops();    

So now the question I have is:  Why does this work on a Delta????

KiteLab, 2015-11-04:
setup_for_endstop_move();
clean_up_after_endstop_move();

is the better pair to use, because of restoring the feedrates.

KiteLab, 2015-11-04:
I guess not the DELTA makes the difference but the ENDSTOPS_ONLY_FOR_HOMING.

Roxy-3DPrintBoard, 2015-11-04:
Ah!!!!!    OK, well...  It has to work for the generic case!!!

KiteLab, 2015-11-04:
<script src="https://gist.github.com/KiteLab/80b34342a1e89a9d181e.js"></script>

Roxy-3DPrintBoard, 2015-11-04:
Code still works on Delta's (as expected) !!!  

Thanks Again!      

We need to get the code_value() functions to ignore white space like they used to.   It kind of did the wrong thing the first time I fired it up because I had spaces between the numbers and the parameters.

AnHardt, 2015-11-04:
Nice job!
https://www.youtube.com/watch?v=gtCgR86rkgE 
is a video from a i3 with a inductive probe performing
M48 P5 X130 Y70 V4 E L5
with a connected servo.

Looks promising.

Roxy-3DPrintBoard, 2015-11-05:
Thanks!   BUT....   I found a bug.   There is an overflow happening on the mod (%) operation to generate a random number....

AnHardt, 2015-11-05:
@Roxy-3DPrintBoard 
I can't see the circle shape at all. So why calculate in polar coordinates at all?

Simplified the `if (n_legs) {` part a bit.

```
      if (n_legs) {
        for (uint8_t l = 0; l < n_legs - 1; l++) {
          X_current = X_MIN_POS + (X_MAX_POS - X_MIN_POS)/200 * random(200);
          Y_current = Y_MIN_POS + (Y_MAX_POS - Y_MIN_POS)/200 * random(200);
          
          #if ENABLED(DELTA)
            // If we have gone out too far, we can do a simple fix and scale the numbers
            // back in closer to the origin.
            while( sqrt( X_current*X_current + Y_current*Y_current ) > DELTA_PROBABLE_RADIUS) {
              X_current /= 1.25;
              Y_current /= 1.25;
            }
          #endif
  
          if (verbose_level > 3) {
            SERIAL_PROTOCOL("Going to:");
            SERIAL_ECHOPAIR("  x: ", X_current);
            SERIAL_ECHOPAIR("  y: ", Y_current);
            SERIAL_ECHOPAIR("  z: ", current_position[Z_AXIS] );
            SERIAL_EOL;
          }
  
          do_blocking_move_to_xy( X_current, Y_current );
          st_synchronize();
        } // n_legs loop
      } // n_legs
```
We don't need real random here. We just want some movement.

Roxy-3DPrintBoard, 2015-11-05:
It is a little bit early to scream "Compiler Bug!" in a crowded theater.   But check this out.   What ever is going wrong here is what is keeping the printer from making nice circular arcs.  If you Cut & Paste the attached M48_gcc_bug.txt file over the front end of your Marlin_main.cpp's M48 function you will get code that prints out all of the sizes of the different data types and prints them.  It also has code to print hex nibbles, hex bytes, hex words, and hex longs.

At the end of this code, it declares an unsigned long (ul) and assigns it a value of 0x12345678L  But only the low word of the value gets stored in ul ????   You might think the prt_hex_long() function is broke, but notice I shift ul<<4  and then shift it ul>>4.    We lose the top hex digit by doing that in the double checking and triple checking part.
~~~~
int ul;
ul = 0x12345678L;
prt_hex_long("checking unsigned long = ", ul );

ul = ul << 4;
prt_hex_long("double checking unsigned long = ", ul );

ul = ul >> 4;
prt_hex_long("triple checking unsigned long = ", ul );
~~~~

SENDING:M48 X0 Y0 P4 V4 L 6
sizeof(short): 2
sizeof(int): 2
sizeof(long): 4
sizeof(unsigned): 2
sizeof(float): 4
sizeof(double): 4
sizeof(void *): 2
sizeof(unsigned long): 4
checking unsigned long = 00005678
double checking unsigned long = 00006780
triple checking unsigned long = 00000678
M48 Z-Probe Repeatability test
Positioning the probe...
Put down Z-Probe.

[M48_gcc_bug.txt](https://github.com/MarlinFirmware/MarlinDev/files/28154/M48_gcc_bug.txt)

Roxy-3DPrintBoard, 2015-11-05:
Oooops....  I had a data type declared wrong!!!

Roxy-3DPrintBoard, 2015-11-05:
@AnHardt 
> I can't see the circle shape at all. So why calculate in polar coordinates at all?

Agreed there is an issue.  The Arduino documentation is horrible.  Check out this page:  http://labs.arduino.org/Sin

It says   rad: an angle measured in radians (PI to -PI)

But the cos() documentation at:

has an example:

double a=cos(90); // a=-0.45;

Huh???    And no restrictions on the size of the input angle.  The 90 is misleading.  It sort of gets people thinking the input is in degrees, and it isn't.

thinkyhead, 2016-04-18:
I suspect we got everything pretty solid in `RCBugFix` now, so let's close this for now….

closed 2016-04-18 by thinkyhead