MarlinFirmware/MarlinDev #274: (Why) Does the ARDUINO 1.0.6 with 1.0.x branch work better with under certain circumstances?

Wackerbarth, 2015-11-07:
It has been stated that firmware built with the Arduino 1.0.6 IDE and older branches works better than newly compiled versions. I'm not disputing that, but, assuming it to be true, would like to isolate the cause.

The following aspects differ:

1) The current IDE uses a newer version of the gcc-avr compiler. (Perhaps the generated code is not quite as efficient)
2) The "core" routines and libraries have been changed.
3) The Marlin code itself has changed. (Are we now trying to do too much in the ISR?)

Additionally, there may be some issues that are the result of pin renumbering, etc. which has caused us to lose support for a particular board. -- This is just a compatibility issue which could be fixed by updating the definition files.

Using the latest IDE (1.6.6), it is possible to eliminate each of the three areas that I identify above.
However, it is desirable to maintain as much compatibility as possible. Therefore, I would like to see a discussion of the underlying issues so that we can address them (and leave the remainder to default to the current Arduino release.

So, I ask that those who know provide some insight into this issue.

Roxy-3DPrintBoard, 2015-11-07:
Can you define what "Works better than newly compiled versions means?"     What kind of difference are we looking for?

Wackerbarth, 2015-11-07:
That's what I want to learn and identify. I'm not the one who made the claim.

AnHardt, 2015-11-07:
As far as I know the history for this:
When we tagged 1.0.2 we all used Arduino 1.0.x.
Around that time some one began to use Arduino 1.5.x and got en error when compiling with 2 or more extruders.
https://github.com/MarlinFirmware/Marlin/issues?utf8=%E2%9C%93&q=is%3Aissue+POINTER_REGS

I think this was the first:
https://github.com/MarlinFirmware/Marlin/issues/1523

We tried a bit back and forward and finally got a fix. I think it was a simplification of the ISR code by shifting out some code to a function.

Since then the code compiles flawlessly under 1.5.x and following Arduino IDEs.

Because the code was changed so much until the fix worked this was never back ported to 1.0.2.

Roxy-3DPrintBoard, 2015-11-07:
> Since then the code compiles flawlessly under 1.5.x and following Arduino IDEs.

Apparently, the code compiles flawlessly all the way up to 1.6.5 because that is what I'm using with RC2 right now.

closed 2016-04-18 by thinkyhead